realsenseai/librealsense #6144: OpenNI2 wrapper build for Windows x64 missing

mediadog, 2020-03-28:
The build instructions using CMake produce a solution/projects that only have x86 as an option.

The vcpkg method of building librealsense generates both x86 and x64 debug and release versions.

Can either the CMake OpenNI2 build add x64, or the wrappers be added to the vcpkg build method?

dorodnic, 2020-03-29:
Hi @mediadog 
You can absolutely build x64 with CMake, including the OpenNi2 wrapper. The option to select build platform is given when first running configure in CMake.

mediadog, 2020-03-29:
Yes, you are right - for some reason when using VS 2019 it defaults to x64, but with VS 2017 it defaults to x86 and I missed that.  It would be good to note in the build instructions that the machine type has to be specified there, and not in VS via the pulldown later, thanks.

But the larger problem is still here, which is if I build librealsense using the method in Readme.md via vcpkg (which fails on Win7 unless you use the fix in PR #10550), the wrappers don't get built and then you have to use the Cmake route which is not at all obvious how that should be done in the vcpkg environment, and requires some manual very non-obvious tweaking to get to build, and then the wrapper does not work anyway (I'll file new bugs on all that).

Ideally these would just be built and included in the SDK, so I could just drop the OpenNI2 drivers into my existing driver folder and do happy work.  Next best would be some clear OpenNI2 build instructions that work with the vcpkg build method.  Note that if I do not use the vcpkg method at all, and use CMake, then the build fails with an undefined symbol GUID_DEVINTERFACE_USB_DEVICE, which looks like INITGUID is not getting set somewhere (I'll also file a bug on that).  So I can only get a good build using the vcpkg method.

closed 2021-01-11 by RealSenseSupport