realsenseai/librealsense #6148: D435 not recognized on Win7 with 2.33.1, OK with 2.17.1

mediadog, 2020-03-28:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version |   Win7 x64 Pro  | 
| Platform                                 | PC ThinkPad X230 |
| SDK Version                            |  2.33.1    | 

### Issue Description
On a Lenovo ThinkPad X230, after installing the latest (2.33.1) SDK package, Realsense Viewer no longer sees a connected D435.  It still shows up with the same two listings in the Device Manager though with no errors.  Replugging/rebooting does not help.  Uninstalling 2.33.1 and going back to 2.17.1 then it works fine.

However on a desktop PC running Win7 x64 Enterprise, 2.33.1 works fine.  I suspect it is a USB chipset issue exposed by the new driver.  The X230 lists the USB controller as "Intel(R) Series/C216 Chipset Family USB Enhanced Host Controller".

I will try going back to 2.33.1 leaving the 2.17.1 driver (23.56.40.396) in place to see if it works then, and will update here.

mediadog, 2020-03-28:
Hmmm, well after installing 2.33.1 driver (which is unsigned BTW) the driver listed for both the D435 RGB and Depth is still 23.56.40.396.  Is the old driver getting in the way, or is it in fact still the 2014 Win7 driver in that latest SDK?

RealSenseCustomerSupport, 2020-04-02:
----------------------------------------------

In the 2.30.0 release note,  installing a new Win7 driver is needed for working with new SDK.
Hope it may give some clues for what you are experiencing. Thank you.
#4975 of https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2300

mediadog, 2020-04-02:
I think you missed my point - I installed 2.33.1, and the install Win7 driver box was checked.  It popped up an unsigned driver warning, and I said install anyway, so it did appear to try and install the driver.  Yet when I went and checked in Device Manager what driver was being used with the D435 RGB and Depth USB devices, it was still the 2014 driver.  And still not being recognized by the 2.33.1 Viewer.

Since this works on a desktop Win7 system, but not this laptop, I suspect there is some issue possibly with the USB chipset that is keeping the D435 from being properly recognized by the new driver, or the driver installation is not working properly.  The unsigned driver part is very suspicious.

RealSenseCustomerSupport, 2020-04-03:
----------------------------------------------

Hi @mediadog, thank you for the clarification. Have you already tried to manually uninstall the old driver on X230 then reinstall the 2.33.1 as mentioned in the following issue?
https://github.com/IntelRealSense/librealsense/issues/5193

mediadog, 2020-04-04:
I manually uninstalled the driver for both depth and "color", and even selected delete the software, and then installed 2.33.1 with the driver.  Still no go; when I plugged in the D435 I got a message the driver installs failed.  The two D435 devices (RGB and Depth) now show up under imaging, but cannot start, and the driver details now show a 2006 Microsoft driver (see attached screenshot).  Maybe the new driver is failing, so it falls back to a generic driver?  Or is the new driver being preempted by an old one?

Happy to send along logs if you can tell me where to find them, or repeat the process to get some if needed.

[image]

RealSenseCustomerSupport, 2020-04-29:
----------------------------------------------

Hi @mediadog,
Just curious if you have somehow resolved this issue.

> Maybe the new driver is failing, so it falls back to a generic driver?

This could be the case. Was there a pop up warning when running the installer asked to verify the driver software before it can be installed?
In addition to that, if you click on that "update driver" button > Browse for driver > Pick from a list of drivers, is the new driver listed in the drop down list? If so, please pick the new driver.

This case will be closed after 7 days if there is no further assistance needed.
Thank you!

closed 2020-05-07 by RealSenseCustomerSupport