realsenseai/librealsense #6153: duplicate udev rules in /etc and /lib ?

weblackey, 2020-03-29:
When following all the instructions for installing the linux distribution with sources on a Jetson Nano with latest Jetpack, we end up with udev rules in both etc and lib and realsense-viewer complains about it. 

why are the duplicates installed, and which one should be deleted ?

MartyG-RealSense, 2020-03-29:
Which error does the RealSense Viewer give you, please?

weblackey, 2020-03-29:
Here is the message:

Timestamp: 1585499111721.604492
Severity: Warn
Description: Multiple realsense udev-rules were found! :
1:/etc/udev/rules.d/99-realsense-libusb.rules
2: /lib/udev/rules.d/60-librealsense2-udev-rules.rules
Make sure to remove redundancies!

MartyG-RealSense, 2020-03-29:
There was a previous case of this error.

https://github.com/IntelRealSense/librealsense/issues/4951

 In that user's case, the multiple udev rules error  occurred after doing an apt install.

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

The eventual solution in their case was to install Librealsense the "manual" way, which I take to mean compiling from source.  Compiling from source is recommendable for an installation on Jetson Nano.

weblackey, 2020-03-29:
thanks for the info. We did not have missing udev rules at any point in time an definitely never installed any udev rules manually.

Our main question is to determine which of these duplicates should be deleted, and of course it would be nice to know why the librealsense installation scripts install both of them in the first place

MartyG-RealSense, 2020-03-29:
I do not have the knowledge of the udev rules to advise on that subject, unfortunately.  One of the other RealSense support team members may be able to help with that in the coming week.

As there is more involved in a Jetson Nano installation process than the details in the standard installation guide, I would recommend removing the current install and following the guide on the JetsonHacks website for installing Librealsense on Nano.

https://github.com/JetsonHacksNano/installLibrealsense

weblackey, 2020-03-29:
ok thank you so much for the pointers and the link to jetsonhacks

it seems if we delete /lib/udev/rules.d/60-librealsense2-udev-rules.rules then it works with no error messages with the D435. Still have issues with D435i but we've given up on that one since simply swapping the D435 for the D435i makes it work... same system same cable same usb port same firmware and with only one udev rule file it would also be guaranteed to be the same udev rule

MartyG-RealSense, 2020-03-29:
I'm very glad I could be of help.  Thanks for sharing your information about the rules in case somebody else experiences a similar problem!

ev-mp, 2020-03-29:
@weblackey , the file ` /lib/udev/rules.d/60-librealsense2-udev-rules.rule` is part of the Debian package `librealsense2-udev-rules`.  I.e. the file has been installed with 
>sudo apt install librealsense2-udev-rules"

or any other Librealsense Debian package (all packages rely on that one).
Deleting it manually should be avoided as it will break Debian packages integrity.

On the other hand the udev file found under `/etc` folder is part of the manual installation. 
Apparently you've triggered both installations thus having both versions installed.  

As the manually-installed file is a stand-alone entity it is recommended to remove that one unless.
The only scenario where you'd might need to remove the Debian package with udev rules is when the Debian SDK package version is not up to date so that you download and build the latest version from source

weblackey, 2020-03-30:
> On the other hand the udev file found under /etc folder is part of the manual installation.
> As the manually-installed file is a stand-alone entity it is recommended to remove that one.

Done, and it works on D435 with no errors. D435i has issues, probably unrelated

closed 2020-03-30 by weblackey

fabiang95, 2022-03-30:
hi, is it possible to share the steps to remove the unnecessary udev rules?

MartyG-RealSense, 2022-03-30:
Hi @fabiang95  It is not a procedure that I have performed myself, but as far as I am aware it is simply a matter of navigating to the **/etc/udev/rules.d/** folder in Linux and deleting a file called **99-realsense-libusb.rules**

This is the only udev file that should be deleted.  Do not delete the rule at **lib/udev/rules.d/60-librealsense2-udev-rules.rules**