realsenseai/librealsense #6155: frames timestamp

galikrosentalik, 2020-03-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400, T200 | 
| Firmware Version                   | 0.0.18.5715 | 
| Operating System & Version |Linux (Ubuntu 18)| 
| Kernel Version (Linux Only)    |  4.18.0-25 | 
| Platform                                 | PC  |
| SDK Version                            |  2.33.1 | 
| Language                            |  C++ | 
| Segment			| Robot | 

### Issue Description
Hi all, I am using D415, D435 and T265 devices, and in order for proper synchronization between all, we are fighting very hard to get to the bottom of all timestamp provided by the frame-metadata.
I already found similar issues, such as #2188 and #2922, still not successful to solve our puzzle.

- Is it possible for me to get some kind of efficient evaluation for the timestamp which the data was sample by your device sensor? (in such a meaner to be translated to pc clock)   
- Is it possible for me to query the device for his clock, so I can conclude diffs between clocks?

Thank's.

dorodnic, 2020-03-30:
If timestamp domain you are getting (frame::get_timestamp_domain) is global time, then you already have hardware timestamp aligned to host clock.
Internally we are querying device time and adjusting the clocks but we do not expose it as an API and do not recommend doing it manually since there are a ton of corner cases needed to be handled.
Of course, the whole thing is open-source, so you are welcomed to read the code and modify it:
https://github.com/IntelRealSense/librealsense/blob/master/src/global_timestamp_reader.cpp#L237
The key limitation, as far as I know, is enabling hardware timestamps to begin with (either via kernel patches or compilation flags)

r91andersson, 2020-03-30:
@dorodnic You mean that the SDK internally is getting the hardware timestamp for each of the frames, and you are then syncing the timestamp to the host clock? And this is the default mode? So I can assume that when I subscribe to a topic, these frames are synced, and that the timestamp corresponds to the time when the frame was taken (translated to ROS-time)?

dorodnic, 2020-03-30:
Yes, this is the default when available (D400, L500, T260, but not SR300 at the moment, and requiring hardware timestamps to be available). You can dynamically check using get_timestamp_domain. 
Of course, this does not equal hardware sync between devices. Moreover different sensors have different attributes (global vs rolling shutter for example) and there are other effects related to exposure and motion blur. But it should eliminate the factor of OS latency per device.

galikrosentalik, 2020-04-03:
@dorodnic, I did find out that (frame::get_timestamp_domain) is set to global, so I started to test the frame::get_timestamp.
Do you have any scale for what should I expect has normal latency from the D415/D435 devices?
I am working with 30 fps, taking both depth&RGB.
Please have a look to the screenshot:  

[image]

The graph represent latency as a function of exposure.
I subtract frame::get_timestamp from the timestamp my app is getting the frame.
Also i compared latest RGB frame to the previous one, bitwise, in case they were different, represent in red, the same in blue.
can you confirm the data looks as expected?

closed 2020-12-15 by RealSenseSupport