realsenseai/librealsense #6156: How to create the vectors in code?

duongbaoquy, 2020-03-30:
Im using D435i camera and making a new tool by my own - but im trying to add the vetors like this (in 3D mode):
[image]
So my question is how to create such vectors? I came across the "motion" example but it moved with the camera - now I want to create vectors that are just standing still like in the realsense viewer.

dorodnic, 2020-03-30:
Hi @duongbaoquy 
The realsense-viewer is using a mix of OpenGL2 (old-school OpenGL) and OpenGL3 (modern OpenGL). This approach is generally not recommended, since it has limitations on performance and portability, but if you are interested, the "vectors" are rendered using the following routine:
https://github.com/IntelRealSense/librealsense/blob/master/common/rendering.h#L1297-L1352

duongbaoquy, 2020-03-30:
ok, thanks for the reply

closed 2020-03-30 by duongbaoquy

duongbaoquy, 2020-04-01:
Thanks for your help - i made it 
But now I cant show camera on it, is it right if i still use command "draw_pointcloud" or kind of " app.show()"?