realsenseai/librealsense #6159: Thin-film linear polarizers

jorge1993, 2020-03-30:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | Any | 
| Operating System & Version |   Win (10)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Hi, 

At this article [https://dev.intelrealsense.com/docs/optical-filters-for-intel-realsense-depth-cameras-d400](article), you mention how to use a  film polarizers to reduce the saturation. 

Which one are you using? or just using any linear polarizers?

MartyG-RealSense, 2020-03-30:
The Chief Technical Officer of the RealSense Group at Intel (agrunnet) recommends Astra Products as a source for filters.

https://github.com/IntelRealSense/librealsense/issues/2875#issuecomment-446793681

agrunnet, 2020-03-30:
Sorry, Astra doesn’t make polarizers just various spectral optical filters. So yes any think film polarizer will do that is linear. The circular polarizers used most often in 3D glasses won’t work.

closed 2020-03-31 by jorge1993