realsenseai/librealsense #6165: D435i lower performance than D435 ?

weblackey, 2020-03-30:
So we've been having lots of problems with this brand new D435i that we got from the Intel RealSense store. Wondering if it is defective or if there is a design issue. In previous issues here we discussed how under some conditions this D435i is not recognized on the same Nvidia Jetson system that has no trouble with a D435 with same firmware (either 5.11.1 or 5.11.15 or 5.12 doesn't matter we tried all)

To summarize previous results, librealsense only works on D435i when compiled with non-default flags while D435 works either way. Now, ignoring all that, and concentrating only on the apt package install of the binary release 2.32.1 which works on both D435 and D435i we noticed the following major difference:

The stereo camera on D435 works in realsense-viewer at the default resolution of 848x480 @ 30fps
The stereo camera on D435i does not work in realsense-viewer at the default resolution of 848x480 @ 30 fps, and only works in realsense-viewer if we drop resolution to 640 x 480 @ 30 fps

Both are showing in realsense-viewer as being connected via USB 3.2. The cameras are used only one at a time, and it doesn't seem to matter if the system is rebooted in between swapping the camera or not. The results are consistent. And we used the same cable on both, same USB port.

dorodnic, 2020-03-30:
Hi @weblackey 
Discovery issues on the Jetson are related to Linux kernel build used for that system. If I recall correctly relevant HID kernel module is not being loaded by Jetson Linux distribution. If you are willing to reconfigure the system, everything should be working with default parameters as well. We provide alternative solution (also via apt-get install) to help setting up the device faster, but it should be functionally equivalent. 
The fact that you cannot stream 848x480 is very concerning, and this is not the expected behavior. Could you share more information about this issue? (are you getting black frames? no frames? bad depth?)

weblackey, 2020-03-30:
Thanks for all the background info, we really appreciate it.

> The fact that you cannot stream 848x480 is very concerning, and this is not the expected 
behavior. 

Thanks, we thought the D435i is just a D435 with added IMU so we are surprised at this lower performance as well

> Could you share more information about this issue? (are you getting black frames? 

white screen, no frames for the stereo camera. RGB camera works fine. after moving the slider in realsense-viewer to stop streaming from the D435i, changing resolution to 640x480 with no other changes, and re-enabling streaming it then shows the stereo view. Maybe with more noise than the D435 in the exact same configuration but that's just a casual observation, not backed with any data

However this has been a stubborn issue, completely reproducible: working at 848 x 480 with D435, not working at 848 x 480 with D435i, and working at 640 x 480.

We're about to send this unit back to the Intel Realsense store. If someone from the support team would like to receive this specific unit, please send a shipping label and please arrange for credit for this unit at the realsense store. We can supply order number, serial number, etc

MartyG-RealSense, 2020-03-30:
@weblackey   The RealSense Store is a separate department from the RealSense GitHub support.  If you purchased your D435i camera within the last 30 days, details of how to return your camera can be found in the returns section at the bottom of the page at the link below.

https://store.intelrealsense.com/customer-service/

If you purchased the camera from a place other than the RealSense Store in the last 30 days, then it should be returned to the retailer that you got it from instead of Intel.

weblackey, 2020-03-30:
thanks, no problem returning this to the realsense store. we just wanted to mention that if this unit is showing unexpected behavior, we would be willing to ship it to realsense hardware engineering instead

MartyG-RealSense, 2020-03-30:
Intel do not offer a repair service for RealSense cameras.  The only option is a return for a refund.  There is also no means for the RealSense engineering team to handle the unit.  Thanks though!

weblackey, 2020-03-30:
ok for future reference it would be interesting to know if all D435i have lower performance than D435 or just this one. Having an IMU built in to the camera is nice as long as it is just an extra function without impacting the stereo camera when the IMU is not even being used.

closed 2020-03-30 by weblackey