realsenseai/librealsense #6166: T265 No device connected with C# Wrapper

RobAutomation, 2020-03-30:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265 } | 
| Firmware Version                   |  | 
| Operating System & Version |   {Win (10) 
| Platform                                 | PC.  |
| SDK Version                            |  (WIN10-2.33.1.1360)                         | 
| Language                            |  {C# }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
I just stated with the T265 to bring it to run. The Demo is running stable. But when I try the C# Wrapper Example on Visual Studio 2019 I can see the T265, but when I try to start a pipe the error is shown "No device connected"
ExternalException: rs2_pipeline_start_with_config(pipe:000002914DA30060, config:000002914DA30300)

The c++ example and also the phyton program is starting the pipe without error. 
I trid also old SDK Version until 2.31, but no differents.
Also I tried to install on a second PC, same result.

Thanks for help
Andreas

RendijsSmukulis, 2020-03-31:
for C#, I've found I need to leave an empty serial number: 

```
    public RsConfiguration DeviceConfiguration = new RsConfiguration
    {
        mode = RsConfiguration.Mode.Live,
        RequestedSerialNumber = string.Empty,
        Profiles = new[]
        {
            new RsVideoStreamRequest { Stream = Stream.Pose, StreamIndex = 0, Width = 0, Height = 0, Format = Format.SixDOF, Framerate = 0 },
            new RsVideoStreamRequest { Stream = Stream.Fisheye, StreamIndex = 1, Width = 0, Height = 0, Format = Format.Y8, Framerate = 0 },
            new RsVideoStreamRequest { Stream = Stream.Fisheye, StreamIndex = 2, Width = 0, Height = 0, Format = Format.Y8, Framerate = 0 }
        }
    };
```

If this doesn't help, you could post a sample of your code how you configure the pipeline.

RobAutomation, 2020-03-31:
I try the example from the wrapper itself and also a simple code like on the python program.
The code is running until pipe.Start(cfg). 
Your code looks like from Unity Example, or? I can't find the RequestedSerialNumber on my example.

Code Example:

            Context ctx = new Context();
            var list = ctx.QueryDevices(); // Get a snapshot of currently connected devices
            if (list.Count == 0)
                throw new Exception("No device detected. Is it plugged in?");
            dev = list[0];
            sensor = dev.Sensors[0];

            textBox1.Text += sensor.Info[CameraInfo.Name] + Environment.NewLine;
            try
            {
                var cfg = new Config();
                cfg.EnableStream(Stream.Pose);

                var pipe = new Pipeline();
                pipe.Start(cfg);
            }
            catch (Exception ex) { MessageBox.Show(ex.Message); }

RendijsSmukulis, 2020-03-31:
Ah, I've seen this issue before. For some reason, after querying for devices, I can no longer call `pipe.Start` as well. Using your code, I see the same issue locally. It goes away if I remove the ` ctx.QueryDevices()` line.

So does ensuring that I dispose of all the objects before calling `pipe.Start`:

```
            using (var ctx = new Context())
            {
                using (var list = ctx.QueryDevices())
                {
                    if (list.Count == 0)
                        throw new Exception("No device detected. Is it plugged in?");

                    using (var dev = list[0])
                    {
                        Console.WriteLine(dev);

                        using (var sensor = dev.Sensors[0])
                        {
                        }
                    }
                }
            }
```

RobAutomation, 2020-04-01:
Yes, thanks for your help. Now is working.
The code that was working for me.

```
            using (var ctx = new Context())
            {
                using (var list = ctx.QueryDevices())
                {
                    if (list.Count == 0)
                    {
                        lastError = "No device detected. Is it plugged in?";
                        throw new Exception(lastError);
                    }

                    using (var dev = list[0])
                    {
                        Console.WriteLine(dev);

                        using (var _sensor = dev.Sensors[0])
                        {
                            sensor = _sensor;
                        }
                    }
                }
            }

            pipe = new Pipeline();
            try
            {
                cfg = new Config();
                cfg.EnableStream(Stream.Pose);
                pipe.Start(cfg);

                threadGetData = new Thread(new ThreadStart(getData));
                threadGetData.Start();
            }
            catch (Exception ex) { lastError = ex.Message; }
```

closed 2020-04-01 by RobAutomation