realsenseai/librealsense #6167: Intermittent connectivity on Ubuntu 16.04 with recommended (Newnex) cabling

MitchAllain, 2020-03-30:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415  | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    | 4.15.0-58                                        | 
| Platform                                 | Cincoze DS-1200 IPC  |
| SDK Version                            |  2.25.0                          | 
| Language                            | realsense-ros                          | 
| Segment			| Robot                          | 
| **librealsense2-dkms** | 1.3.9 |

### Issue Description
We are experiencing intermittent loss of connectivity using one realsense D415 camera. We do have a particularly long run, but are using the recommended [12 m Newnex cable, UL-A01C12-12M](https://www.newnex.com/usb-3-1-type-c-active-cable-legacy.php). We have historically had intermittent disconnects with the realsense devices, maybe on the order of once per month with constant usage. Since switching to the D415 from a D435 for reasons unrelated, we have seen an uptick in device disconnects (currently as often as once every 5 minutes).

For the most part, the device seems to reconnect immediately, however at this high frequency the dropouts are not acceptable for our application. The issues have persisted through one firmware  upgrade, however the current firmware seems to follow the most recent recommendations [on the Intel website](https://dev.intelrealsense.com/docs/firmware-releases#section-recommended-production-configurations)

This excerpt from `dmesg` appears to show some uvcvideo module issues:
```
[  688.669966] usb 2-2: USB disconnect, device number 4
[  689.018015] usb 2-2: new SuperSpeed USB device number 5 using xhci_hcd
[  689.038392] usb 2-2: New USB device found, idVendor=8086, idProduct=0ad3
[  689.038395] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  689.038396] usb 2-2: Product: Intel(R) RealSense(TM) Depth Camera 415
[  689.038397] usb 2-2: Manufacturer: Intel(R) RealSense(TM) Depth Camera 415
[  689.038399] usb 2-2: SerialNumber: 948123021329
[  689.039558] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[  689.041200] uvcvideo: Unable to create debugfs 2-5 directory.
[  689.041235] uvcvideo 2-2:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[  689.041236] uvcvideo 2-2:1.0: Entity type for entity Processing 2 was not initialized!
[  689.041237] uvcvideo 2-2:1.0: Entity type for entity Camera 1 was not initialized!
[  689.041298] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/input/input15
[  689.041721] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[  689.042427] uvcvideo: Unable to create debugfs 2-5 directory.
[  689.042474] uvcvideo 2-2:1.3: Entity type for entity Processing 7 was not initialized!
[  689.042476] uvcvideo 2-2:1.3: Entity type for entity Extension 8 was not initialized!
[  689.042478] uvcvideo 2-2:1.3: Entity type for entity Camera 6 was not initialized!
[  689.746160] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[  689.796519] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[  689.847385] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[  690.027824] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[  690.079351] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[  690.187892] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[  690.287903] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[  690.681576] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
...
[ 1027.077781] usb 2-2: USB disconnect, device number 5
[ 1027.409858] usb 2-2: new SuperSpeed USB device number 6 using xhci_hcd
[ 1027.430178] usb 2-2: New USB device found, idVendor=8086, idProduct=0ad3
[ 1027.430181] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1027.430184] usb 2-2: Product: Intel(R) RealSense(TM) Depth Camera 415
[ 1027.430186] usb 2-2: Manufacturer: Intel(R) RealSense(TM) Depth Camera 415
[ 1027.430188] usb 2-2: SerialNumber: 948123021329
[ 1027.431358] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[ 1027.432801] uvcvideo: Unable to create debugfs 2-6 directory.
[ 1027.432848] uvcvideo 2-2:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[ 1027.432851] uvcvideo 2-2:1.0: Entity type for entity Processing 2 was not initialized!
[ 1027.432853] uvcvideo 2-2:1.0: Entity type for entity Camera 1 was not initialized!
[ 1027.432976] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/input/input16
[ 1027.433458] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[ 1027.434326] uvcvideo: Unable to create debugfs 2-6 directory.
[ 1027.434424] uvcvideo 2-2:1.3: Entity type for entity Processing 7 was not initialized!
[ 1027.434427] uvcvideo 2-2:1.3: Entity type for entity Extension 8 was not initialized!
[ 1027.434429] uvcvideo 2-2:1.3: Entity type for entity Camera 6 was not initialized!
[ 1030.353615] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 1030.403745] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 1030.453829] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 1030.504463] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 1030.554861] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 1030.741706] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 1030.791810] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 1030.841971] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 1030.892121] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 1030.942362] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 1030.994128] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 1031.104166] uvcvideo: Failed to set UVC probe control : -32 (exp. 48).
[ 1031.202995] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
...
```

Here is the output from `lsusb -t`
```
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/10p, 5000M
    |__ Port 2: Dev 56, If 0, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 2: Dev 56, If 1, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 2: Dev 56, If 2, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 2: Dev 56, If 3, Class=Video, Driver=uvcvideo, 5000M
    |__ Port 2: Dev 56, If 4, Class=Video, Driver=uvcvideo, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    |__ Port 5: Dev 2, If 0, Class=Hub, Driver=hub/2p, 480M
``` 

Let me know if I can provide any clarification and thank you for your support.

MartyG-RealSense, 2020-03-31:
You should be able to reliably use Newnex's Active Repeater cable with RealSense at that distance, as demonstrated in the video below.

Have you tried the D435 recently with your currently installed Librealsense setup to confirm whether it is only the D415 affected or whether the problem also now occurs with D435 too, please?

There have been a couple of other RealSense users who have reported this specific error in the past.

**uvcvideo: Unable to create debugfs 2-5 directory.**

One of the users solved the error by carrying out a purge of old files and doing a fresh install of the SDK from the repositories.  Please visit the discussion in the link below and read downward from the linked-to comment for more details about this.

https://github.com/IntelRealSense/librealsense/issues/3426#issuecomment-471589620
https://m.youtube.com/watch?v=GLQgR1jT04M

MartyG-RealSense, 2020-04-15:
This case will be closed after 7 days from the time of writing this message if there are no further comments.  Thanks!

MitchAllain, 2020-04-17:
We decided to build librealsense from source, which did eliminate the disconnects. We have ordered replacement hardware (camera and cables) and I will update here if either solves the issue.

MartyG-RealSense, 2020-04-17:
Thanks so much for the update!  I look forward to hearing the results of your new equipment.

MartyG-RealSense, 2020-05-11:
Could you update us please about whether ordering new equipment solved your disconnect problem please?  Thanks!

MitchAllain, 2020-05-14:
Hi Marty, it seems to have been a cable issue, I'll close for now.

closed 2020-05-14 by MitchAllain