realsenseai/librealsense #6169: D435i depth distance calibration details

dongkaizhou, 2020-03-31:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D435 } | 
| Firmware Version                   | (05.12.02.100) | 
| Operating System & Version  |   Ubuntu18.04 | 
| Kernel Version (Linux Only)     |  (e.g. 4.9.14)    | 
| Platform                                 | NVIDIA Jetson |
| SDK Version                            |  { 2.32.1 }    | 
| Language                            |  {C/opencv/pcl }   | 
| Segment			|  {Robot }     | 

### Issue Description
Hi,
In the project, we want to get the precise depth distance, need to sure the details.
Is the depth distance from object to camera lens? Or from object to camera lens chip? or others?
can you tell me, thanks.
the image for example: 
[image]

dongkaizhou, 2020-03-31:
Is the depth distance from object to camera lens? Or from object to camera chip? or others?

MartyG-RealSense, 2020-03-31:
Page 64 and 65 of the current edition of the data sheet document for the 400 Series cameras describes how the depth start point is calculated for the cased cameras and the caseless Depth Module Kit boards.

Depth origin coordinates are discussed in detail on pages 66 and 67.

https://dev.intelrealsense.com/docs/intel-realsense-d400-series-product-family-datasheet

dongkaizhou, 2020-03-31:
Thank you very much.
The other problem, the depth info`s center, is there two depth camera center? Or the camera center?

MartyG-RealSense, 2020-03-31:
You are very welcome.  I hope that the link below will be helpful in answering your other question.

https://github.com/IntelRealSense/librealsense/issues/2688#issuecomment-436612914

dongkaizhou, 2020-03-31:
soga, left camera is the reference camera and is located at world origin.  
Thank you, Marty.

MartyG-RealSense, 2020-04-15:
I will close this case now due to no further comments being received.  Thanks!

closed 2020-04-15 by MartyG-RealSense