realsenseai/librealsense #6174: Conflict with PyQT library

zapaishchykova, 2020-03-31:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Operating System & Version |   Win 10 |                             | 
| Platform                                 | PC  |
| Language                            |  python                      | 

### Issue Description
So my problem is that when librealsense.py ( ver.2) cannot be imported into the project with PyQT. I open my UI, and then just one simple pressing button stops the app completely. I didn't call any functions from libsense library, it's just simple:
` import pyrealsense2 as rs`
The more frustrating is that it doesn't show any error. Seems to be similar to the https://github.com/IntelRealSense/librealsense/issues/4856 , but i don't get what means "import localy".
Also tried to build wrapper from wheel/ installing with `pip install pyrealsense2`, neither worked. Also tried building from source with this https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md , but after building solution in VS I don't see any files in python wrapper folder so I gave up.
Any suggestions on how to tackle this issue will be much appreciated!

alaricLEA, 2020-04-02:
This sounds similar to the bug I reported. #6141. Importing the pyrealsense2 library (and not calling anything) causes issues with wxPython and Tkinter on Windows.

RealSenseCustomerSupport, 2020-04-16:
----------------------------------------------

@zapaishchykova What's the SDK version when you get the issue? And did you check if python examples in SDK package work normally?  Looking forward to your update. Thanks!

zapaishchykova, 2020-04-16:
Hi! Yes, python samples work just fine, it all happens when such imports are present alongside with librealsence import:
```
from PyQt5.QtWidgets import *
from PyQt5.QtGui import *
from PyQt5.QtCore import *
from PyQt5.QtMultimediaWidgets import *
from PyQt5.QtMultimedia import *
```
I am using SDK 2.0 but as far as I understand it is not used while I import the package in Python?

mths2601, 2020-04-27:
I faced a very similar issue and solved it by adding these lines before any other imports:
`import sys
sys.coinit_flags = 2
import pythoncom`

Now pyrealsense2 can be imported together with the PyQt5 components and opening further windows from the GUI works, too. @alaricLEA maybe you can find a useful hint for your issue, too.

This would be a minimal example which works with these additional lines, but breaks without them as soon as you try to open the file dialogue:
`import sys
sys.coinit_flags = 2
import pythoncom
import pyrealsense2
from PyQt5.QtWidgets import QApplication, QWidget, QPushButton, QFileDialog
from PyQt5.QtCore import pyqtSlot

class App(QWidget):

    def __init__(self):
        super().__init__()
        self.title = 'PyQt5 fail test'
        self.left = 10
        self.top = 10
        self.width = 320
        self.height = 200
        self.initUI()

    def initUI(self):
        self.setWindowTitle(self.title)
        self.setGeometry(self.left, self.top, self.width, self.height)

        button = QPushButton('PyQt5 button', self)
        button.move(100, 70)
        button.clicked.connect(self.on_click)

        self.show()

    @pyqtSlot()
    def on_click(self):
        print('Start function on_click')
        filename, _ = QFileDialog.getOpenFileName(self, "QFileDialog.getOpenFileName()", "",
                                                 "All Files (*);;Python Files (*.py)")
        if filename:
            print(filename)

        print('End function on_click')

if __name__ == '__main__':
    app = QApplication(sys.argv)
    ex = App()
    sys.exit(app.exec_())`

zapaishchykova, 2020-04-27:
That works! Awesome, Thanks!!

closed 2020-04-27 by zapaishchykova

alaricLEA, 2020-04-27:
Works for me too. Many thanks!

AkiraGiShinichi, 2021-05-25:
It works like a charm. Many thanks to @mareikethies !