realsenseai/librealsense #6175: 2.34.0 master does not compile on macOS Catalina

neilyoung, 2020-03-31:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | ? |
| Operating System & Version |  macOS Catalina |
| Kernel Version (Linux Only)    |      | 
| Platform                                 |   |
| SDK Version                            |   2.34.0, master                       | 
| Language                            |                     | 
| Segment			|  others                           | 

### Issue Description

Does not compile on macOS. Was not the case on last development branch, a couple of days ago when I tested it.

Might be related to the update of XCode on March 26th. Having not compiled definitely since then.

Remember to have seen an issue like this yesterday, commented that I would not have probs. Was wrong obviously :)

```
[ 40%] Building C object CMakeFiles/realsense2.dir/third-party/sqlite/sqlite3.c.o
/Users/decades/Documents/librealsense/third-party/sqlite/sqlite3.c:20294:17: warning: 'OSAtomicCompareAndSwapPtrBarrier' is deprecated: first
      deprecated in macOS 10.12 - Use atomic_compare_exchange_strong() from <stdatomic.h> instead [-Wdeprecated-declarations]
      success = OSAtomicCompareAndSwapPtrBarrier(NULL, newzone, 
                ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk/usr/include/libkern/OSAtomicDeprecated.h:547:6: note: 
      'OSAtomicCompareAndSwapPtrBarrier' has been explicitly marked deprecated here
bool    OSAtomicCompareAndSwapPtrBarrier( void *__oldValue, void *__newValue, void * volatile *__theValue );
        ^
1 warning generated.
[ 41%] Linking CXX shared library librealsense2.dylib
ld: can't write output file: librealsense2.2.34.0.dylib for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [librealsense2.2.34.0.dylib] Error 1
make[1]: *** [CMakeFiles/realsense2.dir/all] Error 2
make: *** [all] Error 2

```

dorodnic, 2020-03-31:
Hi @neilyoung 
I'll try to upgrade sqlite dependency, it should help.
I'm building with clang 9 on High Sierra, this is probably why we didn't catch it.

neilyoung, 2020-03-31:
OK, cool. Latest news from the development branch (which also not compiles anymore):

```
44 warnings generated.
[ 90%] Building C object tools/realsense-viewer/CMakeFiles/realsense-viewer.dir/__/__/third-party/glad/glad.c.o
/Users/decades/Documents/librealsense/third-party/glad/glad.c:1724:27: warning: no newline at end of file [-Wnewline-eof]
#pragma GCC diagnostic pop
                          ^
1 warning generated.
[ 91%] Building C object tools/realsense-viewer/CMakeFiles/realsense-viewer.dir/__/__/third-party/tinyfiledialogs/tinyfiledialogs.c.o
/Users/decades/Documents/librealsense/third-party/tinyfiledialogs/tinyfiledialogs.c:321:2: warning: embedding a directive within macro arguments has
      undefined behavior [-Wembedded-directive]
#endif
 ^
1 warning generated.
[ 91%] Building CXX object tools/realsense-viewer/CMakeFiles/realsense-viewer.dir/__/__/third-party/easyloggingpp/src/easylogging++.cc.o
error: unable to open output file 'CMakeFiles/realsense-viewer.dir/__/__/third-party/easyloggingpp/src/easylogging++.cc.o': 'Permission denied'
1 error generated.
make[2]: *** [tools/realsense-viewer/CMakeFiles/realsense-viewer.dir/__/__/third-party/easyloggingpp/src/easylogging++.cc.o] Error 1
make[1]: *** [tools/realsense-viewer/CMakeFiles/realsense-viewer.dir/all] Error 2
make: *** [all] Error 2
~/Documents/librealsense/build $ 

```

Pretty obscure. I will try a fresh clone tomorrow.

neilyoung, 2020-04-01:
OK, some news on this. It happened now the second time to me: A simple `git pull` and a fresh build did run into errors. I have now cloned the entire repo again and could compile w/o problems. Master. Development not tested yet. Very strange...

neilyoung, 2020-04-01:
One thing I noticed: The RealSense-Viewer has become pretty much unusable on macOS. It just hangs.

closed 2021-03-17 by RealSenseSupport