realsenseai/librealsense #6178: "Frame didn't arrived within 5000" error when using a preset (python, D415)

kogan-alex, 2020-04-01:
I'm working with the D415 camera in Python (Firmware version "05.12.02.100"). The following code crashes after several cycles of "connection -> preset -> disconnection" with the error "RuntimeError: Frame didn't arrived within 5000". Subsequently, the "Realsense viewer " also reports that it is unable to receive any depth frames. To recover, the camera has to be reset.

```
import pyrealsense2 as rs

while True:
    config = rs.config()
    config.enable_stream(rs.stream.depth, width=1280, height=720, format=rs.format.z16, framerate=30)
    config.enable_stream(rs.stream.color, width=1280, height=720, format=rs.format.bgr8, framerate=30)

    config.enable_device("#############")

    pipe = rs.pipeline()
    print("starting")
    profile = pipe.start(config)

    depth_sensor = profile.get_device().first_depth_sensor()
    depth_sensor.set_option(rs.option.visual_preset, 3)  # "High Accuracy" preset

    pipe.wait_for_frames()

    print("stopping")
    pipe.stop()

```
Adding a sleep period of 1 second between the preset and "wait for frames" reduces the probability of error significantly but it still crashes. Is there a better solution?

MartyG-RealSense, 2020-04-01:
It's good practice to skip the first several frames to give the auto-exposure time to settle down.  The link below has a Python script that demonstrates such code.

[image]

https://github.com/IntelRealSense/librealsense/issues/4095

kogan-alex, 2020-04-01:
It crashes even when I set the auto exposure to false.

MartyG-RealSense, 2020-04-01:
Is this problem specific to the High Accuracy preset or does it occur with any preset?

kogan-alex, 2020-04-01:
Just tried it with High Density, crashes as well.

MartyG-RealSense, 2020-04-01:
But it works okay with the "Default" preset?

kogan-alex, 2020-04-01:
No, crashes too.

kogan-alex, 2020-04-01:
If I comment the preset line it works fine.

MartyG-RealSense, 2020-04-01:
The line of Python code that you are using to change the preset is correct, I believe:

**depth_sensor.set_option(rs.option.visual_preset, 3)**

I recall that there were some past cases of crashes when changing the preset after the pipeline had been started.  This occurred in the RealSense Viewer as well as scripts.

What hardware and SDK version are you running Librealsense on please?

kogan-alex, 2020-04-01:
Do you mean Firmware version "05.12.02.100"?

MartyG-RealSense, 2020-04-01:
No it wasn't that recent a firmware version.  It was a while ago but I could not find an example link to such a case.

In the RealSense Viewer, does it work if you set the preset to High Accuracy before you enable streaming?

kogan-alex, 2020-04-01:
Yes, it works (I tried it manually a few times).

MartyG-RealSense, 2020-04-01:
BTW, it is worth mentioning that if you are using a serial number (indicated by the #######), identifying by serial number is important in multi-device setups.  But if you just have a single camera, you shouldn't need to use a serial number.

kogan-alex, 2020-04-01:
Yes, I know. We will be using 4, connected to a USB hub. Thanks.

MartyG-RealSense, 2020-04-01:
I finally tracked down a preset-changing case.  In the case in the link below, the RealSense user concluded that the problem was related to the build of Windows that they were using.  Are you running Python on a Windows computer please?

https://github.com/IntelRealSense/librealsense/issues/4571

kogan-alex, 2020-04-01:
Linux

MartyG-RealSense, 2020-04-01:
I located a similar crash case for Ubuntu where a Python program crashed if depth sensor options were set after the pipeline started, but the program did not crash if they were set before the pipeline was started.

https://github.com/IntelRealSense/librealsense/issues/4940

kogan-alex, 2020-04-01:
Looks promising however, I don't understand how you can get a "profile" before calling "start"?

MartyG-RealSense, 2020-04-01:
I was not certain how they achieved that either.  I am still learning advanced Librealsense programming, so I apologise for any awkwardness in my answers.   :)

RealSense team member @ev-mp knows Python well, I believe, so I will tag them into this discussion to seek advice.  I'm sorry for the delay that this may cause to receiving a solution in the meantime.

kogan-alex, 2020-04-19:
Can somebody help in this issue?

MartyG-RealSense, 2020-04-19:
There was a case this week of a Python user who was experiencing problems with receiving a depth stream if using a resolution greater than 640x480.  It was only affecting one particular camera out of a batch of several.

https://github.com/IntelRealSense/librealsense/issues/6262

After eliminating various possibilities for the cause of the problem, it was noticed that with this particular camera, their CPU usage rate was running high.

Would it be possible to check your own CPU usage please and see if it is a high value, please?

kogan-alex, 2020-04-21:
Nothing out of the ordinary with CPU usage. Happens with 3 different cameras on two different computers. Specifically happens when set preset is used.

MartyG-RealSense, 2020-04-21:
The current latest firmware 5.12.3.0 added a fix for the camera not being detected after a "soft reboot" (restarting the computer instead of shutting it down and then restarting with the power button).  I see from the information at the top of this case that you are using the firmware before that current version, 5.12.2.100.

[image]

kogan-alex, 2020-04-22:
I tried to update the firmware of the camera as per your recommendation, but the "update firmware" option in the Realsense viewer is missing. 

[image]

MartyG-RealSense, 2020-04-22:
The firmware updater was not yet in the RealSense Viewer in version 2.20.0 (which dates back to April 2019).  The SDK version is now up to 2.**34**.0.  The ability to introduce firmware in the Viewer was firstintroduced in SDK version 2.24.0 in June 2019.

kogan-alex, 2020-04-22:
How do I upgrade it (Linux)?

I followed the instructions in:
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

And the viewer remained in version 2.20.0.

MartyG-RealSense, 2020-04-22:
Please check in the **Librealsense > Tools** folder for the Viewer.

kogan-alex, 2020-04-30:
I ran the make file in the directory and ran RealSense Viewer from this directory, the version is still 2.20.0

MartyG-RealSense, 2020-04-30:
If you are having difficulties with the usual SDK build process and your computer has an internet connection then there is a "bypass" method of installation that does not rely on Linux versions, kernel versions or patches.

From the Librealsense **Build** directory (Librealsense > Build), run this CMake installation instruction:

**cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true**

kogan-alex, 2020-04-30:
Still nothing changed. Should I do an additional step?
This is the terminal output:

```
~/Realsense/librealsense/build$ cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true
-- Info: REALSENSE_VERSION_STRING=2.20.0
-- Setting Unix configurations
-- Checking internet connection...
-- Internet connection identified, enabling BUILD_WITH_TM2
-- Using X11 for window creation
-- Building with TM2
-- ----------------------------------------------------------------------------
-- T265 Product versions:
-- - HOST 0.19.3.1711 (Default from versions.cmake)
-- - Remote FW 0.0.18.5448 (Default from versions.cmake)
-- - Remote CENTRAL APP 2.0.19.271 (Default from versions.cmake)
-- - Remote CENTRAL BL 1.0.1.112 (Default from versions.cmake)
-- --------------------------------------------------------------------------------------------------------------------------------------------------------------
-- - Downloading FW 0.0.18.5448 from http://realsense-hw-public.s3.amazonaws.com/Releases/TM2/FW/target/0.0.18.5448/target-0.0.18.5448.mvcmd
--   Converting FW version 0.0.18.5448 from target.mvcmd to /home/csr_user/Realsense/librealsense/third-party/libtm/libtm/src/fw.h
--   File /home/csr_user/Realsense/librealsense/third-party/libtm/libtm/src/fw.h wasn't changed
-- - Downloading Central App 2.0.19.271 from http://realsense-hw-public.s3.amazonaws.com/Releases/TM2/FW/app/2.0.19.271/central_app-2.0.19.271.bin
--   Converting Central App version 2.0.19.271 from central_app.bin to /home/csr_user/Realsense/librealsense/third-party/libtm/libtm/src/CentralAppFw.h
--   File /home/csr_user/Realsense/librealsense/third-party/libtm/libtm/src/CentralAppFw.h wasn't changed
-- - Downloading Central BL 1.0.1.112 from http://realsense-hw-public.s3.amazonaws.com/Releases/TM2/FW/bl/1.0.1.112/central_bl-1.0.1.112.bin
--   Converting Central BL version 1.0.1.112 from central_bl.bin to /home/csr_user/Realsense/librealsense/third-party/libtm/libtm/src/CentralBlFw.h
--   File /home/csr_user/Realsense/librealsense/third-party/libtm/libtm/src/CentralBlFw.h wasn't changed
-- --------------------------------------------------------------------------------------------------------------------------------------------------------------
-- Building libtm project on , LIBTM version [0.19.3.1711], API version [10.0], branch [master], FW [0.0.18.5448], Central APP [2.0.19.271], Central BL [1.0.1.112]
-- Creating version file /home/csr_user/Realsense/librealsense/third-party/libtm/libtm/src/Version.h
-- Building project tm as STATIC library lib
-- --------------------------------------------------------------------------------------------------------------------------------------------------------------
-- Building all projects of libtm_samples
-- ----------------------------------------------------------------------------
-- CMake Done
-- ----------------------------------------------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/csr_user/Realsense/librealsense/build
```

MartyG-RealSense, 2020-04-30:
There shouldn't be further installation steps after successful completion of the bypass installation method.  The SDK should now have been built.

I do not know why the SDK keeps insisting on installing 2.20.0.  I will seek advice from Intel about it.

RealSenseSupport, 2020-05-01:
Please try to checkout the latest update to the local repo or you can also pick and download the latest release (source) from [https://github.com/IntelRealSense/librealsense/releases](url). 
The include\librealsense2\rs.h should show the correct version.

kogan-alex, 2020-05-04:
Thank you, the installation problem was solved (apparently there was a couple of realsense-viewers installed).

I updated the camera firmware to 05.12.03.00, however the problem persists.

MartyG-RealSense, 2020-05-11:
Could you update us about whether you are still experiencing problems please?  Thanks!

kogan-alex, 2020-05-11:
Yes, as I have mentioned, the problem is still there.

MartyG-RealSense, 2020-05-11:
@kogan-alex  I carefully considered the problem again with the preset causing a crash, but working if the preset-setting line is commented out.   I also re-reviewed the other past cases of this problem, though most of them conclude without a solution and RealSense team members are unable to replicate the problem.

At this point it seems to be an issue that affects a very small minority of people, as there are only a handful of reported cases.  I appreciate that this does not make it any less frustrating for you.  At this point it is probably not going to be fixed, as if it cannot be replicated by RealSense developers then it is hard to develop a solution for it.  So the best advice I can give is to comment out the preset changing line and try to find other ways to enhance your RealSense image quality.

At this point I will close the case, as a solution for the minority of RealSense users who experience this issue is not likely to be forthcoming in the foreseeable future.  I do apologise.

closed 2020-05-11 by MartyG-RealSense

kogan-alex, 2020-05-13:
I see, this is unfortunate.

Non the less, maybe you can help me with a workaround. I was thinking about checking the preset value before configuring it, and not setting it in case it is correct (I've seen that camera configuration only changes after a power-cycle, if no one else changed it). This will make the problem disappear. However, I've noticed that when I re-start the camera, the returned preset value is always "0" ("custom") even though the values are corresponding to the chosen preset, why is this so? Is it a bug?

MartyG-RealSense, 2020-05-13:
The same behavior is seen on the RealSense Viewer program, where the preset is always set to Custom when you start the program no matter which preset you selected or loaded in on the last session.

Someone reckoned that the default presets come from this SDK file:

https://github.com/IntelRealSense/librealsense/blob/master/src/ds5/advanced_mode/presets.cpp

If you cannot load the High Accuracy preset file, it made me wonder if a workaround might be to copy and paste the High Accuracy json's settings into this presets.cpp file in place of the default settings so that settings equivalent to High Accuracy load by default.