realsenseai/librealsense #6179: More of a question than an issue. I am new to building using CMake, how does installing the library using "pip install pyrealsense2" differ from building the library using CMake? Are they two different tasks with different goals or two different ways to accomplish the same task?

asb2111991, 2020-04-01:


asb2111991, 2020-04-01:
I am working on Windows 10 and Anaconda.

dorodnic, 2020-04-01:
Hi @asb2111991 
Building from source is how everything is done. If you follow the steps you can produce the same binary as the ones built by our build-machines, that we then push to pypi repository. 
The goal of various binary packages (pip, debian, maven, npm, and more) is to save you time.

asb2111991, 2020-04-01:
Thank you @dorodnic for your response. So, this means, it doesn't make a difference whether I build from source or use pip to install the package right?

RealSenseCustomerSupport, 2020-04-13:
----------------------------------------------

The pypi package provides users the convenience and is also produced and built from the same source.

Please note that this issue will be closed if we don’t hear from you in 7 days**, **

closed 2020-04-20 by RealSenseCustomerSupport