realsenseai/librealsense #6181: Need help in using multi cameras

Tom127, 2020-04-02:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Linux (Ubuntu 16)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  pyrealsense2                          | 
| Language                            |  python                         | 
| Segment			|  {Robot }                          | 

### Issue Description
@MartyG-RealSense @dorodnic 
Hi, I'm using two D435 cameras in my PC, I have two situations now:
1. using number1.py read image from camera1, number2.py read image from camera2.
2. code1 read images from camera1, and code2 read images from camera2 in one .py file.
I don't know how to bind .py file with camera, so that number1.py won't turn on camera2, or  code1 won't turn on camera2. How should I do?

ps.  I can already read one camera with my computer by [align-depth2color.py](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py)

MartyG-RealSense, 2020-04-02:
It sounds as though what you need to do is access cameras by their specific serial number in your '1' and '2' scripts, to make sure that each script only talks to a specific camera.  The link below has a Python code example for such a setup.

https://stackoverflow.com/questions/50724909/pyrealsense2-librealsense-sdk-2-0-choose-cam-from-serial-number

Tom127, 2020-04-03:
> It sounds as though what you need to do is access cameras by their specific serial number in your '1' and '2' scripts, to make sure that each script only talks to a specific camera. The link below has an example of Pyrealsense2 code for such a setup.
> 
> [#1735 (comment)](https://github.com/IntelRealSense/librealsense/issues/1735#issuecomment-390840337)

@MartyG-RealSense 

That's exactly what I want. Thank you so much!

closed 2020-04-03 by Tom127