realsenseai/librealsense #6182: cannot compile thread api how-to example

dsvua, 2020-04-02:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I am trying to compile "Do Processing on a Background-Thread" api how to example and it has this line:
```
rs2::depth_frame frame;
```
when, I am trying to compile, I am getting this error:
```
record_camera.cpp: In lambda function:
record_camera.cpp:45:25: error: no matching function for call to ‘rs2::depth_frame::depth_frame()’
             rs2::depth_frame frame;
                         ^~~~~
In file included from /usr/local/include/librealsense2/hpp/rs_sensor.hpp:8:0,
                 from /usr/local/include/librealsense2/hpp/rs_device.hpp:8,
                 from /usr/local/include/librealsense2/hpp/rs_record_playback.hpp:8,
                 from /usr/local/include/librealsense2/hpp/rs_context.hpp:8,
                 from /usr/local/include/librealsense2/rs.hpp:9,
                 from record_camera.cpp:1:
/usr/local/include/librealsense2/hpp/rs_frame.hpp:793:9: note: candidate: rs2::depth_frame::depth_frame(const rs2::frame&)
         depth_frame(const frame& f)
         ^~~~~~~~~~~
/usr/local/include/librealsense2/hpp/rs_frame.hpp:793:9: note:   candidate expects 1 argument, 0 provided
/usr/local/include/librealsense2/hpp/rs_frame.hpp:786:11: note: candidate: rs2::depth_frame::depth_frame(const rs2::depth_frame&)
     class depth_frame : public video_frame
           ^~~~~~~~~~~
/usr/local/include/librealsense2/hpp/rs_frame.hpp:786:11: note:   candidate expects 1 argument, 0 provided
/usr/local/include/librealsense2/hpp/rs_frame.hpp:786:11: note: candidate: rs2::depth_frame::depth_frame(rs2::depth_frame&&)
/usr/local/include/librealsense2/hpp/rs_frame.hpp:786:11: note:   candidate expects 1 argument, 0 provided
```

dorodnic, 2020-04-02:
Hi @dsvua 
`depth_frame frame = frame{};` should work

dsvua, 2020-04-02:
Still issue:
```
record_camera.cpp: In lambda function:
record_camera.cpp:45:38: error: expected ‘,’ or ‘;’ before ‘{’ token
             depth_frame frame = frame{};
                                      ^
record_camera.cpp: In lambda function:
record_camera.cpp:60:38: error: expected ‘,’ or ‘;’ before ‘{’ token
             video_frame frame = frame{};
                                      ^
record_camera.cpp: In lambda function:
record_camera.cpp:75:38: error: expected ‘,’ or ‘;’ before ‘{’ token
             video_frame frame = frame{};
```

dsvua, 2020-04-02:
seems to compile when doing
```
depth_frame frame = frame;
```
I am not sure if it is correct

dsvua, 2020-04-02:
Tested,
```
video_frame = frame;
```
Works, but
```
depth_frame = frame;
```
gives `Segmentation fault`

dsvua, 2020-04-02:
Found the problem. I am `using namespace rs2` and it gets confused with word frame, changing it works, so @dorodnic is correct, this is working:
```
depth_frame tframe = frame{};
```

closed 2020-04-02 by dsvua