realsenseai/librealsense #6183: Camera D435 not detected while launching realsense-viewer on Ubuntu 16.04 LTS

Christopher-anton, 2020-04-02:
| | |
|---|---|
|**librealsense**|2.33.1 RELEASE|
|**OS**|Linux|

Please provide a description of the problem

MartyG-RealSense, 2020-04-02:
Could you tell us please which hardware you are using to run realsense-viewer on (PC, Mac, Raspberry Pi, Jetson, etc.  Thanks!

dorodnic, 2020-04-02:
Also, output from `lsusb` and `dmesg`

MartyG-RealSense, 2020-04-15:
This case will be closed after 7 days from the time of writing this message if there are no further comments.  Thanks!

Christopher-anton, 2020-04-15:
Hello there,

After some recent updates it seems the realsense viewer is detecting the
camera when plugged in. But the framerate is still choppy. If I move the
camera too much, the software crashes because of too much data inrush.
Could this be dependent on the hardware on my PC ?

MartyG-RealSense, 2020-04-15:
The 'Vision Processor D4' hardware inside the camera can help the camera to run well on a low-specification computing device.  If a high resolution, more than one stream and a fast 'frames per second' speed is used though then having a good computer can improve performance.

Christopher-anton, 2020-04-15:
Oh okay, I understand now.

Thanks so much for reaching out!

MartyG-RealSense, 2020-04-15:
You are very welcome.  Good luck with your camera!

closed 2020-04-26 by MartyG-RealSense