realsenseai/librealsense #6189: Error messages after updating from 2.30 to 2.34 on MacOS

dyoo-dev, 2020-04-02:
| Required Info	 |  |
|---|---|
| Camera Model | D435 |
| Firmware Version |  05.12.03.00 |
| Operating System & Version | macOS Mojave | 
| SDK Version	| 2.34.0, master |

### Issue Description
After update SDK from 2.30 to 2.34, when I run realsense-viewer, I have a bunch of the following error messages and the visualization has severe latency of 1 or 2 secs. 
Please give me any suggestions to fix this. 
Thanks in advance. 

```
02/04 10:49:01,010 ERROR [0x700004519000] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 02/04 10:49:01,033 ERROR [0x70000482b000] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 02/04 10:49:01,099 ERROR [0x70000482b000] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 02/04 10:49:01,133 WARNING [0x70000461f000] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Operation timed out, number: 60
 02/04 10:49:01,209 ERROR [0x700004519000] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 02/04 10:49:01,232 ERROR [0x70000482b000] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 02/04 10:49:01,268 WARNING [0x70000461f000] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Operation timed out, number: 60
 02/04 10:49:01,341 ERROR [0x70000482b000] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 02/04 10:49:01,352 WARNING [0x70000461f000] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Operation timed out, number: 60
 02/04 10:49:01,426 ERROR [0x700004519000] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 02/04 10:49:01,437 WARNING [0x70000461f000] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Operation timed out, number: 60
 02/04 10:49:01,511 ERROR [0x70000482b000] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 02/04 10:49:01,564 ERROR [0x700004519000] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 02/04 10:49:01,586 ERROR [0x70000482b000] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 02/04 10:49:01,630 ERROR [0x70000482b000] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
```

MartyG-RealSense, 2020-04-02:
I upgraded to 2.34.0 myself and are experiencing the same error with the RealSense Viewer on a Windows NUC PC with firmware version 5.12.03.00 and a RealSense D435i camera.  The same error occurred on both USB2 and USB 3 connections.

It looks as though the error comes from a file called **global_timestamp_reader.cpp** and a fix is in the works to stop **get_device_time_ms** from being called.

https://github.com/IntelRealSense/librealsense/pull/6150

MartyG-RealSense, 2020-04-26:
Closed due to no further comments.

closed 2020-04-26 by MartyG-RealSense