realsenseai/librealsense #6190: 2.34.0 RealSense Viewer Won't Open (Windows 10)

GOBish, 2020-04-02:
I installed the new build 2.34.0 on Windows 10 and now RealSense Viewer will not open. Screen immediately goes white and crashes.

Previous builds were working fine.

MartyG-RealSense, 2020-04-03:
When diagnosing launch or freeze problems with the RealSense Viewer, the first recommended step is to try updating the graphics drivers on the computer.  This is because the Viewer is dependent on OpenGL graphics, and support for this is typically part of the drivers for the graphics GPU of the computer.

MartyG-RealSense, 2020-04-15:
This case will be closed after 7 days from the time of writing this message if no further comments are made.  Thanks!

iamvukasin, 2020-04-16:
I have a similar problem on Windows 10: the viewer's window isn't even shown. I have uninstalled the SDK and updated the graphics drivers and still have issues.

MartyG-RealSense, 2020-04-16:
@iamvukasin   How did you install the SDK, please?   Did you build it from source or use the pre-built binary?  (the 750 mb **Intel.RealSense.SDK-WIN10-2.34.0.1448.exe** file).

iamvukasin, 2020-04-16:
I used the pre-build binary Intel.RealSense.SDK-WIN10-2.34.0.1448.exe.

MartyG-RealSense, 2020-04-16:
Another cause of Viewer launch problems other than the graphics driver may be a problem with the user documents folder path.  

https://github.com/IntelRealSense/librealsense/issues/4569#issuecomment-526521574

If your user documents folder was called iamvusakin, then you might want to make sure that your user folder path looks like this:

C:\Users\iamvusakin\Documents

I.e make sure that you have a folder called **Documents** inside the root of your user folder.

iamvukasin, 2020-04-16:
I have seen #4569 previously and I have just double-checked that the Documents folder exists. Still have the problem.

BTW, I have a dash in my username. Can it cause the issue?

MartyG-RealSense, 2020-04-16:
I'm not aware of a dash in usernames causing a problem.

Does the Depth Quality Tool have the same non-launch problem please?  They have similar programming, so if the Viewer has a launch problem then the Depth Quality Tool often does too.

iamvukasin, 2020-04-16:
Same problem with the Depth Quality Tool. I tried both standard opening and running the `rs-depth-quality.exe` via Command Prompt, and it stops immediately after.

MartyG-RealSense, 2020-04-16:
Thank you very much for your patience during this diagnostic process.  Are you able to run other SDK programs?   If you go to **C: > Program Files (x86) > Intel RealSense SDK 2.0 > Tools** then there are pre-built versions of SDK example programs that can be run to test the camera with.

iamvukasin, 2020-04-16:
It produces:

```
Could not open OpenGL window, please check your graphic drivers or use the textual SDK tools
```

MartyG-RealSense, 2020-04-16:
It would seem to indicate that the problem is still with the graphics drivers despite your earlier update of them.  Some of the SDK programs such as the Viewer and Depth Quality Tool, and example programs that also use OpenGL, need that driver support to run.

Could you tell me what kind of graphics chip your computer has please?  (Intel, Nvidia, AMD, etc)

iamvukasin, 2020-04-16:
I'm using Nvidia GeForce RTX 2080 Ti running 445.87 version of the driver.

MartyG-RealSense, 2020-04-16:
Nvidia GeForce drivers fully support OpenGL.  So updating the driver should update OpenGL support too.

Research of the error "Could not open OpenGL window, please check your graphic drivers or use the textual SDK tools" suggests that it is something more likely to occur on PCs that are running Ubuntu Linux in Windows than with someone who has directly installed the pre-made Windows SDK onto Windows.

If SDK 2.33.1 and its Viewer worked before for you then you should be able to go back to it easily by downloading and installing the Windows installer for that version.  I actually did it myself earlier today during a different case and Viewer 2.33.1 installed in place of the 2.34.0 one without problems.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.33.1

GOBish, 2020-04-16:
My Depth Quality Tool works, but Viewer crashes immediately (doesn't even open). Updated all graphics drivers.

MartyG-RealSense, 2020-04-16:
Is this Viewer crash on SDK version 2.33.1, please?

GOBish, 2020-04-16:
2.34.0

MartyG-RealSense, 2020-04-16:
Ok.  Were you able to try installing SDK version 2.33.1 as suggested above?

GOBish, 2020-04-16:
Yes 2.33.1 runs fine. Looking to use 2.34 to take advantage of openVino support.

MartyG-RealSense, 2020-04-16:
I installed the full 2.33.1 SDK and then I downloaded the 2.34.0 Viewer individually.  

[image]

The Viewer launched but still has the timestamp message problems in its log that are currently affecting it.  If you download it in this way to your Windows user 'Downloads' folder then you need to launch it from there, as the desktop shortcut will still point to the 2.33.1 version of the Viewer.

[image]

GOBish, 2020-04-16:
OK, that opened the viewer.
But -- OpenVino support not installed, that is part of the 2.34 installation, not just the viewer.

MartyG-RealSense, 2020-04-16:
What if you install the full 2.34.0 SDK and then try running the version of the Viewer in the Downloads folder instead of using the desktop shortcut?

GOBish, 2020-04-16:
still opens, but no openVino support

MartyG-RealSense, 2020-04-16:
Oh sorry, the release notes for SDK 2.34.0 say that the Viewer OpenVINO support requires the full Windows SDK installer and it is not in the standalone version.

[image]

iamvukasin, 2020-04-21:
Installing an older version did work for me. Thanks for the support!

MartyG-RealSense, 2020-04-21:
@iamvukasin   I'm very glad that your own Viewer problem was fixed by using an older version.  Thanks for the update!

dorodnic, 2020-04-23:
It seems that OpenVINO dependency is causing the issue. 
@iamvukasin - can you please share the CPU model you are using?

iamvukasin, 2020-04-23:
I'm using Intel i7 5820K.

MartyG-RealSense, 2020-04-26:
@iamvukasin   I will close this case now as the problem that you had was solved.  Thanks so much!

closed 2020-04-26 by MartyG-RealSense