realsenseai/librealsense #6191: Problem with alternating on off laser on Jetson.

dsvua, 2020-04-03:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | Latest | 
| Operating System & Version |  Linux (Ubuntu 14/16/17)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | NVIDIA Jetson |
| SDK Version                            |  2.34                         | 
| Language                            |  C++                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I have this code to alternate laser and save grayscale IR images when laser is off and depth when laser is on. It works perfectly on my Ubuntu 18.04 with patched kernel, but gives me images with laser pattern when with librealsense compiled on Jetson Nano with `-DFORCE_LIBUVC=true` flag.

Again, code works fine on Ubuntu 18.04 with patched kernel and when on Jetson Nano with UVC it has laser pattern on all images. Librealsense is compiled from current master branch on both systems.

This is the code:
```
int main(int argc, char * argv[]) {
    int width = 848;
    int height = 480;
    int fps = 15;

    cv::String images_path(argv[1]);

    rs2::config config;
    config.enable_stream(RS2_STREAM_INFRARED, 1, width, height, RS2_FORMAT_Y8, fps);
    config.enable_stream(RS2_STREAM_INFRARED, 2, width, height, RS2_FORMAT_Y8, fps);
    config.enable_stream(RS2_STREAM_DEPTH, width, height, RS2_FORMAT_Z16, fps);

    // start pipeline
    rs2::pipeline pipeline;
    rs2::pipeline_profile pipeline_profile = pipeline.start(config);

    rs2::device selected_device = pipeline_profile.get_device();
    auto depth_sensor = selected_device.first<rs2::depth_sensor>();
    depth_sensor.set_option(RS2_OPTION_EMITTER_ON_OFF, 1);

    int emitter_mode, frame_count;

    const auto CAPACITY = 5; // allow max latency of 5 frames
    rs2::frame_queue depth_queue(CAPACITY);
    rs2::frame_queue left_ir_queue(CAPACITY);
    rs2::frame_queue right_ir_queue(CAPACITY);

    std::thread depth_thread([&]() {
        while (true)
        {
            depth_frame tframe = frame{};
            if (depth_queue.poll_for_frame(&tframe))
            {
                // tframe.get_data();
                cv::Mat image = cv::Mat(cv::Size(width, height), CV_16SC1, (void*)tframe.get_data());
                int frame_counter = tframe.get_frame_metadata(RS2_FRAME_METADATA_FRAME_COUNTER);
                cout << images_path + cv::format("depth_%05u.png", frame_counter) << endl;
                cv::imwrite(images_path + cv::format("depth_%05u.png", frame_counter), image);
            }
        }
    });
    depth_thread.detach();

    std::thread left_ir_thread([&]() {
        while (true)
        {
            video_frame tframe = frame{};
            if (left_ir_queue.poll_for_frame(&tframe))
            {
                // tframe.get_data();
                cv::Mat image = cv::Mat(cv::Size(width, height), CV_8UC1, (void*)tframe.get_data());
                int frame_counter = tframe.get_frame_metadata(RS2_FRAME_METADATA_FRAME_COUNTER);
                cout << images_path + cv::format("left_ir_%05u.png", frame_counter) << endl;
                cv::imwrite(images_path + cv::format("left_ir_%05u.png", frame_counter), image);
            }
        }
    });
    left_ir_thread.detach();

    std::thread right_ir_thread([&]() {
        while (true)
        {
            video_frame tframe = frame{};
            if (right_ir_queue.poll_for_frame(&tframe))
            {
                // tframe.get_data();
                cv::Mat image = cv::Mat(cv::Size(width, height), CV_8UC1, (void*)tframe.get_data());
                int frame_counter = tframe.get_frame_metadata(RS2_FRAME_METADATA_FRAME_COUNTER);
                cout << images_path + cv::format("right_ir_%05u.png", frame_counter) << endl;
                cv::imwrite(images_path + cv::format("right_ir_%05u.png", frame_counter), image);
            }
        }
    });
    right_ir_thread.detach();

    // Capture 30 frames to give autoexposure, etc. a chance to settle
    for (auto i = 0; i < 30; ++i) pipeline.wait_for_frames();
        
    while (1){

        // wait for frames and get frameset
        rs2::frameset frameset = pipeline.wait_for_frames();

        // get single infrared frame from frameset
        rs2::video_frame depth_frame = frameset.get_depth_frame();
        emitter_mode = depth_frame.get_frame_metadata(RS2_FRAME_METADATA_FRAME_EMITTER_MODE);
        frame_count = depth_frame.get_frame_metadata(RS2_FRAME_METADATA_FRAME_COUNTER);
        cout << "Emitter mode: " << emitter_mode << "frame count: " << frame_count << endl;

        if (emitter_mode) {
            depth_queue.enqueue(frameset.get_depth_frame());
        } else {
            left_ir_queue.enqueue(frameset.get_infrared_frame(1));
            right_ir_queue.enqueue(frameset.get_infrared_frame(2));            
        }
    }

    return EXIT_SUCCESS;
}
```

Any ideas why?

dorodnic, 2020-04-03:
Just a theory, but are you building from source on the Jetson? If yes, please make sure you build with `-DCMAKE_BUILD_TYPE=Release`

dsvua, 2020-04-03:
This is the cmake command I used: `cmake ../ -DFORCE_LIBUVC=true -DCMAKE_BUILD_TYPE=release -DBUILD_WITH_CUDA=true`

dsvua, 2020-04-03:
Also, I installed librealsense 2.32 from packages as well and it has the same behavior. So, I suspect laser alternating does not work with UVC and needs patched kernel. However, in metadata, it reports that laser is alternating on off when actual IR image has laser pattern. Very strange behavior.

dsvua, 2020-04-03:
I did more tests and it looks like `RS2_FRAME_METADATA_FRAME_EMITTER_MODE` is inverted on UVC setups in comparison to kernel patch. 

When I changed my code to:
```
if (!emitter_mode) {
```
I got images without laser pattern on Jetson when `RS2_FRAME_METADATA_FRAME_EMITTER_MODE == 1` when on desktop with patched kernel I am getting images without laser pattern when `RS2_FRAME_METADATA_FRAME_EMITTER_MODE == 0`

dsvua, 2020-04-03:
@dorodnic Can you look at UVC code and fix `RS2_FRAME_METADATA_FRAME_EMITTER_MODE` to be set correctly? As it is inverted right now.

dorodnic, 2020-04-05:
@dsvua 
There is no difference between V4L2+Patch and libuvc handling of metadata. 
However, there is a known issue in the firmware that emitter mode sometimes is shifted by one. It is being addressed by the firmware group and will be published in one of future releases.

dsvua, 2020-04-05:
Interesting, it is the same camera with the same firmware that behaves differently when connected to different computers. Are you sure it is firmware issue?

RealSenseCustomerSupport, 2020-04-09:
----------------------------------------------

@dsvua What's the firmware version of your camera when get the issue? I tested D435i with RealSenseViewer v2.34 and FW v5.12.3 on Jetson Nano with Ubunt18.04 and kernel 4.9.140, but couldn't reproduce the issue. I can see the laser on/off works well.

dsvua, 2020-04-10:
Firmware version: 05.12.03.00 and I did not check it with realsense_viewer only in my program (code is in first post) as I do not have GUI (X-Windows) on headless Jetson Nano.

dsvua, 2020-04-10:
@RealSenseCustomerSupport  Also, I am not sure you will see any issue visually as laser is correctly toggled on_off, it is metadata that is inconsistent and inverted (it shows laser is on when it is actually off and vice versa) on Jetson Nano.

dsvua, 2020-04-11:
Renamed issue to better explain actual problem.

RealSenseCustomerSupport, 2020-04-15:
----------------------------------------------

@dsvua I tested with RealSenseViewer v2.34 on my Jetson Nano but couldn't see the issue of "metadata that is inconsistent and inverted". I'll attached two snapshots for your reference. Did you see the issue with RealSenseViewer or your own code?

Attachment(s):
PowerMode0.PNG - https://realsensesupport.intel.com/attachments/token/PoykqB3W44kzcAeRW5xhmrw76/?name=PowerMode0.PNG
PowerMode1.PNG - https://realsensesupport.intel.com/attachments/token/mY8osQlCEB8r6J2ouGJEOSFAo/?name=PowerMode1.PNG

RealSenseCustomerSupport, 2020-04-21:
----------------------------------------------

@dsvua Today I could reproduce the issue for the first connection with emitter on-off feature enabled before streaming. And as doronic mentioned previously, this will be addressed by the firmware group and will be published in one of future releases.

###

dsvua, 2020-05-04:
@RealSenseCustomerSupport Thank you for checking it. Feel free to ping me when it is resolved.

RealSenseCustomerSupport, 2020-06-03:
----------------------------------------------

@dsvua The new firmware 5.12.5 with fix is ready now. Please download from the link below to have a try. Thanks!

https://www.intelrealsense.com/download/11232/?_ga=2.208652242.1744298130.1590975453-69693576.1578898374

RealSenseCustomerSupport, 2020-06-22:
----------------------------------------------

@dsvua Did you get chance to try the new firmware 5.12.5 to see if the issue fixed? Looking forward to your update. Thanks!

closed 2020-07-06 by dorodnic

haowu-figure, 2023-02-21:
@RealSenseCustomerSupport  

I'm using FW 5.14.00. Still experiencing this issue, where `get_frame_metadata(RS2_FRAME_METADATA_FRAME_EMITTER_MODE) == 0` gives me emmiter on while `1` gives me emitter off.

MartyG-RealSense, 2023-02-21:
Hi @haowu-figure  In https://github.com/IntelRealSense/librealsense/issues/6191#issuecomment-608519909 it looks as though @dsvua  is not literally saying to put == 0 or == 1 in the program instruction but instead is saying that they get no laser pattern on the image when RS2_FRAME_METADATA_FRAME_EMITTER_MODE has a value of 1 (on) and get the pattern when it has a value of '0' (off).

https://github.com/IntelRealSense/librealsense/issues/9524 demonstrates a C++ method for displaying a metadata parameter name with an **std:: court** print instruction.  So for the emitter metadata, the line should look like this:

`std::cout << df.get_frame_metadata(RS2_FRAME_METADATA_FRAME_EMITTER_MODE) << std::endl;`

******

An alternative method for retrieving the emitter on / off status would be to use a **GET_OPTION** instruction to get the current value of the emitter status (0 for off, 1 for on) instead of setting it.