realsenseai/librealsense #6195: Getting IMU sensor timestamp

oafolabi, 2020-04-03:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | (05.12.03.00) | 
| Operating System & Version |   Win (10)  | 
| Platform                                 | PC  |
| SDK Version                            |   2.33.1                           | 
| Language                            |  C                       | 
| Segment			|  others                          | 

### Issue Description
I am trying to obtain the sensor timestamp from the IMU on my D435i.  As far as I know, given the motion frame, I am able to get the timestamp using frame.get_timestamp(). However, the timestamp is in the global domain. I would like to get the raw hardware timestamp. Currently for the images, I can do this by calling frame.get_frame_metadata(RS2_FRAME_METADATA_SENSOR_TIMESTAMP ). This returns the hardware clock timestamp for the images. Unfortunately, the motion frame does not support RS2_FRAME_METADATA_SENSOR_TIMESTAMP, and I don't know why. 

I have tried setting RS2_OPTION_GLOBAL_TIME_ENABLED to false, but I still get the same range of values (looks like UTC time) from the motion frame. However, using realsense-viewer, I can see that somehow it is able to get the right values. 

I have followed the instructions for getting metadata on windows. I checked to see that the GLOBAL_TIME_ENABLED is set to false using get_option(...) and indeed it is set to false.

ev-mp, 2020-04-05:
@oafolabi , can you attach a screenshot with `Realsense-viewer` that shows "Global Timestamp" when the flag is turned off in the options ?
[image]

I'm using Viewer v2.33.1 from the Release page and cannot reproduce any of the above.

Also the proper metadata attribute for IMU timestamp is **RS2_FRAME_METADATA_FRAME__TIMESTAMP**.  
The attribute RS2_FRAME_METADATA_~~SENSOR~~_TIMESTAMP  is a video sensor-specific  timestamp applicable for UVC streams only by design.

oafolabi, 2020-04-07:
@ev-mp , thanks for the quick response. In realsense viewer everything is fine. The problem I was having was in the code (using librealsense for my own project). I have switched from using  get_timestamp() to get_frame_metadata() using RS2_FRAME_METADATA_FRAME_TIMESTAMP and I now get it in the time  hardware domain (I think, since the number is smaller). However, when I call get_frame_timestamp_domain() it still returns Global Time, even though I have set GLOBAL_TIME_ENABLED to false.

ev-mp, 2020-04-07:
@oafolabi , the `GlobalTimestamp` option is toggled per each sensor individually. Please make sure that you've disabled that option for the `Motion`/All sensors and not only for the first/depth sensor.
You can use Realsense-viewer as a reference to verify the status of that particular option as shown in the screenshot.

RealSenseCustomerSupport, 2020-04-09:
----------------------------------------------

Hi @oafolabi,

Did you have a chance to check @ev-mp's advice out and to disable GlobalTimestamp option for sensors that you interested in or for all the sensors?

Thank you!

RealSenseCustomerSupport, 2020-04-16:
----------------------------------------------

@oafolabi,

Do you still need help with this question?

oafolabi, 2020-04-17:
No, I think I've solved my problem,  although I'm curious as to why I couldn't get the hardware timestamp with get_timestamp() and had to use get_frame_metadata(). Thanks for your help though. @ev-mp , yes I turned it off for each sensor. I will be closing this issue.

closed 2020-04-17 by oafolabi