realsenseai/librealsense #6196: Hand tracking ?

adanacademic, 2020-04-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {SR305 } | 
| Firmware Version                   | (3.27.3.0) | 
| Operating System & Version |   {Win (10)  |                                     | 
| Platform                                 | PC |
| SDK Version                            |  { 2 }                          | 
| Language                            |  {/opencv/python}                          | 

I have just bought SR305 to work with hand joints coordinates, however, I haven't found any place to start! 
is the SDK 2.0 support hand tracking? the page information explicitly describe the hand tracking capability of the camera, where is the place to get such data?

MartyG-RealSense, 2020-04-03:
The SR300 camera technology inside the SR305 is technically capable of processing hand and finger joint information, and was once able to do so in an older RealSense SDK for Windows from 2016. 
 Unfortunately the RealSense SDK 2.0 does not have hand and finger joint tracking functions, and the SR305 is not compatible with this 2016 SDK.

It is possible to detect and track hand joints with "machine learning" algorithms, but this typically requires multiple cameras and the SR305 is not best suited for this as multiple SR305s can interfere with each other (unlike the RealSense 400 Series cameras, which do not experience interference).
  
The link below jumps to a point in an Intel presentation about body tracking on YouTube that discusses finger joint tracking.

https://www.youtube.com/watch?v=VSHDyUXSNqY&feature=youtu.be&t=737

 If OpenCV and Python are your preferred platforms then you may be able to put together a project that can track finger joints by googling for 'opencv hand tracking python'.  Here is an example of a result from such a search:

https://dev.to/amarlearning/finger-detection-and-tracking-using-opencv-and-python-586m

adanacademic, 2020-04-03:
Thank you @MartyG-RealSense . I've spent the past hours googling the issue and searching for solutions. The thing that I bought the camera for is to obtain ready hand joints to use for other applications. I came across this comment [https://github.com/IntelRealSense/librealsense/issues/4435#issuecomment-525617614](url) and managed to detect my sr305 as sr300, however, I am still sticking on how to get the hand pose coordinates using the camera ( how the sr300 was doing it ??) 
any help?

MartyG-RealSense, 2020-04-03:
The 2016 Windows SDK that can detect hand joints can be downloaded from the link below.  I can't guarantee that it will work even if your SR305 is detected as an SR300, but I will provide the link anyway.

https://software.intel.com/en-us/articles/realsense-sdk-windows-eol

Find the 'Full SDK' link on the page and left-click on it to launch a 1.8 gb download of the installer file for the complete SDK in your browser.

[image]

MartyG-RealSense, 2020-04-03:
Are you launching the applications from the Sample Browser program please that is found inside the Intel RealSense SDK Gold folder the SDK installation placed on your Windows desktop?

adanacademic, 2020-04-03:
yes exactly @MartyG-RealSense .
I am launching it from the intel realsense sdk sample browser

MartyG-RealSense, 2020-04-03:
I am recalling these programs from memory so I apologise for inaccuracies.  In the Hand Cursor Viewer program, could you try un-ticking Skeleton in the Tracking Types list before you click Start so it is only tracking Joints, please?

adanacademic, 2020-04-03:
sorry, this option doesn't exist as well. even in cursor mode, the tool crashes as soon as I try to show my hands or moving object.
@MartyG-RealSense

MartyG-RealSense, 2020-04-03:
This SDK was made four years ago in a different era and may have become more unstable on modern Windows versions as it has not been updated since then.  It was also made for the SR300 or SR300-compatible cameras such as the Creative BlasterX Senz3D and the Razer Stargazer, so forcing the SR305 to emulate an SR300 may not be a close enough match.

adanacademic, 2020-04-04:
Thank you @MartyG-RealSense . I have managed to solve the problem. 
the crash reason was the poor PC capabilities, so I moved to a more powerful one and the SDK applications work just fine. I want to know how to save the data acquired by the camera (including the detected hand joints) using the SDK.

MartyG-RealSense, 2020-04-04:
I'm really pleased to hear that you found a solution for your crash problems.  Thanks for the update!

The 2016 SDK also has record and playback functions like the current RealSense SDK 2.0.  They have totally different architectures though.  Whilst the current SDK can record data to files called 'bags', the 2016 SDK uses a file format called RSSDK (which is a form of the video format **h.264**, also known as **MPEG-4 Part 10).**

The Sample Browser has a program called "Clip Editor" that can be used to edit RSSDK files.

https://github.com/IntelRealSense/librealsense/issues/2660#issuecomment-435348648

The documentation for the 2016 SDK's record and playback functions is in the links below:

https://software.intel.com/sites/landingpage/realsense/camera-sdk/v1.1/documentation/html/index.html?doc_essential_record_and_playback_streams.html

https://software.intel.com/en-us/blogs/2015/10/15/how-to-record-and-playback-streaming-sequences-in-intel-realsense-sdk

Documentation about the hand joints is here:

https://software.intel.com/en-us/articles/intel-realsense-sdk-working-with-hands-gestures

https://software.intel.com/sites/landingpage/realsense/camera-sdk/v1.1/documentation/html/index.html?doc_hand_accessing_joint_data.html

https://software.intel.com/sites/landingpage/realsense/camera-sdk/v1.1/documentation/html/index.html?doc_hand_hand_joints.html

https://software.intel.com/sites/landingpage/realsense/camera-sdk/v1.1/documentation/html/index.html?doc_hand_jointdata_properties.html

https://software.intel.com/sites/landingpage/realsense/camera-sdk/v1.1/documentation/html/index.html?doc_hand_local_and_global_joint_rotatio.html

lupusilviu, 2020-04-09:
D435i and D415 cameras, can be used to do Hand Tracking?

MartyG-RealSense, 2020-04-09:
They can follow the hand using software called Cubemos Skeleton Tracking SDK.  It costs $75 but has a free trial version.  

https://www.intelrealsense.com/skeleton-tracking/

To track individual finger joints with a 400 Series camera though, you may need to rely on using vision computing software such as OpenCV.

lupusilviu, 2020-04-09:
> They can follow the hand using software called Cubemos Skeleton Tracking SDK. It costs $75 for a licence but has a free trial version.
> 
> https://www.intelrealsense.com/skeleton-tracking/
> 
> To track individual finger joints with a 400 Series camera though, you may need to rely on using vision computing software such as OpenCV.

Thank you.  Will give it a try.

lupusilviu, 2020-04-09:
Unfortunately, the Skeleton-tracking detects only the  joints, up until the wrist. So it does not do hand tracking.

MartyG-RealSense, 2020-04-09:
The other main skeletal tracking solution compatible with the 400 Series cameras is Nuitrack SDK.  That is about $40 for an annual subscription licence.

https://community.nuitrack.com/t/joints-names-and-locations-scheme/685

https://nuitrack.com/

MartyG-RealSense, 2020-04-26:
This case will be closed after 7 days from the time of writing this message if there are no further comments.  Thank you.

MartyG-RealSense, 2020-05-04:
Closed due to no further comments received.

closed 2020-05-04 by MartyG-RealSense

GFTQ, 2021-08-11:
@adanacademic Dear Friend, I meet the same crushes of hand skeleton tracking modules, in the case of Intel Realsense 2016 SDK in Win10 21H1 version. As your mentioned " the poor PC capabilities", how to solve it? Can you explain more about "move to a more powerful one" in detail? Thanks very much.

MartyG-RealSense, 2021-08-11:
Hi @GFTQ  As time goes by, modern Windows versions are likely to become increasing incompatible with the old RealSense 2016 R2 SDK for Windows, as official support for it ended years ago.  The 2016 R2 SDK is not open source, so there is nothing that can be done by RealSense users to update it for modern Windows computers.  Changing the computer to a more powerful one would therefore likely make no difference.

GFTQ, 2021-08-12:
Dear @MartyG-RealSense,  I just moved 2016 r2 sdk to a powerful computer with Win10 version 1709, and hand tracking did work for a while without interruptions. However, since win 10 cumulative updates, all the mentioned crashes come back.  As you mentioned that the compatibilities of the old SDK to both win10 and unity turns bad. I plan to introduce [hand tracking](https://github.com/IntelRealSense/hand_tracking_samples) into unity with [wrappers](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/unity)
Thank you for your kind and patient responses. 
Best Wishes.

MartyG-RealSense, 2021-08-12:
My understanding is that if you have Windows 10 Pro then once you get a setup that works for you then you can set a 'Group Policy' so that it never installs updates automatically without your permission.  With Windows 10 Home, you can stop updates for 35 days instead of preventing them.

https://www.windowscentral.com/how-stop-updates-installing-automatically-windows-10#disable_automatic_windows_update_settings