realsenseai/librealsense #6197: About D435i

nych9811, 2020-04-05:
Hi. I'm Using D435i. I have questions.

In Intel, they said they measure Depth by active IR Stereo technology. I have a question about this tech.

How many Infrared sensor inside? I understand there are 2 RGB sensors inside, and it measures depth by using parallax. But, I can't check how many Infrared sensor inside. I understood this technology like these.

1. There are 2 RGB sensors inside, and 1 Infrared sensor at the center of camera.
2. There are 2 RGB sensors inside, and 2 Infrared sensor with each RGB sensor.

Which on is right? Please help me!

dorodnic, 2020-04-05:
Hi @nych9811 
2 infrared sensors, 1 RGB (visible light) sensor, 1 infrared projector + an inertial measurement unit (IMU)
Depth is being generated from the two infrared sensors. This is done because indoor or in poor lighting stereo-matching is benefiting from extra projected pattern, and you cannot do this in visible light (you can, but it would be, well, visible) 
Single extra RGB sensor is used to give texture to generated point-cloud.

nych9811, 2020-04-05:
Dear @dorodnic 
Thanks to your reply. Can you tell me more details about generating depth with two infrared sensors? If you have any link about this, please copy it to me.
Thanks!

closed 2020-04-05 by nych9811