realsenseai/librealsense #6206: stereo and RGB stream delay very bad with error get_device_time_ms() took too long (more then 2mSecs)

juliussin, 2020-04-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version |   JetPack 4.3  | 
| Kernel Version (Linux Only)    |  Linux 4.9.140-tegra aarch64                                         | 
| Platform                                 | NVIDIA Jetson Nano  |
| SDK Version                            | 2.34.0 }                          | 

### Issue Description
I'm having a trouble with my RealSense. When I try to stream with `realsense-viewer`, there is a delay like almost 1 second. First, there are some warnings about IMU:
```
WARNING [548345090496] (types.cpp:49) Accel Sensitivity:hwmon command 0x4f failed.
Error type: No data to return (-21).
WARNING [548345090496] (ds5-motion.cpp:464) IMU Calibration is not available, see the previous message
```
And when I stream with stereo module, there is delay and little lag with error:
```
ERROR [547742544320] (type.h:307) get_device_time_ms() took too long (more then 2 mSecs)
ERROR [548299313600] (type.h:307) get_device_time_ms() took too long (more then 2 mSecs)
~ and a lot of this continously
```

Same thing happened with RGB camera. My friend use same hardware (NVIDIA Jetson Nano) and through similar installation process, with another RealSense D435i, didn't have this very bad delay and lag like mine. I wonder why does this happen.

MartyG-RealSense, 2020-04-07:
The **get_device_time_ms** error was reported last week.  It looks as though a fix has been created that is awaiting approval to merge it into the SDK.

https://github.com/IntelRealSense/librealsense/issues/6189

The IMU error may appear if the IMU is uncalibrated.  Calibration can be performed with the tool in the link below.

https://github.com/IntelRealSense/librealsense/tree/master/tools/rs-imu-calibration

An Intel guide document for the process is here:

https://dev.intelrealsense.com/docs/depth-camera-d435i-imu-calibration

MartyG-RealSense, 2020-04-26:
I will close this case now, since the RealSense development team are aware of the issue with the **get_device_time_ms** error and are likely dealing with it.

closed 2020-04-26 by MartyG-RealSense

juliussin, 2020-04-26:
@MartyG-RealSense Is it already fixed?

MartyG-RealSense, 2020-04-26:
@juliussin  It looks like the fix is still waiting to be approved.

@ev-mp   Is there any update please on when the **global_timestamp_reader** pull will be reviewed and approved?  Thanks!

https://github.com/IntelRealSense/librealsense/pull/6150

juliussin, 2020-04-26:
@MartyG-RealSense  okay, I'll downgrade to get over the error for a while

MartyG-RealSense, 2020-04-26:
@juliussin  Thank you very much for your patience.