realsenseai/librealsense #6207: Improving depth resolution using depth table (D435i)?

rwchu, 2020-04-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.12.01.00 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC |
| SDK Version                            | 2.31.0.1234  | 
| Language                            |  C++     | 

### Issue Description
I am imaging an object that I know is between 0.5m and 1.5m away from the camera. I'm trying to maximize the depth resolution as I am trying to track very small movements, and ideally I would like to constrain the depth limits between those two distances while still using the full 16-bit range. I have played around with the settings in the viewer under Depth Table such as disparity shift, but I can't seem to get them right. 

I understand that Intel doesn't offer full documentation for these advanced settings, but I would appreciate any pointers or guidance. Has anyone else tried to play around with the depth table settings?

MartyG-RealSense, 2020-04-07:
I would recommend using the 'Threshold Filter' post-processing filter instead to clamp the depth to the desired minimum and maximum depths.  

You can test this easily by expanding open the 'Threshold Filter' option under "Post-Processing" in the RealSense Viewer options, setting the filter to Enabled and entering the minimum and maximum depths in meters ('0.5' minimum, '1.5' maximum).  Then when you start the streaming, the depth information should be limited to the defined range.

The link below has a script for setting the Threshold Filter with C++ code.

https://stackoverflow.com/questions/59054413/intel-realsense-depth-camera-d435i-noises-and-mounds

For readers using C#, a RealSense community member shared a C# script for the Threshold Filter here:

https://support.intelrealsense.com/hc/en-us/community/posts/360039445773/comments/360010898533

rwchu, 2020-04-07:
Thanks Marty for your quick response. I tried the Threshold Filter but as far as I can tell, this only zeroes the pixels outside of the specified depth range without affecting the depth resolution. Is there no way to constrain the depth while having the full range of 16-bit values within that constrained region?

MartyG-RealSense, 2020-04-07:
You could combine the Threshold Filter (for depth clamping) with reducing the Depth Unit scale for improved close-range accuracy.  The scale on the 400 Series cameras can be reduced from its default of 0.001 (meters), or 1mm, to a minimum of 0.0001

https://support.intelrealsense.com/hc/en-us/community/posts/360034848973/comments/360008694314

Using a high resolution also improves accuracy.  The optimal resolution for D435 models is 848x480.

MartyG-RealSense, 2020-04-26:
This case will be closed after 7 days from the time of writing this message if there are no further comments.  Thank you.

nikostsagk, 2020-04-27:
So, in the end, by changing the depth units you get better accuracy or is just a way to increase the quantisation of the depth scale?

MartyG-RealSense, 2020-04-27:
The general opinion of other RealSense team members is that it has more to do with quantisation.

https://github.com/IntelRealSense/librealsense/issues/5487

MartyG-RealSense, 2020-05-11:
This case will be closed after the next 7 days from the time of writing this message if there are no further comments received.  Thanks!

MartyG-RealSense, 2020-05-24:
Case closed due to no further comments being received.

closed 2020-05-24 by MartyG-RealSense