realsenseai/librealsense #6208: Depth Accuracy of D435/D435i under varying inclination angles (tilt camera).

szx0112, 2020-04-07:
I have read a lot of docs from the forum discussed about the depth accuracy/precision of D435/D435i. It is clear that the accuracy is correlated to the distance between target and camera. 

Such measurement are often performed at a flat wall with camera perpendicular point to the wall. However, I am quite interesting to understand what is the depth accuracy when there is an inclination angle between the camera and the wall (i.e. camera angle is title, not perpendicular anymore).

I can see the accuracy reduction in my experimental test. However, I am not sure if such accuracy degradation is only affected by the distance difference between the pixels (the pixel near or far from the camera) or the camera title angle also play a role (e.g. lens distortion). If yes, how much does it affects the results? 

I see many tests, such as human tracking or drone mapping, mounted the camera with a tilt angle to the ground. I believe better understanding such scenery can help those applications. I put the discussion here because I haven't not find any document provide a full discussion about such parameter. Any comment is really appreciated.

MartyG-RealSense, 2020-04-07:
Section 14a of Intel's camera tuning guide states that in scenes where there Is a repetitive structure such as a grid or fence, tilting the camera up to 20 degrees from the horizontal may be beneficial.

https://www.intel.co.uk/content/www/uk/en/support/articles/000027833/emerging-technologies/intel-realsense-technology.html

In general though, my understanding is that straight-ahead or straight-vertical (e.g pointing 90 degrees downwards) is the optimal orientation.

I recall a case where a RealSense user tilted the camera between 26 and 40 degrees from the straight ahead and experienced decreased accuracy when doing so.

https://github.com/IntelRealSense/librealsense/issues/2723#issuecomment-499153021

szx0112, 2020-04-08:
@MartyG-RealSense Thanks for the reply. I agree the vertical to target is the best setup. However, for our usage, RS camera is applied in a small pipe/tunnel. The minimal sensing distance of realsense make it difficult points directly to the tunnel's inner surface. Thus, camera with tilt angle is selected for the current setup. However, we observe ghost effects at pixels close to the boundary of camera FOV.

In the first document, it seems that the 20 degree tilt angle is to avoid the repetitive features (like use affine transformation to minimize those repetitive features in the left and right images I guess).

The human tracking case (#2723) is very similar as our application scenery, the near points have smaller error than the further points. This can be (1) the distance of those near points are closer to the camera (so has smaller RMS error); (2) the inclination angle of the near points are smaller (if camera points downward to the ground is 0 degree ) than the further points; (3) the effect of both (1) and (2).   

Currently we assume its like (3) but the RMS error is dominated by (1). This make the setup a little bit simple. However, having the clearer understanding would help us the find the optimal setup for our application.

Thanks.

MartyG-RealSense, 2020-04-08:
If by "minimal sensing distance" you mean the minimum distance of objects from the camera before the image starts to break up if you get any closer: you can reduce the minimum distance further by increasing the "Disparity Shift" setting.  This has the effect of reducing minimum distance (MinZ), though also decreasing maximum observable distance (MaxZ).  

A cruder way of reducing minimum distance is to use a lower resolution, though this has an impact on accuracy.

My understanding is that there is a practical limit of around 20 cm from an object, below which the left and right imagers may start getting confused.

RMS error starts off very small at the camera lens and increases linearly over distance.  The D435 has around twice the RMS error over distance of the D415 model, though both models have similar error at close range.  If minimising the minimum sensing distance is important to you then the D435 is a better choice though.

You may be interested in an article on construction inspection with RealSense that was published on the RealSense blog recently, and an associated paper on using RealSense D435 for  inspection of concrete structures.

https://www.intelrealsense.com/the-future-of-construction/

https://rd.springer.com/chapter/10.1007/978-3-030-17798-0_49

szx0112, 2020-04-12:
Agreed. I believe it is all about the parameter tunning (disparity shift, depth units, resolution, exposure,...), and there are trade-offs between accuracy and depth range. 

Thank you for the kindly help.

closed 2020-04-12 by szx0112

MartyG-RealSense, 2020-04-12:
You're very welcome.   Thanks for the update.   Good luck!   :)

MartyG-RealSense, 2026-03-22:
Hi @sunanditaghosh29  On a D435 camera you can find the ground truth of depth by adding to it a minus value of **-4.2** millimeters.

As RealSense depth values are output in meters, you could try adding **-0.0042** meters (the meter equivalent of -4.2 mm) to the measurements to find the ground truth version of the depth value in meters.