realsenseai/librealsense #6209: large lag increase between 2.32.1 and 2.34 (head) on osx

timprepscius, 2020-04-07:
Mac

There was a huge increase in lag between 2.32.1 and 2.34.
I've included screenshots that show the lag for 2.31, 2.32.1 and 2.34

2.31:
[image]

[image]

[image]

Lag between 10 and 12 ms

2.32.1:
[image]

[image]

[image]

Lag between 8 and 11 ms

Head (2.34)
[image]

[image]

[image]

lag: 48ms, 51ms and 42ms

timprepscius, 2020-04-07:
Also, related, but not the bug:

When 90fps is selected, on pretty much any resolution, the is a a large lag for all version >= 2.31.  I don't have any versions pre 2.31 atm.

Do you have tests to find out the lag post usb transfer?
I was looking in your code, and you have things like:

```
        std::vector<byte> pixels(pix, pix + fo.frame_size);
        fr->data = pixels;

```

which sadly aren't optimized to fr->data.assign(pix, pix + fo.frame_size); on my mac.
or a std::move or whatever.

I am wondering how much overhead you have post usb transfer.

If it is more than 1 ms, maybe it would be good to invest some time in optimizing that.  The lowest hanging fruit is using a memory pool for all of your frames, buffer transfers usb.  Memory allocation is amazingly slow, and if you can get rid of it, might be worth it.

Or not.

Hope you all are all healthy and safe.

MartyG-RealSense, 2020-04-07:
In the last few images of your set, the log at the bottom shoes that you are experiencing an error reported last week:

https://github.com/IntelRealSense/librealsense/issues/6189

And also in the past day:

https://github.com/IntelRealSense/librealsense/issues/6206

timprepscius, 2020-04-07:
Here are some images of the 90fps lag:

3 with 1 infrared and 1 depth:
33ms, 23ms, 27ms

[image]

[image]

[image]

3 with just infrared:
17ms, 20ms, 30ms

[image]

[image]

[image]

timprepscius, 2020-04-07:
I'm sorry actually I guess those aren't ms, those are 1/100's of a second.  So it's ten times worse than what I'm saying.

MartyG-RealSense, 2020-04-07:
Looking in the options side panel, your camera is generating temperatures in the range of 44-45 degrees.  That is an abnormal level.  I would expect to see 25-28 in a camera that starts streaming from a cold start.  Once you get past 42 degrees, problems may start manifesting (the official recommended maximum operating temperature is 35).  

You will not burn the camera out though, as it has built in temperature safeguards that can reduce the camera's laser power at first and then shut the laser off if a very high temperature persists after that.

timprepscius, 2020-04-07:
These high lag times at 90fps occur without the laser.  And I think at first cold start.

Let me check in 5 minutes.

MartyG-RealSense, 2020-04-07:
The temperature reporting in the 44-45 range (the ASIC temperature) refers to the temperature of the camera's circuit board.

timprepscius, 2020-04-07:
Ok let the device sit, turned on did test.
still large lag:

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

MartyG-RealSense, 2020-04-07:
Your ASIC temperatures are better in those new tests.  I just did my own test at 90 FPS and 848x480 with Depth, Infrared1 and Infrared2 all turned on.  After a minute of streaming, my ASIC temperature was 33, compared to a max of 28 at 30 FPS.  And it is currently a cool night in my location.   So the 90 FPS speed does seem to have a direct bearing on the operating temperature.

Do you get the same kind of lag if you run at 30 or 60 FPS?

timprepscius, 2020-04-07:
It is dependent on the image size and fps.

I've actually been recording a lot about this.

These aren't using your application:

1 infrared @ 640x480 @ 90FPS
[image]
[image]
[image]

1 infrared @ 848x480 @ 90FPS
[image]
[image]
[image]

1 infrared 1 depth @ 640 @ 90FPS
[image]
[image]
[image]
[image]

1 infrared 1 depth @ 848 @ 90FPS

[image]
[image]
[image]
[image]
[image]
[image]

1 inf 1 depth @ 640 @ 30 FPS
[image]
[image]
[image]
[image]
[image]
[image]

1 infrared 1 depth 848 @ 30 FPS

[image]
[image]
[image]

timprepscius, 2020-04-07:
I can do another pass tomorrow, but I've been doing this all day, trying to figure out what resolution I can use to minimize lag.

In a nutshell, what I think I'm finding, (if the asic temperature didn't skew results too much).

The greater the size of the image(s), the more the lag.
When fps is set to 30fps intead of 90fps, it makes things better, but still not great.

between 60fps and 90fps there is a huge difference in lag.

Will return to this tomorrow.

MartyG-RealSense, 2020-04-07:
The volume of data (bandwidth) is proportional to the resolution and FPS speed.  The data volume will be smaller on lower resolutions / FPS speeds, and greater on higher resolutions / speeds.  90 FPS is a speed I do not often see on this forum, as 60 tends to be the more commonly used maximum.

The Vision Processor D4 image processing hardware inside the camera can do a certain amount of the processing work that a computer might otherwise have to do, but there comes a point where having a good computer hardware specification is helpful too as the processing demands placed on the computer are increased (by raising FPS speed, for example).   

I look forward to your next set of test results.   Good luck!

timprepscius, 2020-04-07:
Is there a way for me to get a timestamp on the frame, when it first arrives from the usb.  Before anything happens in your library before it is passed to me?

timprepscius, 2020-04-07:
I (think) I understand about usb interrupt rates, and what not. 

Do you believe the lag is happening in the device?
During the usb transfer?
Or post usb transfer?

timprepscius, 2020-04-07:
I seem to remember, while working with the freenect, that there were maximum usb frame sizes.  Is this true? Do you know how many usb frames a 640x480 takes vs a 848x480?

Do you know if the device is capable of transferring on adjacent usb frames or will it do one on one off, or something?

MartyG-RealSense, 2020-04-07:
I believe that the timestamp would be retrieved with the command **frame.get_timestamp()**

https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1frame.html#a25f71d45193f2f4d77960320276b83f1

There are various places where processing 'bottlenecks' can occur.   For example, a 400 Series camera's capabilities are considerably more restricted on a USB2 connection than on a USB3 connection, where its full range of capabilities are accessible.

Some types of processing, such as depth to color alignment, may add additional processing demands on the CPU that slow down the running speed of an application.  

When recording frames to disk, there may be a bottleneck if recording to slow-access storage such as a SATA hard drive instead of a fast solid state drive (SSD).

When attaching more than one camera to the same computer, the computer may have insufficient hardware resources to be able to cope with more than one camera if it is a low-spec device.

If you use your own choice of USB cable instead of the official one supplied with the camera then you may experience degraded performance if it is not a high-quality cable, especially if the cable is 2 meters or more in length.

A USB hub may handle data less efficiently than if the camera is attached directly to the computer.  A computer's USB port will tend to have a dedicated USB controller for each port, whereas a hub's USB controller might handle more than one port.

There are other examples of bottlenecks, but you get the idea.

I hope the link below will be helpful to your frame size question.

https://github.com/IntelRealSense/librealsense/wiki/Frame-Buffering-Management-in-RealSense-SDK-2.0#latency-vs-performance

I am off-shift for the day now as it is past midnight in my time zone, but I will be back in 7 hours from the time of writing this message.   Good luck!

timprepscius, 2020-04-10:
I made a small change in how you do frame->data memory in the realsense sdk.
This is 2.33.1:

Before modification (going to try to drag in a video):
Ah videos don't work.

I'll use my site.
This is clean 2.33.1
http://www.timprepscius.com/private/IMG_4376.MOV

This is with the memory pool
http://www.timprepscius.com/private/IMG_4377.MOV

This obviously is not a conclusive profiling, however, I think it shows promise.
The code change is:
(going to try to attach file)
[diff.txt](https://github.com/IntelRealSense/librealsense/files/4461660/diff.txt)

And the framedata allocator (which I did as quick as possible, and didn't test) is:

[framedata_allocator.h.txt](https://github.com/IntelRealSense/librealsense/files/4461666/framedata_allocator.h.txt)

MartyG-RealSense, 2020-04-10:
Thanks so much for sharing your results and code changes with the RealSense community!

2emoore4, 2020-04-16:
[image]

this is my first time using RealSense so I can’t compare to previous versions, but I too am experiencing significant latency on MacOS

what’s the source of this latency? the camera or the software? is there a way to expose the RealSense as a UVC webcam so I can use it without the viewer app?

2emoore4, 2020-04-16:
ah I see you've documented some sources of latency already

MartyG-RealSense, 2020-04-17:
@2emoore4  If you are asking whether the RealSense 400 Series cameras can be used as a standard webcam outside of the RealSense SDK ... yes they can, if you select 'RGB' as the video source (it tends to default to the Depth driver in programs, giving a black screen, so RGB needs to be manually selected in those cases).

I did tests of the D415 and D435i in the video functions of Skype and Discord.

**Skype**

[image]

**Discord**

[image]

If you would like to access streams such as RGB in a RealSense application that is not the RealSense Viewer, then you can certainly do that with programming.  There is a pre-made sample program in the RealSense SDK called **rs-capture** that displays the depth and RGB streams together.

https://github.com/IntelRealSense/librealsense/tree/master/examples/capture

timprepscius, 2020-04-17:
I've done some more optimization of your library.

Is there a way to get an fps with zero processing after a frame has been received from the usb?

so instead of processing a frame and passing it along, all I want to do is receive the frame and discard it.

I cannot get above 70 fps.
I've optimized your memory allocations and your split_frame.

When I look at your code, you are doing an huge number of shared_ptrs which are slow, but revising your code to get rid of the shared_ptrs will be more of an undertaking than I have time for at the moment.

Before doing that, I would like just see what the maximum is.
Is there somewhere I can put a "return ;" directly after a frame is received from the usb, before it is processed and forwarded along?

Thanks

MartyG-RealSense, 2020-04-17:
@timprepscius   Thanks again for your effort in seeking to improve Librealsense!  I will tag @dorodnic into this discussion regarding your question above about discarding received frames.

timprepscius, 2020-04-19:
I have been able to reach 90fps.

Things you need to optimize:
1.  split_frame:
   use a future to put split_a in a different thread

2. get rid of std::vector for your frame.data, I would suggest using a custom class which wraps a memory block, this way you can track exactly what is happening- and since you are dealing with intrinsic types, you don't need to call T::destruct on items, etc and all the other things std::vector does, or doesn't do.

3. only initialize your frame.data when it is metadata of some sort, or remove the need for the initialization altogether:
   
You are doing some things which are a bit weird for frame.data.  I would suggest instead of having a memory block which looks like { frame_pointer*, frame_pointer *, frame_pointer *}

you instead of a struct which looks like this:
struct CompositeFrameInfo {
   size_t framePointerQuantity;
   frame_pointer framesPointers[0];
} ;

This way, your data is does not necessitate initializing the memory buffer to zero.

timprepscius, 2020-04-19:
Also, the way you are doing your frame caching is weird and doesn't really work.  You could pretty much get rid of all of that.

MartyG-RealSense, 2020-04-19:
Thanks again @timprepscius for your ongoing suggestions!   (tag to @dorodnic )

dorodnic, 2020-04-23:
Hi @timprepscius 
You are welcomed to propose a pull-request and your profiling data. 
I was under the impression that latency in Mac OS was due to a bug in 2.34 related to global time queries. 
Proposal to not hold metadata in a vector makes a lot of sense.
I'm not fully following the rest. To the best of my knowledge, frame processing flow is only using shared_ptr value, not modifying the pointer and not allocating new memory (except for the metadata vector)
Running split frame in a background thread is not really free - you may reduce latency but increase CPU utilisation. 
No extra processing is performed on Z16 and YUYV formats. Using the Viewer for latency measurements seems pretty wasteful since the Viewer is doing a lot of extra work. We provide latency tool under opencv examples.

timprepscius, 2020-04-24:
Ummm..

Here is my experience with 2.33, which probably informs on 2.34.  But maybe not.

If I do not optimize your library, I can only get ~70 fps.  (2017 macbook air)
If I do optimize your library I easily reach 90fps.
There are two main optimizations: split_frame, and redoing how you do frame.data. 
Maybe I didn't need to do these if I knew more about your library, but so far, it seems necessary.

For my purposes, I need to handle all frames if possible, even when there is lag (but minimize lag), so I also use your frame queue (in certain builds).

IF I use 90fps without my optimizations, (or 60fps with a bunch of debugging on my end) I see the same sort of lag I saw with 2.34, which I assume is because frame_queue max frame count is some new number.

So basically, time needs to be spent optimizing your library.

Also please put a:

```
.h
// I really want to use size_t, but not defined.  Will just use int for now
int rs2_frame_queue_size(rs2_frame_queue* queue);

.cpp
int rs2_frame_queue_size(rs2_frame_queue* queue) BEGIN_API_CALL
{
    VALIDATE_NOT_NULL(queue);
    auto q = reinterpret_cast<rs2_frame_queue*>(queue); // yes I know, parameter should prob void*, not sure yet
    return q->queue.size();
}
NOEXCEPT_RETURN(0, queue)
```

MartyG-RealSense, 2020-05-24:
Hi @timprepscius    Do you have any further actions that you want to take with this case, or can it be closed?  Thanks!

MartyG-RealSense, 2020-05-30:
Case closed due to no further comments received.   @timprepscius  Please feel free to submit your optimisations as a pull request for the SDK as @dorodnic  suggested above.   Thanks so much for your great effort in seeking to improve Librealsense!

closed 2020-05-30 by MartyG-RealSense