realsenseai/librealsense #6211: No module named pyrealsense2

mynameher3, 2020-04-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D435i } | 
| Firmware Version                   | (Open RealSense Viewer) | 
| Operating System & Version |   {Win 10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Raspberry Pi  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python }                          | 
| Segment			|  {others }                          | 

### Issue Description
<I have sucessfuly instaled intel sdk 2.0 on raspberry and can run easy and simple python codes, but every time i try to run a code that needs to import pyrealsense2 it says no module was found in the first line, although it was generated while installing.  
The code i tried running was:

import pyrealsense2 as rs
pipe = rs.pipeline()
profile = pipe.start()
try:
  for i in range(0, 100):
    frames = pipe.wait_for_frames()
    for f in frames:
      print(f.profile)
finally:
    pipe.stop()

>

MartyG-RealSense, 2020-04-09:
Hi @mynameher3 

After you compiled Pyrealsense2, did you update your PYTHONPATH please with the command below?

**export PYTHONPATH=$PYTHONPATH:/usr/local/lib**

mynameher3, 2020-04-09:
Yes, my path is updated by vim but it´s still appearing the error...

[image]

MartyG-RealSense, 2020-04-09:
Does it make any difference if you move your script into the Librealsense **build** folder and launch it from there?

mynameher3, 2020-04-09:
I have already tried running in the librealsense/build  ,  /Desktop and on /python3 directorys the same file through all them and tried running form the console but its still apearing the same error

MartyG-RealSense, 2020-04-09:
And have you got the **.so** files generated by the wrapper build process in the same location as your script, as described by point number 5 of the instructions?

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#building-from-source

mynameher3, 2020-04-12:
I had to rename to "pyrelasense2.so" only, because the instalation have another name on it, now i have my script and the 2 .so file together in the same folder and had to run it from the console with python3 so now it can recognize it sucessfully!

closed 2020-04-12 by mynameher3

MartyG-RealSense, 2020-04-12:
Thanks so much for the update.  I'm very glad that you found a solution!