realsenseai/librealsense #6213: wrong values from rs2_deproject_pixel_to_point

ZiSo89, 2020-04-09:
I trying to map the background,
I use rs2_deproject_pixel_to_point for every x pixel(0 to 848) and y (constantly 50). 
I get back the xyz,
x is changes as expect, there is a wall on left of the camera at -0.45 meters and an other wall right of the camera +2.1 meters. 
but for the z (depth) got back 0.6 meters at the first pixels. 
Although there is no object above camera.
after the 150 pixel get the normal values 1.9 meters which is the distance from camera to the ceiling.

I was expected that z will be always 1.9 meters. why not? 

the code:

			vector<float> z_dis;
			
			for (int x_pixel = 0; x_pixel < 848; x_pixel++)
			{
				pixel u;
				float upoint[3];

				u.first = (float)x_pixel;
				u.second = (float)50;

				float upixel[2];
				upixel[0] = u.first;
				upixel[1] = u.second;
				auto udist = depth_dis.get_distance((int)upixel[0], (int)upixel[1]);

				rs2_intrinsics intr = depth_dis.get_profile().as<video_stream_profile>().get_intrinsics(); // Calibration data
				rs2_deproject_pixel_to_point(upoint, &intr, upixel, udist);

				z_dis.push_back(upoint[2]);

			}

			for (auto i = z_dis.begin(); i != z_dis.end(); ++i)
			{
				cout << " at x pixel: " << distance(z_dis.begin(), i);
				cout << "  Z is:" << *i << " " << endl;
			}

MartyG-RealSense, 2020-04-09:
Is the camera mounted on a moving mobile robot at a height close to the ground please?  

If the camera is facing straight ahead horizontally whilst mounted at a low height, then I wonder if the camera might be picking up the floor before its view reaches ahead to the ceiling.

It would be very helpful if you could provide an image of what your camera is seeing, and also which model of camera you are using (e.g D415 or D435).  Thanks!

ZiSo89, 2020-04-09:
the camera (D435) is on my desk and not moving.  And is facing straight the ceiling.

[image]

MartyG-RealSense, 2020-04-09:
If the camera is directly facing the ceiling, is it pointing near a light on the ceiling, please?

ZiSo89, 2020-04-09:
My English is not so good so I did not got that, Here is what the camera real see. In the gray image you see only a wall on the left and ceiling.
 
[image]

MartyG-RealSense, 2020-04-09:
Your English is very good, do not worry.  :)

It looks to me as though the reason you are seeing depth at 0.6 m is because that part of the image is not the ceiling but is the side wall.  The wall has lower depth than the ceiling because the bottom of it is closer to the camera. 

As you go higher up the wall on the image, the color of that area changes from dark blue to light blue because the distance of the wall surface becomes further away from the camera

ZiSo89, 2020-04-09:
Ok I understand, thank you.

closed 2020-04-09 by ZiSo89