realsenseai/librealsense #6215: Announcement:  Intel publish white paper on high speed 300 FPS depth capture for D435

MartyG-RealSense, 2020-04-09:
Hi everyone,

Intel have published a new white paper document on the subject of high speed 300 FPS depth capture for the RealSense D435 depth camera.

https://dev.intelrealsense.com/docs/high-speed-capture-mode-of-intel-realsense-depth-camera-d435

There is an accompanying article on the official RealSense blog.

https://www.intelrealsense.com/high-speed-mode-at-300-fps-for-depth-camera/

sam598, 2020-04-09:
Very cool @MartyG-RealSense !

@RealSenseCustomerSupport Would it be possible to have the 256x144 mode be pixel binned instead of center cropped so it can maintain the wide field of view? Having a high speed wide field of view depth camera would be outstanding.

GOBish, 2020-04-16:
The article talks about firmware 5.12.4.0 but I don't see this available anywhere.

MartyG-RealSense, 2020-04-16:
@GOBish  the mention of new firmware versions in documents such as white papers is not always in sync with their listing on the firmware download pages.  Sometimes it takes more time for download links to appear.  Currently, 5.12.3.0 is the most recent available firmware.

closed 2020-04-17 by MartyG-RealSense