realsenseai/librealsense #6216: Why is 2.27.0 displayed after upgrade （I need 2.34.0）

mingyangpeng, 2020-04-10:
| | |
|---|---|
|**librealsense**|2.27.0 RELEASE|
|**OS**|Linux|

Please provide a description of the problem

MartyG-RealSense, 2020-04-10:
If you build the SDK from source, I wonder if it may help if you go to the Librealsense **build** directory and use the instruction below to clean it up and then install Librealsense:

**sudo make uninstall && make clean && make && sudo make install**

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#building-librealsense2-sdk

mingyangpeng, 2020-04-10:
I tried it, it didn't work, thank you!

MartyG-RealSense, 2020-04-10:
I'm sorry that it did not work for you.  I have some questions for you, please.

1.  What computer hardware are you using please?  PC, Mac, Raspberry Pi, Jetson, etc.

2.  Please describe the method that you used when trying to upgrade.

3.  What SDK version were you upgrading from when trying to upgrade to 2.34.0?

mingyangpeng, 2020-04-10:
Thank you again for your warm help，I've uninstalled the old version and plan to reinstall the new one.
1. I used DELL PC -- Ubuntu 18 
2.1 echo 'deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial main' | sudo tee /etc/apt/sources.list.d/realsense-public.list 
2.2 sudo apt-key adv --keyserver keys.gnupg.net --recv-key 6F3EFCDE
2.3 sudo apt update && sudo apt upgrade   
3. old version: 2.27.0

closed 2020-04-10 by mingyangpeng

MartyG-RealSense, 2020-04-10:
Thanks for the details.

I see that you closed this question.  Did you mean to close it or do you still need help please?

mingyangpeng, 2020-04-12:
Thanks for your help, I have reinstalled it.

MartyG-RealSense, 2020-04-12:
Ok, thank you very much!