realsenseai/librealsense #6218: Use rs_options.hpp th

drfg0119, 2020-04-10:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

drfg0119, 2020-04-10:
Here is my code.
I follow the function(called"registerDynamicOption()") from base_realsense_node.cpp
I do not sure the method I used is correct or not, so, is there any tutorials and sample code I should followed? Please let me know. thank you

[image]

drfg0119, 2020-04-12:
Oh, I found the functions are declared in private under the BaseRealSenseNode class, no wonder I couldn't do it... QQ
Is there any method could make me write a dynamic reconfigure project? Although there is a public function called BaseRealSenseNode() which is under BaseRealSenseNode class, I don't know how to use it... I want to write a node that can loop the parameter values autoly, please help me to find the method,thank you.

RealSenseCustomerSupport, 2020-04-14:
----------------------------------------------

The temperature node should provide some reference for doing the similar task.
https://github.com/IntelRealSense/realsense-ros/blob/7449930b7fc07877ca6bcc450da8e2d8fd6ee8c3/realsense2_camera/src/base_realsense_node.cpp#L2236

Please note that this issue will be closed if we don’t hear from you in 7 days.
Thank you.

drfg0119, 2020-04-18:
Hey @RealSenseCustomerSupport ,The temperature nodes didn't work for me, I solved the compile problem, but I didn't know how to write the node by using ddynamic_reconfigure method, because I want to set the values dynamically.

I am trying to write a node can loop the value in a range(for example: motion range 0~60) when run the rs_camera.launch, should I write a register to save the value(EX: registerEnumVariable) and then publish it ? I followed the base_realsense_node.cpp in Line 383 to 486, which is under realsense-ros.
Or even I don't need to do this...

But I can run the rqt_reconfigure to set values in the "coded-light sensor" node dynamically, so I thought that is available for me to write a node to control SR305.

could you please help me thank you.

drfg0119, 2020-04-18:
Hi @RealSenseCustomerSupport  , I have written a node named controller like this, because I didn't know how to deal with the protected member in `class options` which is under **rs_option.hpp**, so I used inheritance to make a new class called `controller`, but when I found there is error when it run into the function defined myself like `ctr.get_MvR(option)`, then I check the function step by step, I found bug is `error::handle(e)` in my hpp file. Is there any ideal could suggest? I'll appreciate.

```C++
#include </opt/ros/melodic/include/ros/ros.h>
#include </home/nvidia/catkin_ws/src/ddynamic_reconfigure/include/ddynamic_reconfigure/ddynamic_reconfigure.h>
#include <string>
#include <map>
#include <iostream>
#include <fstream>
#include "/home/nvidia/librealsense/include/librealsense2/h/rs_option.h"
#include "/home/nvidia/librealsense/include/librealsense2/h/rs_types.h"
#include <algorithm>
#include <cctype>
#include <mutex>
#include "controller.hpp"

using namespace ddynamic_reconfigure;

int main(int argc, char **argv)
{

    ROS_INFO("Spinning node");
    ros::init(argc, argv, "controller");
    ros::NodeHandle nh;
    ddynamic_reconfigure::DDynamicReconfigure ddynrec;
    rs2::controller ctr; // like rs2::options sensor

    rs2_option option = static_cast<rs2_option>(16);  //no.16 is RS2_OPTION_MOTION_RANGE
    const auto sensor_option_value = ctr.get_MvR(option); // like sensor.get_option(option)
    auto option_value = int(sensor_option_value);

    ros::Rate loop_rate(10);
    int c = 0;
    while (ros::ok())
    {
        c++;
        option_value = 0;
        if(!(c%10))
        {
            option_value+=5;
            if(option_value > 60)
                option_value = 0;
        }
        ctr.set_MvR(option, option_value); // like sensor.set_option(option, option_value);

        ddynrec.registerVariable<int>(
            ctr.get_MvR_name(option), int(option_value),
            [option, ctr](int new_value) { ctr.set_MvR(option, new_value); },
            ctr.get_MvR_description(option));

        loop_rate.sleep();
    }
    ROS_INFO("Spinning node shutdown...");
    return 0;
}
```

Here is the controller.hpp written myself, it's copy from **rs_option.hpp**
```C++
#include "/usr/local/include/librealsense2/hpp/rs_options.hpp"
#include "/usr/local/include/librealsense2/hpp/rs_types.hpp"

namespace rs2
{
    class controller : public options
    {
    public:

        const char* get_MvR_description(rs2_option option) const
        {
            rs2_error* e = nullptr;
            auto res = rs2_get_option_description(_ctr, option, &e);
            error::handle(e);
            return res;
        }

        const char* get_MvR_name(rs2_option option) const
        {
            rs2_error* e = nullptr;
            auto res = rs2_get_option_name(_ctr, option, &e);
            error::handle(e);
            return res;
        }

        const char* get_MvR_value_description(rs2_option option, float val) const
        {
            rs2_error* e = nullptr;
            auto res = rs2_get_option_value_description(_ctr, option, val, &e);
            error::handle(e);
            return res;
        }

        float get_MvR(rs2_option option) const
        {
            rs2_error* e = nullptr;
            auto res = rs2_get_option(_ctr, option, &e);
            error::handle(e);
            return res;
        }

        void set_MvR(rs2_option option, float value) const
        {
            rs2_error* e = nullptr;
            rs2_set_option(_ctr, option, value, &e);
            error::handle(e);
        }

        std::vector<rs2_option> get_supported_MvR()
        {
            std::vector<rs2_option> res;
            rs2_error* e = nullptr;
            std::shared_ptr<rs2_options_list> options_list(
                rs2_get_options_list(_ctr, &e),
                rs2_delete_options_list);

            for (auto opt = 0; opt < rs2_get_options_list_size(options_list.get(), &e);opt++)
            {
                res.push_back(rs2_get_option_from_list(options_list.get(), opt, &e));
            }
            return res;
        }

        explicit controller(rs2_options* o = nullptr) : _ctr(o)
        {
        }

        controller& operator=(const controller& other)
        {         
            _ctr = other._ctr;
            return *this;
        }

        controller(const cpntroller& other) : _ctr(other._ctr) {}

        virtual ~controller() = default;

        template<class T>
        controller& operator=(const T& dev)
        {
            _ctr = (rs2_options*)(dev.get());
            return *this;
        }

        rs2_options* _ctr;
    };
}
```

drfg0119, 2020-04-18:
I set the print for finding the bug, 
here is in cpp. 

[image]

Here is under hpp. 
[image]

Here is the result. 
[image]

Please help me, thank you!

drfg0119, 2020-04-18:
hey I solved the problem, because I didn't add this after main code haha. 

[image]

But I need to face another problem now haha, the "e.get_failed_function()" throw the info(said "rs2_get_option"),and "e.get_failed_args()" also throw the info(said "option:nullptr, option:Filter Option"), finally the "e.what()" throw the info(said "null pointer passed for argument"option".").
Totally, the full sentence is 
**RealSense error calling rs2_get_option(option:nullptr, option:Filter Option):
     null pointer passed for argument "options".**
is there any idea could help me? thanks.

drfg0119, 2020-04-18:
but there isn't the same function, should I follow their solution?
And then I change my code, I realize that there is an class called **sensor** could do the same thing I want, so I delete my controller.hpp haha, but I am confused that still has same problem said:
**RealSense error calling rs2_get_option(option:nullptr, option:Filter Option):
null pointer passed for argument "options".**
did I use the wrong way...?

here is my code. thanks

```C++
#include </opt/ros/melodic/include/ros/ros.h>
#include </home/nvidia/catkin_ws/src/ddynamic_reconfigure/include/ddynamic_reconfigure/ddynamic_reconfigure.h>
#include </home/nvidia/librealsense/include/librealsense2/rs.hpp>
#include <vector>
#include <utility>
#include <map>
#include <iomanip>
#include <iostream>

using namespace ddynamic_reconfigure;

int main(int argc, char **argv) try
{
    ROS_INFO("Spinning node");
    printf("c1\n");
    ros::init(argc, argv, "controller");
    printf("c2\n");
    ros::NodeHandle nh;
    printf("c3\n");
    ddynamic_reconfigure::DDynamicReconfigure ddynrec;
    printf("c4\n");
  //rs2::controller ctr;
    rs2::sensor sensor;
    printf("c5\n");

    rs2_option option;
    printf("c6\n");
    option  = static_cast<rs2_option>(16);
    printf("c7\n");
    const auto sensor_option_value =  sensor.get_option(option);
    printf("c8\n");
    auto option_value = int(sensor_option_value);

    ros::Rate loop_rate(10);
    int c = 0;
    while (ros::ok())
    {
        c++;
        option_value = 0;
        if(!(c%10))
        {
            option_value+=5;
            if(option_value > 60)
                option_value = 0;
        }
        sensor.set_option(option, option_value);

        ddynrec.registerVariable<int>(
            sensor.get_option_name(option), option_value,
            [option, sensor](int new_value) { sensor.set_option(option, new_value); },
            sensor.get_option_description(option));

        ddynrec.publishServicesTopics();
        loop_rate.sleep();
    }
    ROS_INFO("Spinning node shutdown...");
    return EXIT_SUCCESS;

}
catch (const rs2::error & e)
{
    std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    "  << e.what() << std::endl;
    return EXIT_FAILURE;
}

catch (const std::exception& e)
{
    std::cerr << e.what() <<std::endl;
    return EXIT_FAILURE;
}
```

And then there is an compile warning, is it OK...??? 
[image]

and this is the error
[image]

drfg0119, 2020-04-19:
Hi @RealSenseCustomerSupport , here is my latest version, I face the problem that I couldn't run the rs_camera.launch with my node together. Could you please help me..? Thanks!

```C++
#include "ros/ros.h"
#include </home/nvidia/catkin_ws/src/ddynamic_reconfigure/include/ddynamic_reconfigure/ddynamic_reconfigure.h>
#include </home/nvidia/librealsense/include/librealsense2/rs.hpp>
#include <vector>
#include <utility>
#include <map>
#include <iomanip>
#include <iostream>
#include <algorithm>

static rs2::device get_a_realsense_device()
{
    rs2::context ctx;
    rs2::device_list devices = ctx.query_devices();
    rs2::device selected_device;
    selected_device = devices[0];
    return selected_device;
}

static rs2::sensor get_a_sensor_from_a_device(const rs2::device& dev)
{
    std::vector<rs2::sensor> sensors = dev.query_sensors();
    return sensors[0];
}

int main(int argc, char **argv) try
{
    ROS_INFO("Spinning node");
    ros::init(argc, argv, "controller");
    ros::NodeHandle nh;
    printf("c1\n");

    rs2::device dev = get_a_realsense_device();
    printf("c2\n");

    rs2::sensor sensor;
    printf("c3\n");

    sensor = get_a_sensor_from_a_device(dev);
    printf("c4\n");

    rs2_option option;
    printf("c5\n");

    option  = static_cast<rs2_option>(15);
    printf("c6\n");
    std::cout << option << std::endl;

    float sensor_option_value = sensor.get_option(option);
    printf("c7\n");
//  printf("%d\n", sensor_option_value);

    auto option_value = int(sensor_option_value);

    ros::Rate loop_rate(1000);

    int c = 0;
    option_value = 0;
    while (ros::ok())
    {
        c++;
        if(c!=0)
        {
            option_value+=5;
            printf("%d\n",option_value);
            if(option_value > 60)
                option_value = 0;
        }
        printf("ya1\n");
        sensor.set_option(option, option_value);
        printf("%d\n",option_value);

        printf("c8\n");
        ros::spinOnce();
        loop_rate.sleep();
    }
    ROS_INFO("Spinning node shutdown...");
    return EXIT_SUCCESS;

}
catch (const rs2::error & e)
{
    std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    "  << e.what() << std::endl;
    return EXIT_FAILURE;
}

catch (const std::exception& e)
{
    std::cerr << e.what() <<std::endl;
    return EXIT_FAILURE;
}
```

[image]

drfg0119, 2020-04-21:
Hi, I am just thinking how to fix the device-conflicting problem, I have 2 idea like:

1. Find the device address under the realsense2_camera when I launch the rs_camera.launch, then 
    include it to my code.
2. Put my main( ) code under the base_realsense_node.cpp. 

But I don't know how to do it, or maybe I am wrong. Could you please help me? Thanks!

RealSenseCustomerSupport, 2020-04-29:
----------------------------------------------

Maybe this can help and as a reference, the multi-cam example, the device serial number was used to avoid conflicts.
https://github.com/IntelRealSense/realsense-ros#work-with-multiple-cameras

drfg0119, 2020-04-30:
Hi @RealSenseCustomerSupport , I didn't use muti-cam in my project, I launch nodes both in **one** device, that shouldn't be a conflicting issue here... Thank you for your attention.

BTW, I'm trying another way to write my auto&dynamic-setting-parameter-controller haha, I want to focus on how to write a client node because there is a service called "**set_parameters**" under ddynamic_reconfigure project.
The request condition is I need to give a **Config** input like the following link:
https://github.com/pal-robotics/ddynamic_reconfigure/blob/kinetic-devel/src/ddynamic_reconfigure.cpp#L20

Is there any method could help me?
I have no idea about where the **Config** in librealsense is, I want to find them then I can use the parameter.
I want to find **Config** under coded_light_depth_sensor, I want to write a method that could help me to find the definition(EX: value range, value maximum and minimum...) of the parameter(EX: motion_range).

Then, I can use the declared parameter do everything I want.
Just like the following code:
https://github.com/WelinLee/ROS_dynamic_reconfig/blob/master/ros_dynamic_test/src/dyn_client.cpp#L25

then I want to do a comparison between this one:
https://github.com/pal-robotics/ddynamic_reconfigure/blob/kinetic-devel/test/test_ddynamic_reconfigure.cpp#L43

Thank you.
If there is any idea could provide, please help me. Thanks

RealSenseSupport, 2020-05-20:
Just curious if there is any update about this case.

This issue will be closed in 7 days if there's no further questions. 
Thank you.

closed 2020-05-27 by RealSenseSupport