realsenseai/librealsense #6219: Where to add realsense2.dll in visual studio 2019

BarRae, 2020-04-10:
I am currently starting to work with our new d435i camera.

I want to write a custom program for it to use in a computer vision application.
However I am currently experiencing some difficulty setting up a project.
The problem is that I don't know where precisely I should add the realsense2.dll file.
With add, I mean where in the visual studio properties tab.
I am a beginner with visual studio.

Could someone help me with this?

MartyG-RealSense, 2020-04-10:
I hope that the advice in the link below will be helpful.

https://github.com/IntelRealSense/librealsense/issues/4277

BarRae, 2020-04-15:
This was helpfull. It is working now.

MartyG-RealSense, 2020-04-15:
Great news - thanks for the update!

BarRae, 2020-04-15:
Sadly I spoke too soon.

I tried copying my project, something went wrong and now both version don't work...
Last time it was solved by including the .dll wile in the debug folder.
That didn't work this time.

I don't know what is wrong now, while running the application I get this error code
[image]

and in the console i get this:
C:\Users\BRR\source\repos\Object detection\x64\Debug\Object detection.exe (process 12852) exited with code -2147483645.
Press any key to close this window . . .

Im sorry for reopening again

MartyG-RealSense, 2020-04-15:
No problem, don't worry.  There is a 'Getting Started' tutorial for integrating RealSense into your own projects that does not involve third-party dependencies.

https://github.com/zivsha/librealsense/tree/getting_started_example/examples/getting-started

closed 2020-04-20 by RealSenseCustomerSupport