realsenseai/librealsense #6221: csv to png color

eltonfernando, 2020-04-10:
I'm building a dataset to extract deep learning as an image with rs-convert -i * .bag -p -v
The __Depth_x.png image (color depth map) is ideal for training, but I would not like to save any data set (there are more than 30000 images). It seems perfectly possible to control __Depth_x.png from the csv file. Try to map the csv values to a 32-bit integer and then combine to place a color map (3 uint 8-bit) but it doesn't get the same result as rs-convert. The source code of the rs-converter script seems very confusing to me regarding this operation. How can I build a __Depth_x.png map using a csv?

MartyG-RealSense, 2020-04-11:
In recent times, RealSense users have been able to convert bags to PNGs with rs-convert but have had problems converting from bag to csv.  That problem is still present in rs-convert as far as I know.

A couple of years ago, @dorodnic the RealSense SDK Manager added the ability to dump depth and color frames to csv in the SDK's **rs-capture** example program.

https://github.com/IntelRealSense/librealsense/issues/1485#issuecomment-380157114

eltonfernando, 2020-04-11:
exactly, everything works fine with rs-convert. 

I would like to generate the png output from rs-convert from a csv.

I tried as follows
``
import numpy as np
csv=np.genfromtxt("my_img.csv",delimiter=",")
csv=np.interp(csv, (csv.min(), csv.max()), (0,2**32))
csv=np.around(csv)
r0 = np.zeros(csv.shape, np.uint8)
r1 = np.zeros(csv.shape, np.uint8)
r2 = np.zeros(csv.shape, np.uint8)
r3 = np.zeros(csvl.shape, np.uint8)
mask=np.array(csv,np.uint32)
r0=mask&255
r1 = (mask>>8) & 255
r2 = (mask>>16) & 255
r3 = (mask>>24) & 255
r=r0-r2
g=r1-r3
b=r3

img=np.zeros((480,848,3), np.uint8)
img[:,:,0]=r
 img[:, :, 1] = g
 img[:, :, 2] = b
``
This is what I expected to get (this is a png generated by rs-convert)

[image]

this is the result I got with this script

[image]

it seems that something is missing, I think rs-convert should use some filter, but I don't see any edge deviation. Maybe there is a problem with my script. don't really know what might be going on

MartyG-RealSense, 2020-04-12:
I have seen a result resembling the output image from your script before compared to a very different output from rs-convert.  I recalled the difference being discussed by users and tracked it down.  The difference was put down to rs-convert comprising the depth data, whereas the other main convertor program available (bag2image) uses the 'real' depth data.

https://github.com/IntelRealSense/librealsense/issues/3208#issuecomment-466146575

MartyG-RealSense, 2020-04-26:
This case will be closed after 7 days from the time of writing this message if there are no further comments.  Thank you.

closed 2020-05-11 by MartyG-RealSense