realsenseai/librealsense #6222: hex color codes in the depth image.

arothenberg, 2020-04-10:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |   D435  | 
| Firmware Version                   | 05.12.02.100 | 
| Operating System & Version |   Win (10)  | 
| Platform                                 | PC |
| SDK Version                            | 2.32                      | 
| Language                            |  C++                      | 
| Segment			|desktop app  | 

### Issue Description
I  converted the rsalignadvanced.cpp code so that the slider works with the depth map instead of the rgb image. 
It works fine except for one problem,  the solution of which is probably obvious, but I'm stuck.

in the   **RemoveBackgound** function  _ std::memset(&p_other_frame[offset], 0x99, other_bpp);_
changes the color of a pixel to grey . the "0X99' value is read as hex 999999 - a grey tone.

I've changed the code to make it work for the depth table and the memset line now uses the depthdata
                _std::memset(&x_depth_frame[offsetx], 0x99, depth_bpp);_
It works but 0x99 translates to 990000 which is red (not the grey above).

So one difference is the 
other_frame (rgb) is 3 bytes per pixel while the depth_frame(the depth frame aligned to the other_frame) is 2 bytes per pixel.

Could someone break down how color is stored in the aligned depth image?

sam598, 2020-04-10:
In the RealSense SDK depth maps are a single channel image where each pixel is a 16 bit unsigned integer. That is why you are seeing 2 bytes per pixels. Each unit is one mm in the range from 0 to 65,535.

In some demos the depth map is encoded using a color ramp so that there is more contrast, and the change in depth is more visible to the human eye. This is just for visualization. When you are actually processing depth data you probably don't want to work with depth data that was, as a linear depth data is much more useful.

arothenberg, 2020-04-13:
Thanks . A very detailed informative response.
 "When you are actually processing depth data you probably don't want to work with depth data that **was**, as a linear depth data is much more useful."
Could you restate this. I'm missing something. Isn't it linear here?

I just noticed that the code casts the depth_frame.getdata() as unit8 data before the memset call. That  might be why I am stuck in red tones. I'll try and take away the cast. 
Actually why do they recast it to unit8 in the code? 

I'd like to find the fastest way to analyze real time data and display a simple representation.  I'd like to avoid aligning and creating copies...

ev-mp, 2020-04-13:
@arothenberg hello, 
The behavior you observe with textured depth is the direct result of the code you posted - 
 `std::memset(&x_depth_frame[offsetx], 0x99, depth_bpp);`. 
The code indeed produces red color, but for a completely different reason:

1. The Depth frame holds raw depth values in [0-65k] range that linearly translate to [0-N] meters according to the selected `depth_units` value.
2 By using the above code you **override the raw depth value** with 0x9999 or 39321 in decimal base, which corresponds to 39.321 meters with default `depth_units`.
3. The single-channel/16bit values cannot be visualized directly using three-channel/8bit RGB monitor, so for display rendering the raw Depth data is being first `colorized` with one of the supported color conversion schemes (Matlab's `jet` being the default) as @sam598 points above. 
4. Finally, the ~40 meter depth value becomes a red pixel when jet colors are applied.

The `background removal` filter in that demo relies on the depth data and modifies the `other` texture, being it RGB or IR image. The Depth and the `other` frames cannot be swapped as they are of different types . In case you want to preserve the range-based filter functionality then you need to review the changes you've made and probably revert to the original code.

arothenberg, 2020-04-16:
I see. That makes sense. I was way off.

I did change the slider to apply to the depth map. It works perfectly. Same exact distances as the video image.  The exercise was worth it.  My only problem was my confusion about the "color" value which wasn't actually a color value but the depth value which in retrospect should have been obvious.

closed 2020-04-16 by arothenberg

arothenberg, 2020-04-16:
Thanks a ton for the help but why would I revert to the original code? That would defeat the whole point. lol.

arothenberg, 2020-04-16:
Also, thanks for 
"By using the above code you override the raw depth value with 0x9999 or 39321 in decimal base, which corresponds to 39.321 meters with default depth_units." 
I'm glad you added this.