realsenseai/librealsense #6224: Tried to create triangulation mesh using greedy triangulation algoritham

thambimani, 2020-04-11:
#include <pcl/point_types.h>
#include <pcl/io/pcd_io.h>
#include <pcl/kdtree/kdtree_flann.h>
#include <pcl/features/normal_3d.h>
#include <pcl/surface/gp3.h>
#include <pcl/io/vtk_io.h>
#include <pcl/io/ifs_io.h>

honestly  I collect code from internet only 

#include <pcl/io/ply_io.h>
#include <pcl/io/obj_io.h>
int main (int argc, char** argv)
{
  // Load input file into a PointCloud<T> with an appropriate type
  pcl::PointCloud<pcl::PointXYZRGB>::Ptr cloud (new pcl::PointCloud<pcl::PointXYZRGB>);
  pcl::PCLPointCloud2 cloud_blob;
  pcl::io::loadPLYFile ("input.ply", cloud_blob);
  pcl::fromPCLPointCloud2 (cloud_blob, *cloud);
  //* the data should be available in cloud

  // Normal estimation*
 pcl::NormalEstimation<pcl::PointXYZRGB, pcl::Normal> n;
    pcl::PointCloud<pcl::Normal>::Ptr normals(new pcl::PointCloud<pcl::Normal>);
    pcl::search::KdTree<pcl::PointXYZRGB>::Ptr tree(new pcl::search::KdTree<pcl::PointXYZRGB>);
    tree->setInputCloud(cloud);
    n.setInputCloud(cloud);
    n.setSearchMethod(tree);
    n.setKSearch(20); // 0.1 -> 10cm
  //  n.setRadiusSearch(5); // 0.1 -> 10cm
    n.compute(*normals);
   

 pcl::PointCloud<pcl::PointXYZRGBNormal>::Ptr cloud_with_normals(new pcl::PointCloud<pcl::PointXYZRGBNormal>);
    pcl::concatenateFields(*cloud, *normals, *cloud_with_normals);

    pcl::search::KdTree<pcl::PointXYZRGBNormal>::Ptr tree2(new pcl::search::KdTree<pcl::PointXYZRGBNormal>);
    tree2->setInputCloud(cloud_with_normals);
    //Using Greedy ProjectionTriangulation
    pcl::GreedyProjectionTriangulation<pcl::PointXYZRGBNormal> gp3;
    pcl::PolygonMesh triangles;

    gp3.setSearchRadius(0.5); // Triangle Size
    gp3.setMu(2.5); // 2.5 ~ 3.0
    gp3.setMaximumNearestNeighbors(150); // 100 ~ 200
    gp3.setMaximumSurfaceAngle(M_PI/4); // 45 degrees
    gp3.setMinimumAngle(M_PI / 18); // 10 degrees
    gp3.setMaximumAngle(2 * M_PI / 3); // 120 degrees
    gp3.setNormalConsistency(true);
    //gp3.setConsistentVertexOrdering(true);
    gp3.setInputCloud(cloud_with_normals);
    gp3.setSearchMethod(tree2);
    gp3.reconstruct(triangles);
  // Additional vertex information
 // std::vector<int> parts = gp3.getPartIDs();
  //std::vector<int> states = gp3.getPointStates();

  pcl::io::saveVTKFile("mesh.vtk",triangles,5);

  // Finish
  return (0);
}
output to file is created but when I going to be zoomed in I never noticed any  triangulation and cloud points then my mesh and its look like a single surface 

I don't know what is the mistake I was made 

here is my output 
[image]

knowing people pls come forward 
and  thanks in advance 
cheers,
Thambi mani

yohanb78, 2020-04-11:
I don't find what is wrong., it should work
Try other values here 
 // Set the maximum distance between connected points (maximum edge length)
    gp3.setSearchRadius (0.025);

Try to save in ply or obj rather than vtk, to be sure...

thambimani, 2020-04-11:
> I don't find what is wrong., it should work
> Try other values here
> // Set the maximum distance between connected points (maximum edge length)
> gp3.setSearchRadius (0.025);
> 
[image]
[image]

> Try to save in ply or obj rather than vtk, to be sure...

[image]
this is for .ply f
[image]
ile here also i nerver noticed triangulation

yohanb78, 2020-04-11:
could I see the file.ply ?
There is a difference, because if you downsized the searchradius it could explain in the screenshot of the .ply that there is a lot of holes where the triangulation didn't occur ?

thambimani, 2020-04-15:
> could I see the file.ply ?
> There is a difference, because if you downsized the searchradius it could explain in the screenshot of the .ply that there is a lot of holes where the triangulation didn't occur ?

what mistake I made I don't know

RealSenseCustomerSupport, 2020-04-29:
----------------------------------------------

Is it alright to close this issue as the similar topic was discussed and closed in #6225?
Thank you.

This case will be closed after 7 days if there is no further comments.

closed 2020-05-07 by RealSenseCustomerSupport