realsenseai/librealsense #6225: how to create  mesh in  delaunay triangulation  using pcl library

thambimani, 2020-04-11:
I search a lot on the internet but unable to find a suitable solution for this.
my input file is input.ply 

knowing persons come forward  with code
thanks in advance

MartyG-RealSense, 2020-04-11:
I researched carefully but could not find a better guide to doing Delaunay triangulation with PCL than the one that you had already linked to.

If it is possible to consider alternatives to the PCL approach, the open source point cloud software **CloudCompare** has a built-in function to carry out the Delaunay process to create a mesh, either as an XY Plane or a Best Fit Plane.  Page 5 of the guide in the link below illustrates the function.

http://kastner.ucsd.edu/ryan/wp-content/uploads/sites/5/2014/03/admin/palissy.pdf

 CloudCompare's documentation also describes the functions:

**XY Plane**
https://www.cloudcompare.org/doc/wiki/index.php?title=Mesh%5CDelaunay_2.5D_(XY_plane)

**Best Fit Plane**
http://www.cloudcompare.org/doc/wiki/index.php?title=Mesh%5CDelaunay_2.5D_(best_fit_plane)

yohanb78, 2020-04-11:
Hello,

I passed a long time the previous week to test differents triangulations with PCL library.

Greedy projection has a serious problem in PCL because it randomly inverse the normal vectors in its reconstruction.

It depends of the source, the kind of object you whould like to recontruct.
The fast reconstruction when you get the frame from an RSE::Pointcloud is the organizedFastMesh.
I tried Greedy, Poisson, MLS, bSpline (very slow but very nice result) reconstructions, Poisson is interesting for reconstruct closed meshs, is fast ans easy to do.

1. Convert your points to PCL pointCloud
```

pcl::PointCloud<pcl::PointXYZ>::Ptr points_to_pcl(const rs2::points& points,{

    pcl::PointCloud<pcl::PointXYZ>::Ptr cloud(new pcl::PointCloud<pcl::PointXYZ>);
    // Config of PCL Cloud object
    cloud->width = width;
    cloud->height = height;
    cloud->is_dense = false;
    cloud->points.resize(points.size());

    auto vertices = points.get_vertices();
    // Iterating through all points and setting XYZ coordinates
    for (size_t i = 0; i < points.size(); ++i)
    {
        cloud->points[i].x = vertices[i].x;
        cloud->points[i].y = vertices[i].y;
        cloud->points[i].z = vertices[i].z;
    }
   return cloud;
}
```
2. Fast Mesh, you can not apply PCL filter otherwise you obtain a point cloud not "organized" and can not use it:

```
pcl::OrganizedFastMesh<pcl::PointXYZ>::Ptr orgMesh (new pcl::OrganizedFastMesh<pcl::PointXYZ>());
    pcl::PolygonMesh::Ptr triangles (new pcl::PolygonMesh);
    //pcl::PolygonMesh triangles ;
    // orgMesh.useDepthAsDistance(true);
    orgMesh->setTriangulationType (pcl::OrganizedFastMesh<pcl::PointXYZ>::TRIANGLE_RIGHT_CUT  );
    orgMesh->setInputCloud(cloud);
    orgMesh->reconstruct(*triangles);
    qDebug("%i", triangles->polygons.size());
    pcl::io::saveOBJFile("D:/filtered.obj", *triangles);
```
If interested, I can add the code for the normal estimation needed to do poisson or greedy or other reconstruction.

On my side, the goal was to have an .obj with the textures. I did it on myself with some for loop and vector for the triangulation :)

Regards.

MartyG-RealSense, 2020-04-11:
@yohanb78   Thanks so much for your help on this case!    :)

thambimani, 2020-04-20:
when we do triangular meshing how to first triangulation happen in Delaunay

thambimani, 2020-04-20:
" a triangle whose circumradius-to-shortest edge ratio is greater than b is said to be skinny"
what is mean??

in Delaunay triangulation how to I find skinny elements

MartyG-RealSense, 2020-04-20:
@thambimani   Wikipedia defines skinny triangles as triangles whose height is much greater than their base.

https://en.wikipedia.org/wiki/Skinny_triangle

My understanding from reading about Delaunay trinagulation is that skinny triangles are something that is desired to be avoided, and that they do not occur often in Delaunay triangulation because the Delaunay process maximizes the minimum angle of all the angles of the triangles in the triangulation.

So you should not need to find skinny elements because they should rarely coccur in Delaunay triangulation.

thambimani, 2020-04-20:
> @thambimani Wikipedia defines skinny triangles as triangles whose height is much greater than their base.
> 
> https://en.wikipedia.org/wiki/Skinny_triangle
> 
> My understanding from reading about Delaunay triangulation is that skinny triangles do not occur often in Delaunay triangulation because the Delaunay process maximizes the minimum angle of all the angles of the triangles in the triangulation.
> 
> So you should not need to find skinny triangles because they should rarely occur in Delaunay triangulation.
[image]

in this picture, I do not find any skinny element as per wiki info.
do you see any skinny element  if you see anyone pls tell me which one and which reason

thambimani, 2020-04-20:
here I had another doubt what is mean by domain edge

MartyG-RealSense, 2020-04-20:
I did careful research of your question about domain edge.  I could not find a single clear answer.  My research suggests though that the 'domain' refers to the overall mesh, and 'domain edge' is the edges of the mesh (which is different from the 'domain face', the surface inside of the edge lines.

thambimani, 2020-04-20:
> I search a lot on the internet but unable to find a suitable solution for this.
> my input file is input.ply
> 
> already I tried http://www.pointclouds.org/documentation/tutorials/greedy_projection.php
> this program but I cannot get the expected output.
> 
> so forward to Delaunay triangulation but I never noticed any PCL code for this.
> 
> now I searching the Delaunay triangulation code.
> 
> knowing persons come forward with code
> thanks in advance

if anyone give code solution for Delaunay triangulation it is helpful for me

MartyG-RealSense, 2020-04-20:
I did further research.  A C++ library called CGAL can do Delaunay triangulation.

https://www.cgal.org/
https://doc.cgal.org/latest/Manual/packages.html#PkgTriangulation3

thambimani, 2020-04-21:
> Hello,
> 
> I passed a long time the previous week to test differents triangulations with PCL library.
> 
> Greedy projection has a serious problem in PCL because it randomly inverse the normal vectors in its reconstruction.
> 
> It depends of the source, the kind of object you whould like to recontruct.
> The fast reconstruction when you get the frame from an RSE::Pointcloud is the organizedFastMesh.
> I tried Greedy, Poisson, MLS, bSpline (very slow but very nice result) reconstructions, Poisson is interesting for reconstruct closed meshs, is fast ans easy to do.
> 
>     1. Convert your points to PCL pointCloud
> 
> 
> ```
> 
> pcl::PointCloud<pcl::PointXYZ>::Ptr points_to_pcl(const rs2::points& points,{
> 
>     pcl::PointCloud<pcl::PointXYZ>::Ptr cloud(new pcl::PointCloud<pcl::PointXYZ>);
>     // Config of PCL Cloud object
>     cloud->width = width;
>     cloud->height = height;
>     cloud->is_dense = false;
>     cloud->points.resize(points.size());
> 
>     auto vertices = points.get_vertices();
>     // Iterating through all points and setting XYZ coordinates
>     for (size_t i = 0; i < points.size(); ++i)
>     {
>         cloud->points[i].x = vertices[i].x;
>         cloud->points[i].y = vertices[i].y;
>         cloud->points[i].z = vertices[i].z;
>     }
>    return cloud;
> }
> ```
> 
>     1. Fast Mesh, you can not apply PCL filter otherwise you obtain a point cloud not "organized" and can not use it:
> 
> 
> ```
> pcl::OrganizedFastMesh<pcl::PointXYZ>::Ptr orgMesh (new pcl::OrganizedFastMesh<pcl::PointXYZ>());
>     pcl::PolygonMesh::Ptr triangles (new pcl::PolygonMesh);
>     //pcl::PolygonMesh triangles ;
>     // orgMesh.useDepthAsDistance(true);
>     orgMesh->setTriangulationType (pcl::OrganizedFastMesh<pcl::PointXYZ>::TRIANGLE_RIGHT_CUT  );
>     orgMesh->setInputCloud(cloud);
>     orgMesh->reconstruct(*triangles);
>     qDebug("%i", triangles->polygons.size());
>     pcl::io::saveOBJFile("D:/filtered.obj", *triangles);
> ```
> 
> If interested, I can add the code for the normal estimation needed to do poisson or greedy or other reconstruction.
> 
> On my side, the goal was to have an .obj with the textures. I did it on myself with some for loop and vector for the triangulation :)
> 
> Regards.

@yohanb78 
pls can you ADD the Whole code for greedy triangulation that is more helpful for us
Thank you

thambimani, 2020-04-21:
#include <pcl/point_types.h>
#include <pcl/io/pcd_io.h>
#include <pcl/kdtree/kdtree_flann.h>
#include <pcl/features/normal_3d.h>
#include <pcl/surface/gp3.h>
#include <pcl/io/vtk_io.h>
#include <pcl/io/ifs_io.h>
#include <pcl/io/ply_io.h>
#include <pcl/io/obj_io.h>
int main (int argc, char** argv)
{
  // Load input file into a PointCloud<T> with an appropriate type
  pcl::PointCloud<pcl::PointXYZRGB>::Ptr cloud (new pcl::PointCloud<pcl::PointXYZRGB>);
  pcl::PCLPointCloud2 cloud_blob;
  pcl::io::loadPLYFile ("2.ply", cloud_blob);
  pcl::fromPCLPointCloud2 (cloud_blob, *cloud);
  //* the data should be available in cloud

  // Normal estimation*
 pcl::NormalEstimation<pcl::PointXYZRGB, pcl::Normal> n;
    pcl::PointCloud<pcl::Normal>::Ptr normals(new pcl::PointCloud<pcl::Normal>);
    pcl::search::KdTree<pcl::PointXYZRGB>::Ptr tree(new pcl::search::KdTree<pcl::PointXYZRGB>);
    tree->setInputCloud(cloud);
    n.setInputCloud(cloud);
    n.setSearchMethod(tree);
    n.setKSearch(20); // 0.1 -> 10cm
  //  n.setRadiusSearch(5); // 0.1 -> 10cm
    n.compute(*normals);
   

 pcl::PointCloud<pcl::PointXYZRGBNormal>::Ptr cloud_with_normals(new pcl::PointCloud<pcl::PointXYZRGBNormal>);
    pcl::concatenateFields(*cloud, *normals, *cloud_with_normals);

    pcl::search::KdTree<pcl::PointXYZRGBNormal>::Ptr tree2(new pcl::search::KdTree<pcl::PointXYZRGBNormal>);
    tree2->setInputCloud(cloud_with_normals);
    //Using Greedy ProjectionTriangulation
    pcl::GreedyProjectionTriangulation<pcl::PointXYZRGBNormal> gp3;
    pcl::PolygonMesh triangles;

    gp3.setSearchRadius(0.5); // Triangle Size
    gp3.setMu(2.5); // 2.5 ~ 3.0
    gp3.setMaximumNearestNeighbors(150); // 100 ~ 200
    gp3.setMaximumSurfaceAngle(M_PI/4); // 45 degrees
    gp3.setMinimumAngle(M_PI / 18); // 10 degrees
    gp3.setMaximumAngle(2 * M_PI / 3); // 120 degrees
    gp3.setNormalConsistency(true);
    //gp3.setConsistentVertexOrdering(true);
    gp3.setInputCloud(cloud_with_normals);
    gp3.setSearchMethod(tree2);
    gp3.reconstruct(triangles);
  // Additional vertex information
 // std::vector<int> parts = gp3.getPartIDs();
  //std::vector<int> states = gp3.getPointStates();
  pcl::io::savePLYFile("op.ply",triangles,5);
  pcl::io::saveOBJFile("op.obj",triangles,5);
  pcl::io::saveVTKFile("op.vtk",triangles,5);

  // Finish
  return (0);
}

@yohanb78 

in this code op.vtk file looks a single structure with RGB when I try to zoom I cannot find any triangulation.

[image]
when I zooming that output file 
[image]

 what I am wrong

MartyG-RealSense, 2020-04-21:
@thambimani  Your images have a red error message in the log at the bottom of the window.  In the one showing the full hand, it says:

"Mesh has no normal!  You can manually compute them (select it then call "Edit > Normals > Compute)"

thambimani, 2020-04-21:
> @thambimani Your images have a red error message in the log at the bottom of the window. It says:
> 
> "Mesh has no normal! You can manually compute them (select it then call "Edit > Normals > Compute)"

@MartyG-RealSense  
give solution for my problem or  direct me in the correct way

MartyG-RealSense, 2020-04-21:
Please try going to the menu of your program and selecting "Edit > Normals > Compute" to manually compute normals, and see if that improves your image.

yohanb78, 2020-04-21:
Please attach the .obj, you can rename it suffixed with .txt. Otherwise you can open it and tell if you can find lines which begins with f (for faces/triangles), you shoud have too lines begining with v and vn (if normals have been computed).
I don't know cloudcompare software and can not be sure that there is a specific vizualization of the triangles whereas it may be exists.

Your code is ok, the differences I see with mine, are:
*  the conversion to a pcl point cloud because I use my own.
* this way for normal computation, I don't remember why I have replaced NormaEstimation by NormalEstimationOMP class:
```
 pcl::NormalEstimationOMP<pcl::PointXYZRGB, pcl::Normal> n;
    pcl::PointCloud<pcl::Normal>::Ptr normals (new pcl::PointCloud<pcl::Normal>);
    pcl::search::KdTree<pcl::PointXYZRGB>::Ptr tree (new pcl::search::KdTree<pcl::PointXYZRGB>);
    tree->setInputCloud (cloud);
    n.setInputCloud (cloud);
    n.setSearchMethod (tree);
    //n.setRadiusSearch(0.1);
    n.setViewPoint(0, 0, -1);
    n.setKSearch (20);
    n.compute (*normals);
```

I'm not sure but if normals have not been well computed, you will have no triangulation.
And a console log after computation to display normals->points.size()

Regards

thambimani, 2020-04-21:
> Please try going to the menu of your program and selecting "Edit > Normals > Compute" to manually compute normals, and see if that improves your image 

Please try going to the menu of your program and selecting "Edit > Normals > Compute" to manually compute normals, and see if that improves your image.

I have not expected this type of output

[image]
[image]

this is not my expectation
 http://www.pointclouds.org/documentation/tutorials/greedy_projection.php
this link contains one video that is what my expectation

guide me what changes I make in my code

MartyG-RealSense, 2020-04-21:
@thambimani   Please remember that it is not the responsibility of Librealsense to provide support for PCL programming questions.  It is a kindness, not something that has to be done.

@yohanb78   Thanks so much again for your patient answering!

thambimani, 2020-04-21:
> Please attach the .obj, you can rename it suffixed with .txt. Otherwise you can open it and tell if you can find lines which begins with f (for faces/triangles), you shoud have too lines begining [...]
@yohanb78 
Thank you somuch friend

yohanb78, 2020-04-21:
No pb. 
As i already said. There is better to do with realsense post processing. There is food tuto about it this. For your hand. Start with décimation and space filters ans Alonso disparities. Save in. ply, import it in pcl and work hard until your goal.  Try opencv if you want to colorize as you wish your vertices.
I cant help further. It is an other level of skills

thambimani, 2020-04-21:
> No pb.
> As i already said. There is better to do with realsense post processing. There is good tuto about this. For your hand. Start with décimation and space filters ans Alonso disparities. Save in. ply, import it in pcl and work hard until your goal. Try opencv if you want to colorize as you wish your vertices.
> I cant help further. It is an other level of skills

I can thank for your supporting

closed 2020-04-24 by thambimani

Amayess, 2022-08-31:
> Hello,
>
> I passed a long time the previous week to test differents triangulations with PCL library.
>
> Greedy projection has a serious problem in PCL because it randomly inverse the normal vectors in [...]

Hi @yohanb78 

I was able to try the Fast Organized Mesh method to generate a mesh in OBJ format. I get a result but it only contains vertices and no faces.

Do you have an explanation for this please? 

How to make the method generate faces too? 

Thanks for your help.