realsenseai/librealsense #6227: Gravity Vector on T265

bobdavies2000, 2020-04-12:
I am running 2.34.0 version of librealsense under Windows 10 (latest release) and the acceleration vector looks incorrect on the T265.  It is correct on the D400 series of cameras.  To see the problem, use the rs-pose-predict app in the librealsense distribution and modify the following line:

std::cout << "Predicted " << std::fixed << std::setprecision(3) << dt_s*1000 << "ms " <<
"Confidence: " << pose_data.tracker_confidence << " T: " <<
predicted_pose.translation.x << " " <<
predicted_pose.translation.y << " " <<
predicted_pose.translation.z << " (meters) \r";

and change "translation" to "acceleration".  All 3 values are near zero on my T265 which is incorrect.  If the camera is vertical, the y value should be close to 9.8 - earth's gravitational constant.

In poking around in the pose data, I can see the 9.8 value so this appears to be a structure alignment issue.

Bob Davies

ev-mp, 2020-04-13:
@bobdavies2000 hello, the "acceleration" attribute of the "pose" sensor denotes the sensor's velocity derivative ( dv/dt), not the gravitational/inertial forces.

T265 sensor provides 5 different streams types:  Pose, Accelerometer, Gyro, Fisheye Left, Fisheye Right.   
To find the forces applied to the sensor's body try the **Accelerometer** stream values.

[image]

bobdavies2000, 2020-04-13:
Thanks for clearing that up and I can easily fix my code.  It does stimulate a question: why would you allow the pose data to contain an acceleration variable when that variable does not contain acceleration but velocity?  Thanks again for your response.

closed 2020-04-13 by bobdavies2000

ev-mp, 2020-04-13:
The pose struct contains both the linear velocity and linear acceleration values . Actually it contains 21 different parameters.  See [the documentation](https://intelrealsense.github.io/librealsense/doxygen/structrs2__pose.html) for the full list of attributes.