realsenseai/librealsense #6228: T265 and D435i in matlab

hojunson, 2020-04-12:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10)   | 
| Kernel Version (Linux Only)    |  most recent                                    | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  matlab}                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I want to run D435i and T265 cams in matlab simultaneously. 

===========================
for dev = ctx.query_devices() % Loop through all devices detected
    pipe = realsense.pipeline(); 
    SN = dev{1}.get_info(realsense.camera_info.serial_number)
    cfg = realsense.config();
    cfg.enable_device(SN);
    pipe.start(cfg);

end
=============
I have USB 3.0 and 2.0 port connected to D435i and T265 perspectively.
But 
I got the error below.
[image]

I checked both cameras work well separately.

Can anyone help me?

JimmyFroost, 2020-06-19:
you should try to connect T265 first, then D435.
I think I had same problem with D415 taking over, as if there was no T265.

RealSenseSupport, 2020-12-05:
Hi,

Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

hojunson, 2020-12-05:
No Thanks.

closed 2020-12-11 by RealSenseSupport