realsenseai/librealsense #6229: Looking for detail instructions for Peter Schmidt's bot

RoboBil, 2020-04-12:
I'm looking for any detailed information of the bot shown at 17:40 of the YouTube video of [Peter Schmidt's Feb 13, 2019 presentation "Visual navigation for wheeled autonomous robots – using Intel® RealSense™ Tracking Camera T265."](https://www.youtube.com/watch?v=62vm0_RZ1nU&t=1174s)

 

I'm a real beginner and know very little about programming. I have recently purchased a T265. I would like to build a smaller version of his bot.

 

Thanks

 

RoboBill

neilyoung, 2020-04-12:
It doesn't necessarily be such a big boy

https://www.youtube.com/watch?v=riBsZoSrcpI&feature=youtu.be

Details: https://neilyoung.serveblog.net/

Regards

RoboBil, 2020-04-13:
Your website/video is very helpful. If I understand your project correctly, it is a system that displays the Cartesian coordinates of your remote controlled robot's position. In the video you mentioned no 
 self navigation yet. Have you begun to work on that? 

Thanks

neilyoung, 2020-04-13:
No, not yet. This is a project that I only do in my spare time and at the moment I am so busy at work that there is no time for that. In fact, knowing your position is a prerequisite for successful navigation. Surely there are still missing fusions, which can be achieved e.g. by combination with distance sensors (ultrasonic) or Depth Cams (D435). At the moment I am satisfied with the level reached at first. Maybe it helps others.

Thanks

closed 2021-03-09 by RealSenseSupport