realsenseai/librealsense #6230: A-factor does not improve depth accuracy and depth_units gives inconsistent results (tested under python wrapper)

szx0112, 2020-04-13:
In the last few days, I try to improve the depth accuracy as far as I can by playing with the following four parameters: 
(1) a-factor [https://dev.intelrealsense.com/docs/white-paper-subpixel-linearity-improvement-for-intel-realsense-depth-cameras](https://dev.intelrealsense.com/docs/white-paper-subpixel-linearity-improvement-for-intel-realsense-depth-cameras)

(2)  depth units (1 mm or 0.1 mm)

(3) visual preset [https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets](https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets)

(4) post_processing [https://dev.intelrealsense.com/docs/depth-post-processing](https://dev.intelrealsense.com/docs/depth-post-processing)

I test those params by collecting the center point of the consecutive 500 depth frames with camera points to a flat wall at fixed distance (~2.3 meters). Resolution is fixed at 848x480 as recommended by the white paper. The mean and standard deviation are computed to test the accuracy of the collected depth values under different settings.

Default setting is set as reference (depth_unit=0.001, post_processing=disabled, a-factor=0, visual_preset=disabled)
**mean: 2372.87 (mm)
std: 26.029 (mm)**

_(1) Enable visual Presets:_
a) With highAccuracy preset enabled (other params keep the same as default)
    **mean: 2375.042
    std: 23.296**

b) With highDensity preset enabled (other params keep the same as default)
    **mean: 2373.838
    std:  23.660**

_(2) Enable a-factor:_
a) a-factor=0.08 with high density preset enabled (other params keep the same as default)
     **mean: 2385.644
     std: 35.359**

b) a-factpr=0.18 with high accuracy preset enabled  (other params keep the same as default)
    **mean: 2408.772
    std: 43.127**

_(3) Enabled Post-processing:_
    spatial + temporal filtering (other params keep the same as default)
    **mean: 2379.518
    std: 12.427**

_(4) Depth_units change to 0.0001_
     **mean: 23747.014 (1/10 mm)
     std: 237.0574  (1/10 mm)**

Although the specific values (mean and std) may not exactly accurate as I just select one result for each setting. However, we can still tentative conclude that:
(1) post-processing can significantly reduce the std compared to other listed params
(2) visual_preset also reduce the std but the performance is non-significant compared to post-processing
(3) a-factor shows even worse results (which confuse me a lot because the result is totally different from the discussion provided in the document)
(4) depth_units has no effect on the accuracy, but it may increase the precision. Also I notice when changes the depth_units to 0.0001, sometime the result is in 1mm (23xx.xxx) while sometime the result is in 1/10 mm (23xxx.xxx).

Currently, I only combine the post-processing with the high accuracy preset as (1) a-factor does not help the accuracy improvement (2) with depth_units=0.0001, it sometime provides values in 1mm while sometime in 0.1mm. 

While I am still searching for other solutions that can further improves the depth accuracy. Any comment or suggestion is really appreciated.

RealSenseCustomerSupport, 2020-04-13:
----------------------------------------------

Hi szx0112,
since you measure depth on a plain white wall, that means the texture camera can see mainly the projected laser dots..

you can refer to another white paper, see if i such as ncreasing the laser power can help
by the way,
the RMS error increase with measured distance
and D415 has  lower depth RMS error than D435 at same distance
which also mentioned in this white paper
https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance

szx0112, 2020-04-14:
Hi @RealSenseCustomerSupport ,

We still prefer the D435 camera as the camera need to be mounted on a moving vehicle. 

I observed that increases laser power indeed reduce the RMS error. Thank you for the suggestion.

However, weird performance shows when both visual_preset is loaded and set the depth_units to 0.0001. The sensor reading become incorrect, also the we notice the changed depth sensing range. Here, I show an example (the depth sensing range is clipped up to 1 meter):

Only change depth_units from 0.01 to 0.001 (sometime need re-streaming the camera to show the correct result) Depth reading at the middle of the image: ~0.7 meter
[image]

Only load visual_preset (default.json used in this case) Depth reading at the middle of the image: ~0.7 meter
[image]

Both load visual preset and change depth_units to 0.0001, the sensor reading is incorrect (Depth reading at the middle of the image is ~0.03) and also notice depth sensing range is different, the depth image can see the objects further than 1 meter and cannot senses objects closer than the target pipe. 
[image]

To help reproduce the problem, I attached the code below. Same results show both on a D435 and a D435i camera. The code is modified based on the librealsense python example [align-depth2color.py](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py).
```
## License: Apache 2.0. See LICENSE file in root directory.
## Copyright(c) 2017 Intel Corporation. All Rights Reserved.

#####################################################
##              Align Depth to Color               ##
#####################################################

# First import the library
import pyrealsense2 as rs
# Import Numpy for easy array manipulation
import numpy as np
# Import OpenCV for easy image rendering
import cv2
import json

json_preset_path = './default.json'                         # path of visual preset
with open(json_preset_path) as f:
    json_obj = json.load(f)

json_str = str(json_obj).replace("'", '\"')

# when multiple devices
ctx = rs.context()                                                
devices = ctx.query_devices()                                     
print('devices: ', devices)
advance_mode = rs.rs400_advanced_mode(devices[0])
advance_mode.load_json(json_str)                       # comment this line to enable/disable visual_preset

# Create a pipeline
pipeline = rs.pipeline()

# Create a config and configure the pipeline to stream
# different resolutions of color and depth streams
config = rs.config()
config.enable_stream(rs.stream.depth, 848, 480, rs.format.z16, 15)
config.enable_stream(rs.stream.color, 848, 480, rs.format.bgr8, 15)

# Start streaming
profile = pipeline.start(config)

# change depth units
depth_sensor = profile.get_device().first_depth_sensor()
depth_sensor.set_option(rs.option.depth_units, 0.0001)                # change depth_units from 0.001 to 0.0001

# Getting the depth sensor's depth scale (see rs-align example for explanation)
depth_sensor = profile.get_device().first_depth_sensor()
depth_scale = depth_sensor.get_depth_scale()
print("Depth Scale is: " , depth_scale)

# We will be removing the background of objects more than
#  clipping_distance_in_meters meters away
clipping_distance_in_meters = 1  #1 meter
clipping_distance = clipping_distance_in_meters / depth_scale

# Create an align object
# rs.align allows us to perform alignment of depth frames to others frames
# The "align_to" is the stream type to which we plan to align depth frames.
align_to = rs.stream.color
align = rs.align(align_to)

# Streaming loop
try:
    while True:
        # Get frameset of color and depth
        frames = pipeline.wait_for_frames()
        # frames.get_depth_frame() is a 640x360 depth image

        # Align the depth frame to color frame
        aligned_frames = align.process(frames)

        # Get aligned frames
        aligned_depth_frame = aligned_frames.get_depth_frame() # aligned_depth_frame is a 640x480 depth image
        color_frame = aligned_frames.get_color_frame()

        # Validate that both frames are valid
        if not aligned_depth_frame or not color_frame:
            continue

        depth_image = np.asanyarray(aligned_depth_frame.get_data())
        print(depth_image[424][240])
        print((depth_image[424][240]) * depth_scale)
        color_image = np.asanyarray(color_frame.get_data())

        # Remove background - Set pixels further than clipping_distance to grey
        grey_color = 153
        depth_image_3d = np.dstack((depth_image,depth_image,depth_image)) #depth image is 1 channel, color is 3 channels
        bg_removed = np.where((depth_image_3d > clipping_distance) | (depth_image_3d <= 0), grey_color, color_image)

        # Render images
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)
        images = np.hstack((color_image, bg_removed, depth_colormap))
        cv2.namedWindow('Align Example', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('Align Example', images)
        key = cv2.waitKey(1)
        # Press esc or 'q' to close the image window
        if key & 0xFF == ord('q') or key == 27:
            cv2.destroyAllWindows()
            break
finally:
    pipeline.stop()

```

RealSenseCustomerSupport, 2020-04-17:
----------------------------------------------

Hi  szx0112,
another approach is to modify the default.json file to have the depth unit setting you preferred as below, which change the unit to 0.001 to 0.0001

   "param-depthclampmax": "65536",
    "param-depthclampmin": "0",
    "param-depthunits": "100",   ##change from 1000 to 100 ##
    "param-disableraucolor": "0",
    "param-disablesadcolor": "0",
    "param-disablesadnormalize": "0",

You can use the viewer to save or load your customized configuration to json file, that you can compared different json file  to know what is changed

check the code you provided which is helpful, there is a chance that the firmware is not prepared/initialized completed for pipeline.start
based on the default.json (unit 0.001)
and the update depth unit request (unit 0.0001) is sent

if possible, would suggest to update the json file
second option is to add delay between pipeline.start and depth unit update, you can start with a big delay first to see if that really help

szx0112, 2020-04-17:
Thank you @RealSenseCustomerSupport Set delay after camera initialization really solved the problem. I am going to close this issue.

closed 2020-04-17 by szx0112