realsenseai/librealsense #6231: on-chip calibration via Intel realsense viewer v2.34.0" is not responding.

mingyangpeng, 2020-04-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {   D415 } | 
| Firmware Version                   | (05.12.03.00) | 
| Operating System & Version |   {  Linux (Ubuntu 18.04.4 LTS)  }  |  
| Platform                                 | PC   |
| SDK Version                            |  {2.34.0 }                          |  

### Issue Description
Intel realsense viewer :  When I try to use on-Chip Calibration(Very fast /fast/Medium/Slow/White Wall)/Tare Calibration, after a while,it says " Intel realsense viewer v2.34.0" is not responding. 

Output:
 types.h:307 " get_device_time_ms() took too long(more then 2mSecs)";
 synthetic-stream.cpp:48 "Exception was thrown during user processing callback";

MartyG-RealSense, 2020-04-13:
There is currently a known issue with the RealSense Viewer in 2.34.0 and the error that you quoted.

https://github.com/IntelRealSense/librealsense/issues/6189

dingbaiyin, 2020-04-28:
I am trying, but it is not finished yet. Don't know what to do next

MartyG-RealSense, 2020-04-28:
I just tested the On-Chip Calibration with 2.34.0 and experienced the same freezing problem (as the **get_device_time_ms** error in 2.34.0 is not yet fixed).  I tested On-Chip Calibration in the Viewer with the previous SDK version **2.33.1** and it worked perfectly.  So the only suggestion I can offer is to use SDK 2.33.1 for now if you need to perform an On-Chip Calibration.

dingbaiyin, 2020-04-29:
I need to use C and C ++ to implement the self-calibration function under the Linux system, but I do n’t know how to call the calibration function. I have been stuck for some time now, and I have not found any practical examples，Do you have any implementation examples ？Thank you .

MartyG-RealSense, 2020-04-29:
The appendices of Intel's white paper document on on-chip calibration provide code in C, C++ and Python for running the calibration.

**C**
https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras#section-appendix-b-more-detailed-on-chip-calibration-c-api-description

**Python**
https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras#section-appendix-c-on-chip-calibration-python-api

**C++**
https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras#section-appendix-d-on-chip-calibration-c-api

**LabVIEW**
https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras#section-appendix-e-self-calibration-with-lab-view

dingbaiyin, 2020-04-29:
Thank you very much .I have read the document .I hope I can realize the function

dingbaiyin, 2020-04-29:
I also encountered problems when using Intel.RealSense.Viewer-2.34.exe. When applying new calibration parameters, the software crashed

MartyG-RealSense, 2020-04-29:
The Viewer is having a freeze problem at the moment with on-chip calibration in 2.34.0.  Myself and another RealSense user have also experienced it.  I tested on-chip calibration with the previous SDK 2.33.1 yesterday and it worked perfectly, so my recommendation is to use 2.33.1 until the problems are fixed.

dingbaiyin, 2020-04-29:
Thank you, your suggestions are very valuable to me

songshan0321, 2020-05-11:
> The Viewer is having a freeze problem at the moment with on-chip calibration in 2.34.0. Myself and another RealSense user have also experienced it. I tested on-chip calibration with the previous SDK [...]

May I know how to downgrade realsense-viewer into 2.33.1?

MartyG-RealSense, 2020-05-11:
If you are using Windows, it is simple to download an earlier version of the RealSense Viewer.  Go to the 'Assets' file list at the bottom of the details for SDK 2.33.1 on the SDK 'Releases' page and click on the link titled 'Intel.RealSense.Viewer.exe' to download this version of the Viewer.  

https://github.com/IntelRealSense/librealsense/releases/tag/v2.33.1

[image]

You should launch the Viewer from the folder location that you downloaded it to, as the desktop shortcut icon will still point to the 2.34.0 version of the Viewer.

If you are using Linux though, I would think that you would have to download the source code for the 2.33.1 SDK from the asset list above and compile the entire 2.33.1 SDK to install the 2.33.1 Viewer.

closed 2020-05-30 by MartyG-RealSense