realsenseai/librealsense #6233: couldn't find  the corresponding function of rs2_deproject_pixel_to_point in Android

Ben-Lee-2019, 2020-04-14:
Hi, I recently used Realsense in Android projects, what is the corresponding function of 'rs2_deproject_pixel_to_point' and 'get_intrinsics' in Android?

MartyG-RealSense, 2020-04-14:
If the Android Librealsense project is using Java code then the example Librealsense Java script in the link below may be useful.

https://github.com/bytedeco/javacpp-presets/blob/master/librealsense/src/gen/java/org/bytedeco/librealsense/intrinsics.java

This script also contains code for deprojecting from image coordinates to 3D points.

https://github.com/bytedeco/javacpp-presets/blob/master/librealsense/src/gen/java/org/bytedeco/librealsense/intrinsics.java#L36

Ben-Lee-2019, 2020-04-14:
@MartyG-RealSense Thanks a Ton！

MartyG-RealSense, 2020-04-14:
@Ben-Lee-2019   You are very welcome.  I'm glad I could help!

MartyG-RealSense, 2020-05-11:
Case closed due to no further comments received.   Thanks!

closed 2020-05-11 by MartyG-RealSense