realsenseai/librealsense #6234: What are the ranges of texture coordinates

viisautta, 2020-04-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i                                                     | 
| Firmware Version                   | 5.12.3                                                      | 
| Operating System & Version |  Fedora 32                                              | 
| Kernel Version (Linux Only)    | 5.6.3                                                       | 
| Platform                                 | PC                                                          |
| SDK Version                           |  2.34.0                                                    | 
| Language                               |  C/opencv/pcl                                        | 
| Segment		        	|   others                                                   | 

### Issue Description
Normally, for texture coodinates, the range of U and V are from 0 to 1.
Using get_texture_coordinates(), the range of U and V are approximately from -0.2 to 1.2.
Could you provide the explanation to the range oftexture coodinates on realsense?

MartyG-RealSense, 2020-04-14:
A member of Intel created a script that retrieved the texture coordinates with get_texture_coordinates() and said that the range should be between 0 and 1.

https://forums.intel.com/s/question/0D70P0000068WnYSAU

The part of their script that mapped the 0-1 UV coordinates to the color frame was this:

int x = tex_coords[i].u * color_intrinsic.width;
int y = tex_coords[i].v * color_intrinsic.height;
int colorLocation = y * color_intrinsic.width + x;
unsigned char color[3] = {colorStream[colorLocation], colorStream[colorLocation +1], colorStream[colorLocation +2]};

ev-mp, 2020-04-14:
The Field of View (FOV) of the Depth sensor is bigger than the RGB sensor's, hence while the sensors overlap you can't have RGB data coverage on the boundaries of the depth frame.
The [U,V] outliers designate pixels for which the texture mapping occurs outside of the RGB sensor's FOV. #2355

ev-mp, 2020-04-20:
@viisautta , do you need more info to proceed? 
Please update

viisautta, 2020-04-26:
thanks for your detailed explaination

closed 2020-04-26 by viisautta