realsenseai/librealsense #6235: realsense-viewer crashes on UP2 board and restarts system

AndreasAZiegler, 2020-04-14:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version |   Linux (Ubuntu 18)  | 
| Kernel Version (Linux Only)    |  4.15.0)                                         | 
| Platform                                 | UP2 board  |
| SDK Version                            |  2.34.0                          | 
| Language                            |                            | 
| Segment			|  Robot                          | 

### Issue Description
When I start ```realsense-viewer``` and turn on the "RGB Camera" and the "Stereo Module", the whole system restarts and I get the following in the syslog.
```
Apr 14 14:54:08 UP2-board systemd[1]: Starting Firmware update daemon...
Apr 14 14:54:09 UP2-board dbus-daemon[723]: [system] Successfully activated service 'org.freedesktop.fwupd'
Apr 14 14:54:09 UP2-board systemd[1]: Started Firmware update daemon.
Apr 14 14:54:09 UP2-board org.gnome.Shell.desktop[1108]: Window manager warning: Buggy client sent a _NET_ACTIVE_WINDOW message with a timestamp of 0 for 0x2800015 (Intel Real)
Apr 14 14:54:10 UP2-board kernel: [   51.365538] ------------[ cut here ]------------
Apr 14 14:54:10 UP2-board kernel: [   51.365543] Unknown pixelformat 0x00000000
Apr 14 14:54:10 UP2-board kernel: [   51.365602] WARNING: CPU: 1 PID: 1766 at /build/linux-hwe-KDyd1s/linux-hwe-4.15.0/drivers/media/v4l2-core/v4l2-ioctl.c:1291 v4l_fill_fmtdesc+0xf41/0x1250 [videodev]
Apr 14 14:54:10 UP2-board kernel: [   51.365603] Modules linked in: snd_hda_codec_hdmi nls_iso8859_1 snd_soc_skl snd_soc_skl_ipc snd_hda_ext_core snd_soc_sst_dsp snd_soc_sst_ipc snd_soc_acpi snd_soc_core snd_compress ac97_bus snd_pcm_dmaengine intel_rapl spi_pxa2xx_platform intel_telemetry_pltdrv intel_punit_ipc intel_telemetry_core 8250_dw intel_pmc_ipc x86_pkg_temp_thermal coretemp kvm_intel kvm pinctrl_upboard irqbypass leds_upboard crct10dif_pclmul crc32_pclmul ghash_clmulni_intel pcbc aesni_intel aes_x86_64 crypto_simd glue_helper cryptd intel_cstate intel_rapl_perf snd_hda_intel snd_hda_codec snd_hda_core hid_sensor_accel_3d snd_hwdep snd_pcm hid_sensor_gyro_3d hid_sensor_trigger industrialio_triggered_buffer kfifo_buf pwm_lpss_pci hid_sensor_iio_common uvcvideo pwm_lpss lpc_ich snd_seq_midi i915 videobuf2_vmalloc snd_seq_midi_event
Apr 14 14:54:10 UP2-board kernel: [   51.365641]  snd_rawmidi videobuf2_memops videobuf2_v4l2 videobuf2_core snd_seq input_leds videodev media idma64 snd_seq_device virt_dma snd_timer drm_kms_helper intel_lpss_pci intel_lpss drm snd i2c_algo_bit mei_me fb_sys_fops shpchp mei syscopyarea soundcore sysfillrect sysimgblt ti_adc128s052 industrialio video mac_hid upboard_fpga sch_fq_codel parport_pc ppdev lp parport ip_tables x_tables autofs4 hid_generic hid_sensor_custom hid_sensor_hub usbhid hid mmc_block dwc3 ulpi udc_core sdhci_pci sdhci dwc3_pci r8169 ahci mii libahci pinctrl_broxton
Apr 14 14:54:10 UP2-board kernel: [   51.365688] CPU: 1 PID: 1766 Comm: realsense-viewe Not tainted 4.15.0-37-generic #40~upboard06-Ubuntu
Apr 14 14:54:10 UP2-board kernel: [   51.365689] Hardware name: AAEON UP-APL01/UP-APL01, BIOS UPA1AM42 03/12/2019
Apr 14 14:54:10 UP2-board kernel: [   51.365696] RIP: 0010:v4l_fill_fmtdesc+0xf41/0x1250 [videodev]
Apr 14 14:54:10 UP2-board kernel: [   51.365698] RSP: 0018:ffffae7281c1fc30 EFLAGS: 00010282
Apr 14 14:54:10 UP2-board kernel: [   51.365700] RAX: 0000000000000000 RBX: ffffae7281c1fd68 RCX: 0000000000000006
Apr 14 14:54:10 UP2-board kernel: [   51.365701] RDX: 0000000000000007 RSI: 0000000000000086 RDI: ffff919c7fc96490
Apr 14 14:54:10 UP2-board kernel: [   51.365702] RBP: ffffae7281c1fc40 R08: 0000000000000001 R09: 000000000000039d
Apr 14 14:54:10 UP2-board kernel: [   51.365703] R10: ffff919c6dbc7e00 R11: 0000000000000000 R12: ffff919c3a7f0b00
Apr 14 14:54:10 UP2-board kernel: [   51.365704] R13: ffffffffc0667900 R14: ffff919c6dbc7e40 R15: ffff919c3a7f0b00
Apr 14 14:54:10 UP2-board kernel: [   51.365706] FS:  00007fa4e3fff700(0000) GS:ffff919c7fc80000(0000) knlGS:0000000000000000
Apr 14 14:54:10 UP2-board kernel: [   51.365707] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Apr 14 14:54:10 UP2-board kernel: [   51.365708] CR2: 00007fa4dcb18048 CR3: 0000000232044000 CR4: 00000000003406e0
Apr 14 14:54:10 UP2-board kernel: [   51.365710] Call Trace:
Apr 14 14:54:10 UP2-board kernel: [   51.365720]  v4l_enum_fmt+0x74/0x110 [videodev]
Apr 14 14:54:10 UP2-board kernel: [   51.365727]  __video_do_ioctl+0x370/0x380 [videodev]
Apr 14 14:54:10 UP2-board kernel: [   51.365732]  ? terminate_walk+0x8e/0xf0
Apr 14 14:54:10 UP2-board kernel: [   51.365739]  video_usercopy+0x24d/0x640 [videodev]
Apr 14 14:54:10 UP2-board kernel: [   51.365745]  ? video_ioctl2+0x20/0x20 [videodev]
Apr 14 14:54:10 UP2-board kernel: [   51.365749]  ? security_file_alloc+0x68/0xa0
Apr 14 14:54:10 UP2-board kernel: [   51.365756]  video_ioctl2+0x15/0x20 [videodev]
Apr 14 14:54:10 UP2-board kernel: [   51.365762]  v4l2_ioctl+0xc8/0xf0 [videodev]
Apr 14 14:54:10 UP2-board kernel: [   51.365764]  do_vfs_ioctl+0xa8/0x630
Apr 14 14:54:10 UP2-board kernel: [   51.365768]  ? get_unused_fd_flags+0x30/0x40
Apr 14 14:54:10 UP2-board kernel: [   51.365770]  SyS_ioctl+0x79/0x90
Apr 14 14:54:10 UP2-board kernel: [   51.365773]  do_syscall_64+0x73/0x130
Apr 14 14:54:10 UP2-board kernel: [   51.365778]  entry_SYSCALL_64_after_hwframe+0x3d/0xa2
Apr 14 14:54:10 UP2-board kernel: [   51.365780] RIP: 0033:0x7fa4ef4c75d7
Apr 14 14:54:10 UP2-board kernel: [   51.365781] RSP: 002b:00007fa4e3ffcf98 EFLAGS: 00000246 ORIG_RAX: 0000000000000010
Apr 14 14:54:10 UP2-board kernel: [   51.365782] RAX: ffffffffffffffda RBX: 00007fa4dc002bc0 RCX: 00007fa4ef4c75d7
Apr 14 14:54:10 UP2-board kernel: [   51.365783] RDX: 00007fa4e3ffd150 RSI: 00000000c0405602 RDI: 000000000000000b
Apr 14 14:54:10 UP2-board kernel: [   51.365785] RBP: 000000005a313620 R08: 00007fa4dc867450 R09: 00007fa4dc867370
Apr 14 14:54:10 UP2-board kernel: [   51.365786] R10: 000000000000002b R11: 0000000000000246 R12: 00007fa4e3ffd190
Apr 14 14:54:10 UP2-board kernel: [   51.365787] R13: 00007fa4e3ffd120 R14: 00000000c02c564a R15: 00007fa4e3ffd3d0
Apr 14 14:54:10 UP2-board kernel: [   51.365788] Code: 30 0f 84 b4 01 00 00 3d 4a 4c 32 30 0f 84 97 01 00 00 3d 4d 33 31 30 0f 84 7a 01 00 00 89 c6 48 c7 c7 d7 a2 6a c0 e8 bf 67 3f f4 <0f> 0b 80 7b 0c 00 0f 85 e2 f3 ff ff 8b 43 2c 48 c7 c1 7e 96 6a 
Apr 14 14:54:10 UP2-board kernel: [   51.365827] ---[ end trace 9a1fa1de9d2a2628 ]---
Apr 14 14:54:10 UP2-board kernel: [   51.365871] ------------[ cut here ]------------
Apr 14 14:54:10 UP2-board kernel: [   51.365872] Unknown pixelformat 0x00000000
Apr 14 14:54:10 UP2-board kernel: [   51.365895] WARNING: CPU: 1 PID: 1766 at /build/linux-hwe-KDyd1s/linux-hwe-4.15.0/drivers/media/v4l2-core/v4l2-ioctl.c:1291 v4l_fill_fmtdesc+0xf41/0x1250 [videodev]
Apr 14 14:54:10 UP2-board kernel: [   51.365895] Modules linked in: snd_hda_codec_hdmi nls_iso8859_1 snd_soc_skl snd_soc_skl_ipc snd_hda_ext_core snd_soc_sst_dsp snd_soc_sst_ipc snd_soc_acpi snd_soc_core snd_compress ac97_bus snd_pcm_dmaengine intel_rapl spi_pxa2xx_platform intel_telemetry_pltdrv intel_punit_ipc intel_telemetry_core 8250_dw intel_pmc_ipc x86_pkg_temp_thermal coretemp kvm_intel kvm pinctrl_upboard irqbypass leds_upboard crct10dif_pclmul crc32_pclmul ghash_clmulni_intel pcbc aesni_intel aes_x86_64 crypto_simd glue_helper cryptd intel_cstate intel_rapl_perf snd_hda_intel snd_hda_codec snd_hda_core hid_sensor_accel_3d snd_hwdep snd_pcm hid_sensor_gyro_3d hid_sensor_trigger industrialio_triggered_buffer kfifo_buf pwm_lpss_pci hid_sensor_iio_common uvcvideo pwm_lpss lpc_ich snd_seq_midi i915 videobuf2_vmalloc snd_seq_midi_event
Apr 14 14:54:10 UP2-board kernel: [   51.365926]  snd_rawmidi videobuf2_memops videobuf2_v4l2 videobuf2_core snd_seq input_leds videodev media idma64 snd_seq_device virt_dma snd_timer drm_kms_helper intel_lpss_pci intel_lpss drm snd i2c_algo_bit mei_me fb_sys_fops shpchp mei syscopyarea soundcore sysfillrect sysimgblt ti_adc128s052 industrialio video mac_hid upboard_fpga sch_fq_codel parport_pc ppdev lp parport ip_tables x_tables autofs4 hid_generic hid_sensor_custom hid_sen [... cut ...]
Apr 14 14:54:16 UP2-board kernel: [   57.734169] R10: 000055c0d9c57010 R11: 0000000000000246 R12: 00007ffecfa502d0
Apr 14 14:54:16 UP2-board kernel: [   57.734170] R13: 00007fa4f08371e0 R14: 000001e000000280 R15: 00007ffecfa50490
Apr 14 14:54:16 UP2-board kernel: [   57.734172] Code: 30 0f 84 b4 01 00 00 3d 4a 4c 32 30 0f 84 97 01 00 00 3d 4d 33 31 30 0f 84 7a 01 00 00 89 c6 48 c7 c7 d7 a2 6a c0 e8 bf 67 3f f4 <0f> 0b 80 7b 0c 00 0f 85 e2 f3 ff ff 8b 43 2c 48 c7 c1 7e 96 6a 
Apr 14 14:54:16 UP2-board kernel: [   57.734211] ---[ end trace 9a1fa1de9d2a262d ]---
```

MartyG-RealSense, 2020-04-14:
Does the board restart if you only turn on the Stereo module, please?

AndreasAZiegler, 2020-04-14:
@MartyG-RealSense: Yes, it also restarts if I only turn on the Stereo module.

MartyG-RealSense, 2020-04-14:
I would speculate that the RealSense Viewer crash is the first event to occur when the stereo streamng is activated.  The second event (the board reset) may be occurring because the crash causes an event that the board's operating system cannot recover from, and so an auto-reset is initiated to correct the problem.

In the first two errors that are logged:

UP2-board kernel: [   51.365543] Unknown pixelformat 0x00000000
UP2-board kernel: [   51.365602] WARNING: CPU: 1 PID: 1766 at /build/linux-hwe-KDyd1s/linux-hwe-4.15.0/drivers/media/v4l2-core/v4l2-ioctl.c:1291 v4l_fill_fmtdesc+0xf41/0x1250 [videodev]

I believe that the appropriate course of action with the Unknown pixelformat error is to perform a kernel patching.

AndreasAZiegler, 2020-04-14:
Thanks for the information. How does the kernel patching work with the UP2
kernel?

MartyG-RealSense, 2020-04-14:
The librealsense kernel patching process has a kernel script that patches kernel modules affected by RealSense.  I believe that the appropriate LTS patching instruction is:

./scripts/patch-realsense-ubuntu-lts.sh

AndreasAZiegler, 2020-04-15:
So I followed the instruction [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) with ```./scripts/patch-realsense-ubuntu-lts.sh```. As the up2 kernel is ```Linux UP2-board 4.15.0-37-generic #40~upboard06-Ubuntu SMP Wed Nov 27 11:29:44 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux```, I don't know if this is right.

As soon as I turn on the Stereo module, the system reboots and I get this is the syslog
```
Apr 15 07:22:09 UP2-board kernel: [  306.727518] ------------[ cut here ]------------
Apr 15 07:22:09 UP2-board kernel: [  306.727522] Unknown pixelformat 0x00000000
Apr 15 07:22:09 UP2-board kernel: [  306.727561] WARNING: CPU: 0 PID: 1998 at /build/linux-hwe-KDyd1s/linux-hwe-4.15.0/drivers/media/v4l2-core/v4l2-ioctl.c:1291 v4l_fill_fmtdesc+0xf41/0x1250 [videodev]
Apr 15 07:22:09 UP2-board kernel: [  306.727563] Modules linked in: uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_core videodev media nls_iso8859_1 snd_hda_codec_hdmi snd_soc_skl snd_soc_skl_ipc snd_hda_ext_core snd_soc_sst_dsp snd_soc_sst_ipc snd_soc_acpi snd_soc_core snd_compress ac97_bus snd_pcm_dmaengine spi_pxa2xx_platform 8250_dw leds_upboard pinctrl_upboard intel_rapl intel_telemetry_pltdrv intel_punit_ipc intel_telemetry_core intel_pmc_ipc x86_pkg_temp_thermal coretemp kvm_intel kvm irqbypass crct10dif_pclmul crc32_pclmul ghash_clmulni_intel pcbc aesni_intel aes_x86_64 crypto_simd glue_helper cryptd intel_cstate intel_rapl_perf snd_hda_intel snd_hda_codec snd_hda_core snd_hwdep lpc_ich input_leds pwm_lpss_pci pwm_lpss snd_pcm ti_adc128s052 industrialio i915 snd_seq_midi snd_seq_midi_event snd_rawmidi snd_seq
Apr 15 07:22:09 UP2-board kernel: [  306.727612]  idma64 virt_dma snd_seq_device intel_lpss_pci snd_timer intel_lpss video drm_kms_helper drm upboard_fpga mei_me i2c_algo_bit fb_sys_fops snd syscopyarea mac_hid sysfillrect sysimgblt mei soundcore shpchp sch_fq_codel parport_pc ppdev lp parport ip_tables x_tables autofs4 hid_generic usbhid hid mmc_block dwc3 ulpi udc_core sdhci_pci sdhci r8169 dwc3_pci mii ahci libahci pinctrl_broxton
Apr 15 07:22:09 UP2-board kernel: [  306.727652] CPU: 0 PID: 1998 Comm: rs-enumerate-de Not tainted 4.15.0-37-generic #40~upboard06-Ubuntu
Apr 15 07:22:09 UP2-board kernel: [  306.727653] Hardware name: AAEON UP-APL01/UP-APL01, BIOS UPA1AM42 03/12/2019
Apr 15 07:22:09 UP2-board kernel: [  306.727660] RIP: 0010:v4l_fill_fmtdesc+0xf41/0x1250 [videodev]
Apr 15 07:22:09 UP2-board kernel: [  306.727661] RSP: 0018:ffffb4df81977c30 EFLAGS: 00010282
Apr 15 07:22:09 UP2-board kernel: [  306.727663] RAX: 0000000000000000 RBX: ffffb4df81977d68 RCX: 0000000000000006
Apr 15 07:22:09 UP2-board kernel: [  306.727664] RDX: 0000000000000007 RSI: 0000000000000086 RDI: ffff95bb3fc16490
Apr 15 07:22:09 UP2-board kernel: [  306.727666] RBP: ffffb4df81977c40 R08: 0000000000000001 R09: 000000000000038c
Apr 15 07:22:09 UP2-board kernel: [  306.727667] R10: ffff95bb08649000 R11: 0000000000000000 R12: ffff95baf0191300
Apr 15 07:22:09 UP2-board kernel: [  306.727668] R13: ffffffffc097f900 R14: ffff95bb08649000 R15: ffff95baf0191300
Apr 15 07:22:09 UP2-board kernel: [  306.727670] FS:  00007f7517a27540(0000) GS:ffff95bb3fc00000(0000) knlGS:0000000000000000
Apr 15 07:22:09 UP2-board kernel: [  306.727671] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Apr 15 07:22:09 UP2-board kernel: [  306.727672] CR2: 00007f7517839880 CR3: 0000000231ca2000 CR4: 00000000003406f0
Apr 15 07:22:09 UP2-board kernel: [  306.727674] Call Trace:
Apr 15 07:22:09 UP2-board kernel: [  306.727685]  v4l_enum_fmt+0x74/0x110 [videodev]
Apr 15 07:22:09 UP2-board kernel: [  306.727692]  __video_do_ioctl+0x370/0x380 [videodev]
Apr 15 07:22:09 UP2-board kernel: [  306.727696]  ? terminate_walk+0x8e/0xf0
Apr 15 07:22:09 UP2-board kernel: [  306.727703]  video_usercopy+0x24d/0x640 [videodev]
Apr 15 07:22:09 UP2-board kernel: [  306.727709]  ? video_ioctl2+0x20/0x20 [videodev]
Apr 15 07:22:09 UP2-board kernel: [  306.727714]  ? security_file_alloc+0x68/0xa0
Apr 15 07:22:09 UP2-board kernel: [  306.727720]  video_ioctl2+0x15/0x20 [videodev]
Apr 15 07:22:09 UP2-board kernel: [  306.727726]  v4l2_ioctl+0xc8/0xf0 [videodev]
Apr 15 07:22:09 UP2-board kernel: [  306.727729]  do_vfs_ioctl+0xa8/0x630
Apr 15 07:22:09 UP2-board kernel: [  306.727732]  ? get_unused_fd_flags+0x30/0x40
Apr 15 07:22:09 UP2-board kernel: [  306.727734]  SyS_ioctl+0x79/0x90
Apr 15 07:22:09 UP2-board kernel: [  306.727738]  do_syscall_64+0x73/0x130
Apr 15 07:22:09 UP2-board kernel: [  306.727742]  entry_SYSCALL_64_after_hwframe+0x3d/0xa2
Apr 15 07:22:09 UP2-board kernel: [  306.727744] RIP: 0033:0x7f75158565d7
Apr 15 07:22:09 UP2-board kernel: [  306.727745] RSP: 002b:00007fff6142ce68 EFLAGS: 00000246 ORIG_RAX: 0000000000000010
Apr 15 07:22:09 UP2-board kernel: [  306.727747] RAX: ffffffffffffffda RBX: 000055794c1ff070 RCX: 00007f75158565d7
Apr 15 07:22:09 UP2-board kernel: [  306.727748] RDX: 00007fff6142d020 RSI: 00000000c0405602 RDI: 0000000000000005
Apr 15 07:22:09 UP2-board kernel: [  306.727749] RBP: 000000005a313620 R08: 000055794c1f74d0 R09: 0000000000000000
Apr 15 07:22:09 UP2-board kernel: [  306.727750] R10: 000055794c036010 R11: 0000000000000246 R12: 00007fff6142d060
Apr 15 07:22:09 UP2-board kernel: [  306.727751] R13: 00007fff6142cff0 R14: 00000000c02c564a R15: 00007fff6142d2a0
[... cut ...]
Apr 15 07:28:27 UP2-board /usr/lib/gdm3/gdm-x-session[938]: (**) Option "xkb_layout" "us"
Apr 15 07:28:27 UP2-board /usr/lib/gdm3/gdm-x-session[938]: (WW) Option "xkb_variant" requires a string value
Apr 15 07:28:27 UP2-board /usr/lib/gdm3/gdm-x-session[938]: (WW) Option "xkb_options" requires a string value
Apr 15 07:28:27 UP2-board /usr/lib/gdm3/gdm-x-session[938]: (II) event11 - Intel(R) RealSense(TM) Depth Ca: is tagged by udev as: Keyboard
Apr 15 07:28:27 UP2-board /usr/lib/gdm3/gdm-x-session[938]: (II) event11 - Intel(R) RealSense(TM) Depth Ca: device is a keyboard
```

MartyG-RealSense, 2020-04-15:
Apologies for the delay in responding further, as I was considering your case very carefully but was unable to find a solution.  I am going to refer it internally to Intel for further advice.

MartyG-RealSense, 2020-04-15:
Apologies for the delay in responding further, as I was considering your case very carefully.  My advice at this point would be to try the "bypass" method of Librealsense installation if possible for you.  It needs an internet connection but allows Librealsense to be installed without reliance on Linux versions, kernel versions, patches, etc.  It is a useful technique when the standard installation methods are not working.

https://support.intelrealsense.com/hc/en-us/community/posts/360037657234-downloading-the-packages-dkms-and-sdk-on-ubuntu-19-10?page=1#community_comment_360009799533

AndreasAZiegler, 2020-04-16:
Thanks for the information @MartyG-RealSense. I now built the librealsense-2.34.0 with:
```
cmake -DCMAKE_BUILD_TYPE=Release -DFORCE_RSUSB_BACKEND=true ..
```

When I now start the RealSense viewer, I get this in the syslog
```
Apr 16 07:11:56 UP2-board org.gnome.Shell.desktop[1108]: Window manager warning: Buggy client sent a _NET_ACTIVE_WINDOW message with a timestamp of 0 for 0x1c00015 (Intel Real)
Apr 16 07:11:56 UP2-board org.gnome.Shell.desktop[1108]: Window manager warning: Buggy client sent a _NET_ACTIVE_WINDOW message with a timestamp of 0 for 0x1c00015 (Intel Real)
Apr 16 07:11:56 UP2-board upowerd[1114]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:1.2
Apr 16 07:11:56 UP2-board upowerd[1114]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:1.0
Apr 16 07:11:56 UP2-board upowerd[1114]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:1.1
Apr 16 07:11:56 UP2-board upowerd[1114]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:1.0
Apr 16 07:11:56 UP2-board upowerd[1114]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:1.2
Apr 16 07:11:56 UP2-board upowerd[1114]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:1.1
Apr 16 07:11:56 UP2-board upowerd[1114]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:1.1
Apr 16 07:11:56 UP2-board upowerd[1114]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:1.2
Apr 16 07:11:56 UP2-board upowerd[1114]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:1.0
Apr 16 07:11:56 UP2-board upowerd[1114]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:1.0
Apr 16 07:11:56 UP2-board upowerd[1114]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:1.1
Apr 16 07:11:56 UP2-board upowerd[1114]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:1.2
```
This goes on and on until I start the Stereo module, when the system restarts. Something else that I notice is that when I have the RealSense connected to the UP2 board during start, Ubuntu is not able to boot sometimes. I get the Ubuntu booting screen and when the screen turns black before the the Ubuntu background would appear, the system restarts.

MartyG-RealSense, 2020-04-16:
Apparently, upowerd is a power management system.  

The only case I could find where this bind error occurred with Librealsense was with an Up Core board.   The camera was being used on the board alongside an AI Core device.  It worked with USB3 when plugged in without the AI Core, but otherwise would only work on USB2 if the AI Core was also plugged in.

Are you using an AI Core or similar device such as a Neural Compute Stick please?

AndreasAZiegler, 2020-04-16:
I'm using a plain UP2 board without any extension.
[image]

MartyG-RealSense, 2020-04-16:
Do you get errors if you open the Viewer with the camera unplugged?  If not, do the bind / unbind errors start as soon as the camera is plugged in?

When the camera is plugged in, does the camera information in the Viewer side panel always remain there, or does the camera appear and disappear?

AndreasAZiegler, 2020-04-16:
> Do you get errors if you open the Viewer with the camera unplugged? 

No

>  If not, do the bind / unbind errors start as soon as the camera is plugged in?

Yes, see syslog below

```
Apr 16 08:18:34 UP2-board kernel: [  106.296172] usb 2-2: new SuperSpeed USB device number 3 using xhci_hcd
Apr 16 08:18:34 UP2-board kernel: [  106.317001] usb 2-2: New USB device found, idVendor=8086, idProduct=0b07
Apr 16 08:18:34 UP2-board kernel: [  106.317008] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Apr 16 08:18:34 UP2-board kernel: [  106.317012] usb 2-2: Product: Intel(R) RealSense(TM) Depth Camera 435 
Apr 16 08:18:34 UP2-board kernel: [  106.317016] usb 2-2: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435 
Apr 16 08:18:34 UP2-board kernel: [  106.317020] usb 2-2: SerialNumber: 823313020254
Apr 16 08:18:34 UP2-board mtp-probe: checking bus 2, device 3: "/sys/devices/pci0000:00/0000:00:15.0/usb2/2-2"
Apr 16 08:18:34 UP2-board mtp-probe: bus: 2, device: 3 was not an MTP device
Apr 16 08:18:34 UP2-board kernel: [  106.489459] media: Linux media interface: v0.10
Apr 16 08:18:34 UP2-board kernel: [  106.502825] Linux video capture interface: v2.00
Apr 16 08:18:34 UP2-board kernel: [  106.520153] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
Apr 16 08:18:34 UP2-board kernel: [  106.520183] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
Apr 16 08:18:34 UP2-board kernel: [  106.520292] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
Apr 16 08:18:34 UP2-board kernel: [  106.522360] uvcvideo: Unable to create debugfs 2-3 directory.
Apr 16 08:18:34 UP2-board kernel: [  106.522450] uvcvideo 2-2:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
Apr 16 08:18:34 UP2-board kernel: [  106.522453] uvcvideo 2-2:1.0: Entity type for entity Processing 2 was not initialized!
Apr 16 08:18:34 UP2-board kernel: [  106.522455] uvcvideo 2-2:1.0: Entity type for entity Camera 1 was not initialized!
Apr 16 08:18:34 UP2-board kernel: [  106.522581] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:1.0/input/input10
Apr 16 08:18:34 UP2-board kernel: [  106.522884] uvcvideo: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
Apr 16 08:18:34 UP2-board kernel: [  106.522891] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
Apr 16 08:18:34 UP2-board kernel: [  106.523689] uvcvideo: Unable to create debugfs 2-3 directory.
Apr 16 08:18:34 UP2-board kernel: [  106.523765] uvcvideo 2-2:1.3: Entity type for entity Processing 7 was not initialized!
Apr 16 08:18:34 UP2-board kernel: [  106.523767] uvcvideo 2-2:1.3: Entity type for entity Extension 8 was not initialized!
Apr 16 08:18:34 UP2-board kernel: [  106.523768] uvcvideo 2-2:1.3: Entity type for entity Camera 6 was not initialized!
Apr 16 08:18:34 UP2-board kernel: [  106.523869] usbcore: registered new interface driver uvcvideo
Apr 16 08:18:34 UP2-board kernel: [  106.523870] USB Video Class driver (1.1.1)
Apr 16 08:18:34 UP2-board upowerd[1099]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:15.0/usb2/2-2
Apr 16 08:18:34 UP2-board upowerd[1099]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:1.4
Apr 16 08:18:34 UP2-board upowerd[1099]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:1.2
Apr 16 08:18:34 UP2-board upowerd[1099]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:1.1
Apr 16 08:18:34 UP2-board upowerd[1099]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:1.3
Apr 16 08:18:34 UP2-board /usr/lib/gdm3/gdm-x-session[911]: (II) config/udev: Adding input device Intel(R) RealSense(TM) Depth Ca (/dev/input/event10)
Apr 16 08:18:34 UP2-board /usr/lib/gdm3/gdm-x-session[911]: (**) Intel(R) RealSense(TM) Depth Ca: Applying InputClass "libinput keyboard catchall"
Apr 16 08:18:34 UP2-board /usr/lib/gdm3/gdm-x-session[911]: (II) Using input driver 'libinput' for 'Intel(R) RealSense(TM) Depth Ca'
Apr 16 08:18:34 UP2-board upowerd[1099]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:1.0
Apr 16 08:18:34 UP2-board /usr/lib/gdm3/gdm-x-session[911]: (II) systemd-logind: got fd for /dev/input/event10 13:74 fd 55 paused 0
Apr 16 08:18:34 UP2-board /usr/lib/gdm3/gdm-x-session[911]: (**) Intel(R) RealSense(TM) Depth Ca: always reports core events
Apr 16 08:18:34 UP2-board /usr/lib/gdm3/gdm-x-session[911]: (**) Option "Device" "/dev/input/event10"
Apr 16 08:18:34 UP2-board /usr/lib/gdm3/gdm-x-session[911]: (**) Option "_source" "server/udev"
Apr 16 08:18:34 UP2-board /usr/lib/gdm3/gdm-x-session[911]: (II) event10 - Intel(R) RealSense(TM) Depth Ca: is tagged by udev as: Keyboard
Apr 16 08:18:34 UP2-board /usr/lib/gdm3/gdm-x-session[911]: (II) event10 - Intel(R) RealSense(TM) Depth Ca: device is a keyboard
Apr 16 08:18:34 UP2-board /usr/lib/gdm3/gdm-x-session[911]: (II) event10 - Intel(R) RealSense(TM) Depth Ca: device removed
Apr 16 08:18:34 UP2-board /usr/lib/gdm3/gdm-x-session[911]: (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:1.0/input/input10/event10"
Apr 16 08:18:34 UP2-board /usr/lib/gdm3/gdm-x-session[911]: (II) XINPUT: Adding extended input device "Intel(R) RealSense(TM) Depth Ca" (type: KEYBOARD, id 11)
[... cut ...]
Apr 16 08:18:36 UP2-board upowerd[1099]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:1.1
Apr 16 08:18:36 UP2-board upowerd[1099]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:1.2
Apr 16 08:18:36 UP2-board upowerd[1099]: unhandled action 'unbind' on /sys/devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:1.0
```

> When the camera is plugged in, does the camera information in the Viewer side panel always remain there, or does the camera appear and disappear?

The camera information remains.

MartyG-RealSense, 2020-04-16:
It looks as though udev is classifying the camera as being a keyboard!  I had been wondering if the problem was related to the udev device rules for Librealsense.

I do not want to risk performing a firmware update during these errors.  I will refer the case to an Intel RealSense support manager on the GitHub team for advice.  I apologise for the delay in resolving your issue in the meantime.

MartyG-RealSense, 2020-04-16:
I have had acknowledgement of my request for Intel engineering support on this case.  

In the meantime, would it be possible to consider using a mains electricity powered USB 3 hub to help deal with any problems related to power stability on the USB ports that might be occurring.  Thanks!

AndreasAZiegler, 2020-04-17:
The start up problems seems gone when I connect the RealSense via a powered USB3.0 hub. In the final setup I will use two RealSense cameras. Is there a bandwidth problem when I connect them both via the  USB3.0 hub?

MartyG-RealSense, 2020-04-17:
I'm very pleased that a powered hub helped with your problems!  I will let Intel support know of your success.

There is a bandwidth cost when connecting multiple cameras to a hub.  Below is a table that Intel did when testing up to 4 cameras with an AmazonBasics hub and without hardware sync active.

https://files.readme.io/d466828-m10.png

Intel's full white paper on multiple cameras an be viewed here:

https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration

The Up Squared (UP2) board is powerful enough to handle two 400 Series cameras, so you should be fine in that respect.

AndreasAZiegler, 2020-04-23:
As my previous progress looked promising, I went further and experimented with a D435 and a D435i simultaneously connected to the UP2 board. Unfortunately this doesn't work as expected. I have them both connected via a powered USB3 hub.

When I install the librealsense2 via ```apt install```, and use the realsense viewer, the application freezes and/or the system restarts when I activate stereo on either of the cameras. It also happened that the viewer kicked out the cameras (they were not longer listed) and when I re(added) them, they were listed as USB 2 devices. In the file realsense_ubuntu.txt there are part of the syslog covering the events.
[realsense_ubuntu.txt](https://github.com/IntelRealSense/librealsense/files/4522968/realsense_ubuntu.txt)

When I build librealsense2 with ```cmake -DCMAKE_BUILDE_TYPE=Release ..```, the realsense viewer also freezes and/or the system restarts when I start the stereo module. Some corresponding syslog entries are in realsense_build.txt.
[realsense_build.txt](https://github.com/IntelRealSense/librealsense/files/4522982/realsense_build.txt)

When I build it with ```cmake -DCMAKE_BUILD_TYPE=Release -DFORCE_RSUSB_BACKEND=true ..```, the system reboots or I get the message in the figure below, when I activate the stereo module. The syslog are in the file realsense_build_special.txt.
[realsense_build_special.txt](https://github.com/IntelRealSense/librealsense/files/4523002/realsense_build_special.txt)
[image]

MartyG-RealSense, 2020-04-23:
Apologies for the delay in responding.  I have been considering your case carefully.

Can you confirm to me please whether you are using the short official USB cables that are supplied with the cameras, or whether you are using your own choice of USB cable that is longer (for example, 2 meters or more).  Thank you very much.

AndreasAZiegler, 2020-04-23:
I'm using the official cables that came with the cameras.

MartyG-RealSense, 2020-04-23:
Thank you very much.  It is an unusual case.   Usually the Up Squared does not have problems with 2 cameras.  The hub does seem to work, as it is able to work with a single camera.

Forgive the question - I am trying to eliminate possibilities - but is the powered hub's mains plug switched on at the electricity mains socket on the wall?

AndreasAZiegler, 2020-04-24:
I just checked and the powered hub is connected (power cable) and I could measure the required 5V with a multi meter. I'm using [this USB3 hub](https://www.digitec.ch/de/s1/product/hama-39879-4ports-usb-hub-3824095). Unfortunately I don't have another one to check if the hub causes the problem.

I would actually prefer if I can connect the two RealSense cameras directly with the UP2 board. With another (an older) UP2 board that worked.

MartyG-RealSense, 2020-04-24:
My recollection of USB standards is that it sends 5V even if it is an "unpowered" connection plugged into a non-mains hub or directly into the computer.  So it may not demonstrate that the hub is working correctly.

You should be able to at least connect 2 cameras to the board directly and have them detected, as their power usage should be relatively low whilst they are not streaming.

You are right that Up Squared should be able to support 2 cameras.  The link below has detailed advice from a RealSense user who did so successfully.

https://forums.intel.com/s/question/0D70P0000069F8LSAU

MartyG-RealSense, 2020-05-11:
This case will be closed after the next 7 days from the time of writing this message if there are no further comments received.  Thanks!

closed 2020-05-17 by MartyG-RealSense