realsenseai/librealsense #6238: How to resize draw_pointcloud?

duongbaoquy, 2020-04-15:
Hi, 
I'm using D435i camera, when i use command draw_pointcloud it works fine. But now i want to resize the view to see only the red zone ( in the image):
[image]

MartyG-RealSense, 2020-04-15:
It sounds as though you do not want to change the scale of the point cloud, but instead exclude all of the data outside of the red box.  This can be done with a 'bounding box'.  

It would likely be more complex to define a three-dimensional bounding box for a point cloud than it would be for a 2D image though.   One RealSense user approached the problem by using a **struct** to define minimum and maximum values for X and Y and then getting the point cloud points.

https://github.com/IntelRealSense/librealsense/issues/2016#issuecomment-403804157

duongbaoquy, 2020-04-16:
@MartyG-RealSense 
Thanks for your reply,maybe I'll find something useful with it
But actually, I want to remove all the data outside the red box so I can only see the object I want

closed 2020-05-15 by duongbaoquy