realsenseai/librealsense #6239: How to get depth value of a specific cordinatet

BarRae, 2020-04-15:
I am currently working on a program to detect blue marbles and localise them in 3D space.
I am using OpenCV for this.

I am currently able to detect the marbles in a 2D colour image coming from the D435i camera.
This gives me the X, Y position of this marble in the 2D image.

Does anyone know how I can get the depth value at that specific X, Y position.
Feel free to ask for more information.

MartyG-RealSense, 2020-04-15:
If you need to get the XYZ coordinates in real-time, the only OpenCV script I could find for this is in the link below.

https://github.com/IntelRealSense/librealsense/issues/1904#issuecomment-398434434

For getting the coordinates from an image file, there is this script:

https://github.com/IntelRealSense/librealsense/issues/2200#issuecomment-411774418

BarRae, 2020-04-15:
Thank you for your reply.

I have been looking into the #1904 (comment) that you linked and have been trying to implement it.
I am programming in C++ and am a complete novice in python.
so I'm having a hard time to implement it in my code.

I'll try to give a better description of what i'm doing an what my goal is:
Using a pipeline I start a color stream.
From that color stream I take one frame and store it as Mat color.
I then convert that image from BGR to HSV called imgHSV.
using the inRange function I filter out every color that is not blue in a new output called OutputImage.
Next I use the HoughCircles function to detect circles in OutputImage and "print" these circles on the imgHSV image. At this point I know the X,Y coordinates of the centre of the circles and the radius of the circles.

Now I need to know how I can get the depth value of that certain X,Y coordinate.

Im sorry if this is basically repeating my question.
I am quite new to this.

Would it help if I posted my current code, keep in mind that it is a mess at the moment because I am currently using it for trying different methods and functions

MartyG-RealSense, 2020-04-15:
The requirements of your project exceed my current level of OpenCV programming knowledge, unfortunately.  I will tag @dorodnic in for advice, as he has contributed some OpenCV examples.

ev-mp, 2020-04-15:
@BarRae , to find the depth of a given RGB pixel you should either call (1) [rs2_project_color_pixel_to_depth_pixel](https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/rsutil.h#L170) API or (2) utilize "align" processing block. 
The choice would depend on the number of correlations you need to perform per frame.  If less than ~100 then the first method will be more effective, otherwise - you can switch to the processing block.
#5173, #5944, #2948,

BarRae, 2020-04-16:
Thank you both, I will definitely look at the linked examples today, as soon as I got my application working again (it broke yesterday while copying a few files). I will come back to you with my results. If anyone needs extra information feel free to ask. If you guys need my code than I will gladly provide it after I cleaned it up a bit :-) .

BarRae, 2020-04-16:
Hello again, @ev-mp and @MartyG-RealSense 

I have chosen to work with option 1 as @ev-mp suggested, scince I will be detecting 1 item per image.
I was wondering if you could help me fill in some parameters, I have found post in which some are explained but not all.

I would like to know what the function is of the following parameters and what I would enter for those parameters.

const uint16_t* data, float depth_scale,
const float from_pixel[2])

for from_pixel[2] I mainly want to know what I would have to fill in if the pixel in question is Point(c[0], c[1])

I'm sorry if this is a basic question.

MartyG-RealSense, 2020-04-16:
@BarRae   I will have to defer to the superior programming knowledge of @ev-mp on this particular question.  I apologise for the delay in the meantime!

BarRae, 2020-04-16:
I would like to add something,

Could you help me with the intrinsic and extrinsic parameters.
I've seen a lot of posts about this an most of them refer to the API-how-to.
The code does run currently, but I don't know what to do with these intrinsic parameters

I will provide my current code:
`#include <iostream>
#include <opencv2/opencv.hpp>
#include <opencv2/core.hpp>
#include <opencv2/highgui.hpp>
#include <rsutil.h>
#include <librealsense2/rs.hpp>

using namespace std;
using namespace cv;

Mat image;
Mat imgHSV;
Mat OutputImage;

int iLowH = 0;
int iHighH = 12;
int iLowS = 60;
int iHighS = 255;
int iLowV = 0;
int iHighV = 245;

int acc = 1;
int rows = 10;
int para1 = 5;
int para2 = 10;
int minRad = 5;
int maxRad = 70;

float from_pixel[2] = { 0 }, to_pixel[2] = { 0 };
float depth_scale = 0.001;

static void HSVthreshold(int, int, int, int, int, int, void*)
{
    inRange(imgHSV, Scalar(iLowH, iLowS, iLowV), Scalar(iHighH, iHighS, iHighV), OutputImage);
}

static void Circle_detector(int, int, int, int, int, void*)
{
    vector<Vec3f> circles;
    HoughCircles(OutputImage, circles, HOUGH_GRADIENT, 1,
        OutputImage.rows / rows,      //change to detect circles that are closer to eachother
        para1, para2, minRad, maxRad);        //chang last to parameters to detect larger or smaller circles

    for (size_t i = 0; i < circles.size(); i++)
    {
        Vec3i c = circles[i];
        Point center = Point(c[0], c[1]);
        // circle center
        circle(imgHSV, center, 1, Scalar(0, 255, 0), 3, LINE_AA);
        // circle outline
        int radius = c[2];
        circle(imgHSV, center, radius, Scalar(255, 0, 0), 3, LINE_AA);

        cout << "The center of the detection is located at pixel: " << Point(c[0], c[1]);
        //rs2_project_color_pixel_to_depth_pixel(to_pixel, reinterpret_cast<const uint16_t*>(depth.get_data()), depth_scale, 0.1f, 10, &depth_intrin, &color_intrin, &color_extrin_to_depth, &depth_extrin_to_color, from_pixel);
    }
}

int main()
{
    // Contructing piplines and other stuff to receive data from the realsense camera.

    //Contruct a pipeline which abstracts the device
    rs2::pipeline pipe;     //for color
    rs2::pipeline pipe_in;  //for instrinsic

    //Create a configuration for configuring the pipeline with a non default profile
    rs2::config cfg;

    //Add desired streams to configuration
    cfg.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_BGR8, 30);

    //Instruct pipeline to start streaming with the requested configuration
    pipe.start(cfg);    //for color

    // Camera warmup - dropping several first frames to let auto-exposure stabilize
    rs2::frameset frames;
    for (int i = 0; i < 30; i++)
    {
        //Wait for all configured streams to produce a frame
        frames = pipe.wait_for_frames();
    }

    //Get color each frame
    rs2::frame color_frame = frames.get_color_frame();

    // Creating OpenCV Matrix from a color image
    Mat color(Size(640, 480), CV_8UC3, (void*)color_frame.get_data(), Mat::AUTO_STEP);

    //intrinsic video parameters (copied from exaple)
    rs2::pipeline_profile selection = pipe_in.start();
    auto depth_stream = selection.get_stream(RS2_STREAM_DEPTH)
        .as<rs2::video_stream_profile>();
    auto resolution = std::make_pair(depth_stream.width(), depth_stream.height());
    auto i = depth_stream.get_intrinsics();
    auto principal_point = std::make_pair(i.ppx, i.ppy);
    auto focal_length = std::make_pair(i.fx, i.fy);
    rs2_distortion model = i.model;

    // cheking if an image was read
    if (color.empty())
    {
        cerr << "image was not generated !" << endl;
        return 1;
    }

    namedWindow("Display Image", WINDOW_AUTOSIZE);
    imshow("Display Image", color);

    //convert RGB to HSV
    cvtColor(color, imgHSV, COLOR_BGR2HSV);

    //showHSV image
    imshow("image", imgHSV);

    //Create windows
    namedWindow("image", WINDOW_AUTOSIZE); //window for original image
    namedWindow("Output", WINDOW_AUTOSIZE); //window for output mask

    //aplying color filter to HSV image
    HSVthreshold(iLowH, iHighH, iLowS, iHighS, iLowV, iHighV, 0);

    //Optional filter --> does not work properly at the moment needs more setup <--
    
    //morphological opening (remove small objects from the foreground)
    erode(OutputImage, OutputImage, getStructuringElement(MORPH_ELLIPSE, Size(2, 2)));
    dilate(OutputImage, OutputImage, getStructuringElement(MORPH_ELLIPSE, Size(2, 2)));

    //morphological closing (fill small holes in the foreground)
    dilate(OutputImage, OutputImage, getStructuringElement(MORPH_ELLIPSE, Size(2, 2)));
    erode(OutputImage, OutputImage, getStructuringElement(MORPH_ELLIPSE, Size(2, 2)));

    imshow("Output", OutputImage);
    

    Circle_detector(rows, para1, para2, minRad, maxRad, 0);
    imshow("image", imgHSV);
 
    waitKey();

    return 0;
}`

Again I'm really sorry for my code being a mess and for misunderstanding.

MartyG-RealSense, 2020-04-16:
Would this discussion with a C++ script for using **rs2_project_color_pixel_to_depth_pixel** be of help to you, please?

https://github.com/IntelRealSense/librealsense/issues/2982

AbdelsalamHaa, 2020-04-17:
@BarRae I actually have the same question as you. I'm kinda new to Liprealsense but I'm familier with Opencv. since u used opencv to get the X,Y values why don't u just use the index values to get the distance from depthFram 

```
try (FrameSet frames = pipe.waitForFrames()) {
               try (Frame f = frames.first(StreamType.DEPTH))
               {
                   DepthFrame depth = f.as(Extension.DEPTH_FRAME);
                   final float deptValue = depth.getDistance(depth.getWidth()/2, depth.getHeight()/2);
                   runOnUiThread(new Runnable() {
                       @Override
                       public void run() {
                           TextView textView = findViewById(R.id.distanceTextView);
                           textView.setText("Distance: " + df.format(deptValue));
                       }
                   });
               }
            }
```

this code will get the distance at the center of the image using this line of code 
`final float deptValue = depth.getDistance(depth.getWidth()/2, depth.getHeight()/2);`
so instead of converting your RGB or HSV image to get the distance you just take the indexs values from the image and get the distance from the DepthFrame
In other word u would use opencv to get the X,Y values only and the distance from the DepthFrame

ev-mp, 2020-04-17:
@BarRae, in order to have provision for RGB-Depth correlations you should slightly modify the flow:
1 . You don't need two `pipe` instances, so switch `pipe` object that will stream  color **and depth** data. The depth frame is required for the ultimate goal.
```cpp
    //Add desired streams to configuration
    cfg.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_BGR8, 30);
    cfg.enable_stream(RS2_STREAM_DEPTH, TBD, TBD, RS2_FORMAT_Z16, TBD);
  
    pipe.start(cfg);    //for color

    while(condition)
    {
            frames = pipe.wait_for_frames();
            auto depth_frame = frames.get_depth_frame();
            auto color_frame = frames.get_color_frame();
            // Make sure that both depth and  color are present for calculations
            if (!depth_frame  || !color_frame )
                continue;
            ...
            // Do the RGB detection stuff
```

2. Skip the RGB detection and assume that you've managed to establish the center of a blue marble as coordinate (i,j) in the RGB frame. 

3. Gather the prerequisites to use `rs2_project_color_pixel_to_depth_pixel`  and call the algorithm.
I haven't actually compiled the below code so some fixes might be needed
```cpp
    auto depth_profile = depth_frame.get_profile().as<rs2::video_stream_profile>();
    auto color_profile = color_frame .get_profile().as<rs2::video_stream_profile>();

    auto depth_intrin = depth_profile.get_intrinsics();
    auto color_intrin = color_profile.get_intrinsics();
    auto depth2color_extrin = depth_profile.get_extrinsics_to(color_profile);
    auto color2depth_extrin = color_profile.get_extrinsics_to(depth_profile);

    float rgb_src_pixel[2] = { i,j }; // The RGB coordinate for the center of the marble
    float dpt_tgt_pixel[2] = { 0 }, // The depth pixel that has the best match for that RGB coordinate

     // Search along a projected beam from 0.1m to 10 meter. This can be optimized to the concrete scenario, e.g. if you know that the data is bounded within [min..max] range
    rs2_project_color_pixel_to_depth_pixel(dpt_tgt_pixel, reinterpret_cast<const uint16_t*>(depth_frame.get_data()), depth_frame.get_depth_units(), 0.1f, 10,
                &depth_intrin, &color_intrin,
                &color2depth_extrin , &depth2color_extrin , rgb_src_pixel);

       // Verify that the depth correspondence is valid, i.e within the frame boundaries
       if ( dpt_tgt_pixel >= ... && 

      // Get the depth value for the pixel found
     auto distance = depth_frame.get_distance(dpt_tgt_pixel[0], dpt_tgt_pixel[1]);

```

BarRae, 2020-04-17:
@ev-mp thanks for your reaction

I'm looking at it now and have a question
In your code snippets you put this:
```
while(condition)
    {
            frames = pipe.wait_for_frames();
            auto depth_frame = frames.get_depth_frame();
            auto color_frame = frames.get_color_frame();
            // Make sure that both depth and  color are present for calculations
            if (!depth_frame  || !color_frame )
                continue;
```
I suppose this means that my
```
for (int i = 0; i < 30; i++)
    {
        //Wait for all configured streams to produce a frame
        frames = pipe.wait_for_frames();
    }
```

Has to go.
Furthermore I is not really clear to me what I should put in the condition for the while loop and till how far in the code this while loop should go.

Thank you in advance

ev-mp, 2020-04-17:
@BarRae, the `while(condition)`  designates the entry point of the main loop in your code.
I presume that this is not a single-shot scenario and that you try to find the correspondence continuously as it may not be present in every single frame, especially in dynamic environment.

In the above code I outline only the core blocks so you still can use the camera `warm-up` routine with 30 `black` frames if needed.

Q:
>I am trying to get the depth_scale this way (because depth_frame.get_depth_units() did not work):

A) Make sure you use the latest v2.34.0.
B) Make sure to downcast the frame instance properly -  `depth_frame.as<depth_frame>().get_units()`.  
There is no need for `auto depth_sensor = dev.query_sensors().front().as<depth_sensor>();` - the `get_units` does exactly the same in a concise form. In addition it just might not work syntactically.

@AbdelsalamHaa, 
>so instead of converting your RGB or HSV image to get the distance you just take the index values from the image and get the distance from the DepthFrame
something like this
final float deptValue = depth.getDistance(X, Y);

That proposition cannot work in the above scenario. The Depth and RGB images are not aligned spatially. There is a _very loose_ correlation between data stored in pixel (i,j) in RGB frame versus the data stored in the same (i,j) index in Depth frame.

BarRae, 2020-04-17:
Hello again, @ev-mp @MartyG-RealSense @AbdelsalamHaa 

I am having a proud moment.
My code works (almost) as it's supposed to!!!

The camera takes an image,
detects my object
tell's me the pixel in the picture which is the centre of the object and tells me the distance to this object.
If I press a button the cycle repeats.

The only thing I cannot get to work is the part at the end that checks if the depth pixel is within the frame. (to me this is not a big issue I think) 
If you guys know any improvements then I will gladly hear them.

For now I thank you all very much.
I will put my code down here for anyone interested in how I did it.
```
#include <iostream>
#include <opencv2/opencv.hpp>
#include <opencv2/core.hpp>
#include <opencv2/highgui.hpp>
#include <rsutil.h>
#include <librealsense2/rs.hpp>

using namespace std;
using namespace cv;

Mat image;
Mat imgHSV;
Mat OutputImage;

int iLowH = 0;
int iHighH = 12;
int iLowS = 60;
int iHighS = 255;
int iLowV = 0;
int iHighV = 245;

int acc = 1;
int rows = 10;
int para1 = 5;
int para2 = 10;
int minRad = 5;
int maxRad = 70;

int x;
int y;

int width = 640;
int height = 480;

int input;

static void HSVthreshold(int, int, int, int, int, int, void*)
{
    inRange(imgHSV, Scalar(iLowH, iLowS, iLowV), Scalar(iHighH, iHighS, iHighV), OutputImage);
}

static void Circle_detector(int, int, int, int, int, void*)
{
    vector<Vec3f> circles;
    HoughCircles(OutputImage, circles, HOUGH_GRADIENT, 1,
        OutputImage.rows / rows,      //change to detect circles that are closer to eachother
        para1, para2, minRad, maxRad);        //chang last to parameters to detect larger or smaller circles

    for (size_t i = 0; i < circles.size(); i++)
    {
        Vec3i c = circles[i];
        Point center = Point(c[0], c[1]);
        // circle center
        circle(imgHSV, center, 1, Scalar(0, 255, 0), 3, LINE_AA);
[... 105 lines cut ...]
        auto scale = sensor.get_depth_scale();

        // Search along a projected beam from 0.1m to 10 meter. This can be optimized to the concrete scenario, e.g. if you know that the data is bounded within [min..max] range
        rs2_project_color_pixel_to_depth_pixel(dpt_tgt_pixel, reinterpret_cast<const uint16_t*>(depth_frame.get_data()), scale, 0.1f, 2,
            &depth_intrin, &color_intrin,
            &color2depth_extrin, &depth2color_extrin, rgb_src_pixel);

        // Verify that the depth correspondence is valid, i.e within the frame boundaries
        
        if (dpt_tgt_pixel >= depth_frame.get_width && depth_frame.get_height)
        {
            auto distance = depth_frame.get_distance(dpt_tgt_pixel[0], dpt_tgt_pixel[1]);
            // Get the depth value for the pixel found

            cout << "The distance to the object is: " << distance << endl;
        }
        

        
        waitKey();

        }
        return 0;
}
```

MartyG-RealSense, 2020-04-17:
Thanks to everyone who contributed to this discussion and gave their valuable time!

BarRae, 2020-04-21:
Before I close this issue I wanted to ask one more question.
When using `rs2_project_color_pixel_to_depth_pixel` what is the reference point.
For where is the distance measured when using `get_distance`?

And furthermore if I use `rs2_deproject_pixel_to_point` to get the XYZ coordinate of my object, is this still from the same reference point?

I'm basically trying to ask from which image these distances are measured.

MartyG-RealSense, 2020-04-21:
On 400 Series cameras, the center of the left imager is the origin point.

https://github.com/IntelRealSense/librealsense/issues/2688#issuecomment-436612914

The dynamic calibration guide for the 400 Series, meanwhile, says: "Assume left camera is the reference camera and is located at world origin".

BarRae, 2020-04-21:
thank you for the information. 
I'll close the issue now.

closed 2020-04-21 by BarRae