realsenseai/librealsense #6241: Problem when Align depth and color images

Ayangdanny, 2020-04-15:
My sensor are SR 300 and SR 305. When I run the "align" demo in re-examples. I got the result below. The alignment between depth and color image is not good. There are big cracks in the edges. How to get a good alignment result?
[image]

Ayangdanny, 2020-04-15:
another Question. Which is the highest precision equipment in kinds of realsense sensors for near range (<1m).

MartyG-RealSense, 2020-04-15:
I ran tests with the **rs_align_advanced** sample program with my own SR300, replicating your pose exactly and moving my arms to see how the image quality reacted.  It most often broke up in the way that your image showed when I moved my hands (and so my lower arms) outwards horizontally, away from the center-line of the camera.

I also observed that it was necessary to be close to the camera (about 0.5 meters away) for the face and chest to appear in the image as much as it does in your image.

The results were not surprising to me.  When I spent a couple of years using the SR300 with hand tracking in the Unity game engine, tracking of the joints of the hand would often be lost when moving the hands horizontally outwards to the distance from the camera center-line where the lower arm is breaking up.  Face tracking was much more reliable because the head remained relatively in the center-line of the camera's field of view.

Unfortunately there is not likely to be much that can be done in this situation.  The SR300 / SR305 has a limited depth sensing distance that requires the camera to be close up to the body, and widening the field of view with two of these cameras would not be practical because (unlike the 400 Series cameras), SR300 type cameras experience interference when placed in proximity to each other.

As your picture shows, face recognition with the SR300 / SR305 seems to be working well though.

In regard to the best near-range performance RealSense cameras: my first recommendation would be the new L515 lidar depth camera.  My second choice would be the D435, as it has a smaller minimum distance than the D415 model and so can get closer to objects.   Though the D435 model experiences greater RMS Error (depth noise that increases linearly over distance) than the D415, at close range the differences in error between the D415 and D435 models should be small.

Chen777-cloud, 2020-04-17:
Excuse me, I want to recognize something and acquire its 6D pose with D435i.  Can you give me some instructions？I’m a novice.

MartyG-RealSense, 2020-04-17:
@Steven-ZY   Although the D435i and the RealSense T265 Tracking camera use the same IMU component, they process the data in different ways.  So the T265 has built in support for accessing 6DOF pose data but the D435i does not.

There are workarounds though to get the pose of objects with a D435 type camera using the vision softwares OpenCV or ROS.

https://github.com/IntelRealSense/librealsense/issues/6038#issuecomment-598089815

closed 2020-05-17 by MartyG-RealSense