realsenseai/librealsense #6244: Example request for changing resolution

timprepscius, 2020-04-15:
The simplest example that:

(configurations are arbitrary, but they must have different resolutions)

1.  starts sensor at 848x640 both infrareds and depth
2. grabs X frames and prints out some computation
3.  stops sensor
4.  starts sensor at 1280x800 one infrared
5. grabs X frames and prints out some computation
6.  stops sensor
7. goto 1

hardware reset + pause is not working 
pipe stop is not working
There is some magic code between 3 & 4 which I can't figure out.

ev-mp, 2020-04-20:
@timprepscius, 
What are the camera type, SDK and FW  versions that you're using ? 
What OS is the issue is reproduced on?
Can you share librealsense log file and a code snippet with reproduction for review (if possible)?

RealSenseCustomerSupport, 2020-04-27:
----------------------------------------------

@timprepscius Could you please provide the information requested by ev-mp? Thanks!

closed 2020-05-04 by ev-mp