realsenseai/librealsense #6246: pip install pyrealsense2 DOES NOT work on MAC

vikram09, 2020-04-16:
I bought brand new d435i from Intel website. I want to use it in my python code . I am using OpenCV too in my python. 
    When I do 
 pip install pyrealsense2

I get this error. Please help. Mac OS is very popular platform, how can we not have support for it. 
    Lots of people have mentioned this issue. 

DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Defaulting to user installation because normal site-packages is not writeable
ERROR: Could not find a version that satisfies the requirement pyrealsense2 (from versions: none)
ERROR: No matching distribution found for pyrealsense2

MartyG-RealSense, 2020-04-16:
Pip install for pyrealsense2 on Mac is not supported, unfortunately.  The most recent information on Mac installation of Pyrealsense2 is in the link below:

https://github.com/IntelRealSense/librealsense/issues/5275

dorodnic, 2020-04-16:
Actually it should work with python 3.7.
Check and update the version of python

MartyG-RealSense, 2020-04-16:
@dorodnic  I wasn't aware that pip compatibility had been added for Python 3.7.  Thanks for the update, I will pass it to others who have asked.

vikram09, 2020-04-16:
Yes I have python 3.7 and using Pycharm, The following steps worked BUT, I had to see a youtube video how to install "cmake" and make it run on Mac Terminal  window. 

I wish someone from Intel can make a batch file for all of these cmds. The Camera should work out of box.

From #5275 
git clone https://github.com/IntelRealSense/librealsense
mkdir build & cd build
cmake ../ -DBUILD_PYTHON_BINDINGS=bool:true
make -j4
sudo make install
You should still be in the /build directory so now you want to find the .so files. cd wrapper & cd python
You'll see a bunch of .so files in here, just copy them and then paste them to the root project directory of your project that you want to use pyrealsense2 in. Then you can just import pyrealsense2 in your project modules using import pyrealsense2

closed 2020-04-16 by vikram09